espressif/esp-idf #15911: programming a ESP32C6 over IO12 and IO13

ravencarcass, 2025-05-06:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have an ESP32C6-DevKitC-1 and want to program the device with the IO12 and IO13 pins. 
This one has 2 USBs one with UART and one with serial/JTAG (IO12 and IO13). 
I can now program it with the UART and that works fine, but I am wondering if I can program it with the serial/JTAG.
- I am using ESP-IDF V5.4
- set the following things: JTAG, /dev/ttyUSB0, esp32c6

Does anyone know what I am doing wrong?

wanckl, 2025-05-07:
@ravencarcass Hi, Of course, when connect ESP with USB serial jtag port, it will be an `ttyACM` device in computor, or other name if you are not using Linux, 

Then just similarly `idf.py flash -p /dev/ttyACM0`

Just notice the default print log port is UART, to monitor log on serial jtag port, enable the `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG`

suda-morris, 2025-05-08:
@ravencarcass 

you can also get answer from the AI chatbot in the esp-idf documentation page. e.g.

[image]

ravencarcass, 2025-05-08:
I followed what the AI bot said. 
1) I opened Visual Studio Code
2) I plugged the right USB in my PC
3) I selected the port /dev/ttyACM0
4) I gave the ACM0 port the right rights with `sudo chmod 666 /dev/ttyACM0`
4) I build the project without any problems
5) I press on the flash button and it gave the following things in the terminal.

```
esptool.py v4.8.1
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-C6 (QFN40) (revision v0.0)
Features: WiFi 6, BT 5, IEEE802.15.4
Crystal is 40MHz
MAC: 40:4c:ca:ff:fe:50:65:40
BASE MAC: 40:4c:ca:50:65:40
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Traceback (most recent call last):
  File "/home/robbe/.espressif/python_env/idf5.4_py3.10_env/lib/python3.10/site-packages/esptool/__init__.py", line 1314, in _main
    main()
  File "/home/robbe/.espressif/python_env/idf5.4_py3.10_env/lib/python3.10/site-packages/esptool/__init__.py", line 999, in main
    esp.flash_set_parameters(flash_size_bytes(flash_size))
  File "/home/robbe/.espressif/python_env/idf5.4_py3.10_env/lib/python3.10/site-packages/esptool/loader.py", line 1286, in flash_set_parameters
    self.check_command(
  File "/home/robbe/.espressif/python_env/idf5.4_py3.10_env/lib/python3.10/site-packages/esptool/loader.py", line 510, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "/home/robbe/.espressif/python_env/idf5.4_py3.10_env/lib/python3.10/site-packages/esptool/loader.py", line 479, in command
    p = self.read()
  File "/home/robbe/.espressif/python_env/idf5.4_py3.10_env/lib/python3.10/site-packages/esptool/loader.py", line 412, in read
    return next(self._slip_reader)
StopIteration

A fatal error occurred: The chip stopped responding.

 *  The terminal process "/home/robbe/.espressif/python_env/idf5.4_py3.10_env/bin/python '/home/robbe/esp/esp-idf/components/esptool_py/esptool/esptool.py', '-p', '/dev/ttyACM0', '-b', '460800', '--before', 'default_reset', '--after', 'hard_reset', '--chip', 'esp32c6', 'write_flash', '--flash_mode', 'dio', '--flash_freq', '80m', '--flash_size', '2MB', '0x0', 'bootloader/bootloader.bin', '0x10000', 'spi_lcd_touch.bin', '0x8000', 'partition_table/partition-table.bin'" terminated with exit code: 2. 
```
For me does it look like I didn't gave my USB not the right rights. But I find it weird because I gave them the chmod 666 rights. What could be wrong I can't find directly somethings about it.

wanckl, 2025-05-12:
@ravencarcass Dose you tried only once or it failed every time ?

ravencarcass, 2025-05-12:
@wanckl I've tried it multiple times and failed every time.