espressif/esp-idf #15912: ESP-IDF v5.3.3 TinyUSB can't be switched back to USB/JTAG (regression since 5.3.2)

johnboiles, 2025-05-07:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.3

### Espressif SoC revision.

ESP32-S3 v0.1

### Operating System used.

macOS

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

Related to #9826, previously it was possible to switch out the device from being a TinyUSB peripheral and go back to the built-in USB/JTAG peripheral (e.g. for programming). You could do that with this code:

```C
tusb_cdc_acm_deinit(TINYUSB_CDC_ACM_0);
tinyusb_driver_uninstall();
usb_phy_config_t phy_config = {.controller = USB_PHY_CTRL_SERIAL_JTAG,
                               .target = USB_PHY_TARGET_INT};
usb_phy_handle_t jtag_phy;
usb_new_phy(&phy_config, &jtag_phy);
```

### What is the actual behavior?

This no longer works in v5.3.3 (but works in v5.3.2). I also tested v5.4.1 and master (`v5.5-dev-3510-gaaebc37467`) and those don't work either.

### Steps to reproduce.

Build and flash my attached sample project on an ESP32-S3 device using various IDF versions. Push the GPIO0 button to switch from TinyUSB CDC mode back to USB/JTAG mode. On v5.3.2 the `Espressif Device` will go away and the `USB JTAG/serial debug unit` will show up. On v5.3.3 and above, this won't happen.

[20250506_tinyusb_usb_jtag_regression.zip](https://github.com/user-attachments/files/20073942/20250506_tinyusb_usb_jtag_regression.zip)

### Debug Logs.

```plain

```

### More Information.

I'm testing on macOS but I doubt this is host OS specific, since the USB/JTAG device typically works consistently on every platform.

jrahlf, 2025-05-07:
Try this:
```c
esp_err_t tinyusb_driver_uninstall()
{
    tinyusb_free_descriptors();
    usb_phy_action(phy_hdl, USB_PHY_ACTION_HOST_FORCE_DISCONN);  <<<< added
    return usb_del_phy(phy_hdl);
}
```

johnboiles, 2025-05-08:
@jrahlf same behavior with that change unfortunately. I'm not seeing any non-null `esp_err_t` from any of those functions either (via logging on the hardware UART). It happily succeeds but doesn't bring back the USB/JTAG device.
[image]

atanisoft, 2025-05-08:
What I've seen was /dev/ttyACM0 would become /dev/ttyACM2 (/dev/ttyACM1 is a non-espressif device). There is no persistence unfortunately which causes problems for tools which are trying to toggle DTR/RTS etc bits to reboot the device automatically (which is detected and handled).

atanisoft, 2025-05-08:
I should note, this is with the above "fix" in place btw

johnboiles, 2025-05-08:
@atanisoft do you think that's related to the ability to switch between the USB_ACM TinyUSB device and USB/JTAG built-in device? Or is that a separate issue?

atanisoft, 2025-05-08:
I'm leaning towards related, the switch of the device definitely creates a new ttyACMx device due to the shift from TinyUSB to USB/JTag. There is also a related problem though with ESPHome web installer or esptool.py in that neither of them detect the port change which is problematic as well.

atanisoft, 2025-05-08:
it should really persist the device name (ttyACMx) between restarts regardless of the TinyUSB or USB/JTag. That should fix most of the issues from what I've seen.

jrahlf, 2025-05-09:
When we had this problem (only tested on Windows), we noticed that the switching actually worked, but the PC would not re-enumerate the device. So if you unplugged it and plugged it in again (we have battery power), it worked fine. The added line above fixed the problem for us.

atanisoft, 2025-05-09:
Unfortunately reiterating / changing the port will fail the use case of either bit twiddling of DTR/RTS *OR* simply changing from TinyUSB to/from USB/JTag.

That is one core difference here unfortunately that was working in v5.3.2 but seems broken in v5.3.3/v5.4.1/v5.5.x (master).

johnboiles, 2025-05-09:
> if you unplugged it and plugged it in again (we have battery power), it worked fine

I tried that just now on macOS with your `USB_PHY_ACTION_HOST_FORCE_DISCONN` change and a battery. Same behavior for me -- the device no longer enumerates at all. I'll try Windows too when I'm back in the office.

@atanisoft I bet the behavior of the port staying the same is a very Linux-specific thing. From my experience, that's never been the behavior on macOS or Windows. It may be a clue -- perhaps the TinyUSB interface doesn't get released entirely when going away and that's why the kernel is incrementing the tty #.

I'm going to bite the bullet and bisect the IDF changes :/ (sidenote: I'm so close to having Claude Code be able to do this for me automatically, except that when the USB device goes away I have no way to restart the MCU!)

atanisoft, 2025-05-09:
You are likely right in that the tool isn't releasing the connection immediately (known bug in the tool) but there is still the esptool.py / esphome web tool that also fail similarly for bit twiddling which should work and result in the same TTY device upon completion. I think if we solve the issue of sorting out restarting reliably we can fix the tool issues in parallel.

johnboiles, 2025-05-09:
Ok several commits wouldn't build (probably related to backporting changes from master to v5.3.3) but this is the result of the `git bisect`

```
There are only 'skip'ped commits left to test.
The first bad commit could be any of:
005ae0554a7f19b33c33e4ff500f85e5c382c249
144f8a660e6a032bb535ad7aa962c51df19ddcd1
47fd8aac236050b44f0c94a7ee9b6a3f3b7384c7
19bdc77e558b3da520b9e8234938227681efae8e
We cannot bisect more!
```

I manually confirmed this on the v5.3.3 tag by testing those commits again (starting with oldest):

* `1ef5be3b39 fix(ulp): remove redefinition and unify gpio enum in ulp` - Works (switches back to USB/JTAG)
* `47fd8aac23 feat(hal/usb): Explicitly enable clock and reset USB WRAP on init` - Fails to compile
* `005ae0554a refactor(usb/phy): Do not use deprecated variables in usb_phy` - Fails to compile
* `144f8a660e refactor(usb/phy): Start using values from usb_dwc_info in PHY driver` - Fails to compile
* `19bdc77e55 fix(usb): Backport v5.3 compatibility commit` - Compiles but does not work (does not switch back to USB/JTAG)

johnboiles, 2025-05-09:
Ok, it looks like 47fd8aac23 `feat(hal/usb): Explicitly enable clock and reset USB WRAP on init` definitely causes the issue. It's possible 005ae0554a, 144f8a660e also cause problems. You can test it yourself by checking out 47fd8aac23 and then applying this diff to get it to compile

```diff
diff --git i/components/usb/usb_phy.c w/components/usb/usb_phy.c
index 1de4ebee3b..93778e6cf9 100644
--- i/components/usb/usb_phy.c
+++ w/components/usb/usb_phy.c
@@ -258,7 +258,7 @@ static esp_err_t usb_phy_install(void)
     }
     // Enable USB peripheral and reset the register
     portEXIT_CRITICAL(&phy_spinlock);
-    USB_WRAP_RCC_ATOMIC() {
+    USB_PHY_RCC_ATOMIC() {
         usb_wrap_ll_enable_bus_clock(true);
         usb_wrap_ll_reset_register();
     }
```

johnboiles, 2025-05-09:
And I've found the smoking gun:

```diff
diff --git i/components/usb/usb_phy.c w/components/usb/usb_phy.c
index 1de4ebee3b..f1b4250879 100644
--- i/components/usb/usb_phy.c
+++ w/components/usb/usb_phy.c
@@ -357,7 +357,8 @@ static void phy_uninstall(void)
         p_phy_ctrl_obj = NULL;
         USB_PHY_RCC_ATOMIC() {
             // Disable USB peripheral without reset the module
-            usb_wrap_hal_disable();
+            usb_wrap_ll_enable_bus_clock(false);
         }
     }
     portEXIT_CRITICAL(&phy_spinlock);
```

Changing this line fixes the issue. The weird thing is that [`usb_wrap_hal_disable`](https://github.com/espressif/esp-idf/blob/47fd8aac236050b44f0c94a7ee9b6a3f3b7384c7/components/hal/include/hal/usb_wrap_hal.h#L57) calls [`_usb_wrap_hal_disable`](https://github.com/espressif/esp-idf/blob/47fd8aac236050b44f0c94a7ee9b6a3f3b7384c7/components/hal/usb_wrap_hal.c#L21) which calls [`_usb_wrap_ll_enable_bus_clock(false)`](https://github.com/espressif/esp-idf/blob/47fd8aac236050b44f0c94a7ee9b6a3f3b7384c7/components/hal/esp32s3/include/hal/usb_wrap_ll.h#L219). That should be the same as [`usb_wrap_ll_enable_bus_clock`](https://github.com/espressif/esp-idf/blob/47fd8aac236050b44f0c94a7ee9b6a3f3b7384c7/components/hal/esp32s3/include/hal/usb_wrap_ll.h#L225) calling [`_usb_wrap_ll_enable_bus_clock`](https://github.com/espressif/esp-idf/blob/47fd8aac236050b44f0c94a7ee9b6a3f3b7384c7/components/hal/esp32s3/include/hal/usb_wrap_ll.h#L219)

atanisoft, 2025-05-09:
Interesting finding, I wonder if we can get a debug dump showing the include path used for that specific file. It may yield a clue where there is a missing include perhaps?

johnboiles, 2025-05-09:
What makes you think it would be related to a missing include?

atanisoft, 2025-05-09:
I'm looking at the call to `usb_wrap_hal_disable();` which you noted, which has this call:
```
#if SOC_RCC_IS_INDEPENDENT
#define usb_wrap_hal_disable(...)   _usb_wrap_hal_disable(__VA_ARGS__)
#else
// Use a macro to wrap the function, force the caller to use it in a critical section
// the critical section needs to declare the __DECLARE_RCC_ATOMIC_ENV variable in advance
#define usb_wrap_hal_disable(...) do {(void)__DECLARE_RCC_ATOMIC_ENV; _usb_wrap_hal_disable(__VA_ARGS__);} while(0)
#endif
```

in hal/include/hal/usb_wrap_hal.h which should pick the first option and likely will not call the API we expect due to `__VA_ARGS__` not being used by the subsequent calls (at all!) which should ultimately [this](https://github.com/espressif/esp-idf/blob/b62c2cef1b7e78c96fad4d4aa6a9f41320c3013c/components/hal/usb_wrap_hal.c#L21-L24) function as you noted. But I'm wondering if it maybe not calling it at all due to the include paths or simply linking not working correctly here.

johnboiles, 2025-05-09:
My bad everyone I must have made a mistake; I can no longer repro the issue on 47fd8aac23. I now believe the issue is in 005ae0554a. Specifically this part (which makes much more sense). I'll triple check this and open a PR for further discussion.

```diff
diff --git a/components/usb/usb_phy.c b/components/usb/usb_phy.c
index 1de4ebee3b..43cceeb595 100644
--- a/components/usb/usb_phy.c
+++ b/components/usb/usb_phy.c
@@ -13,12 +13,9 @@
 #include "esp_private/usb_phy.h"
 #include "soc/usb_dwc_periph.h"
 #include "hal/usb_wrap_hal.h"
-#include "hal/usb_serial_jtag_hal.h"
 #include "esp_rom_gpio.h"
 #include "driver/gpio.h"
@@ -307,13 +307,6 @@ esp_err_t usb_new_phy(const usb_phy_config_t *config, usb_phy_handle_t *handle_r
         usb_wrap_hal_phy_set_external(&phy_context->wrap_hal, (config->target == USB_PHY_TARGET_EXT));
 #endif
     }
-#if SOC_USB_SERIAL_JTAG_SUPPORTED
-    else if (config->controller == USB_PHY_CTRL_SERIAL_JTAG) {
-        usb_serial_jtag_hal_phy_set_external(&phy_context->usj_hal, (config->target == USB_PHY_TARGET_EXT));
-        phy_context->otg_mode = USB_OTG_MODE_DEVICE;
-        phy_context->otg_speed = USB_PHY_SPEED_FULL;
-    }
-#endif
 
     if (config->target == USB_PHY_TARGET_INT) {
         gpio_set_drive_capability(USBPHY_DM_NUM, GPIO_DRIVE_CAP_3);

atanisoft, 2025-05-09:
That looks extra promising, almost like an unintentional removal since it has no comments on why it was there.
   
   
   Message ID: ***@***.***>

johnboiles, 2025-05-09:
I've tested the fix on v5.3.3, v5.4.1 and master (v5.5-dev-3568-g565cca2fee) and it is working! On v5.3.3 and v5.4.1 it looks like this:

```diff
diff --git i/components/usb/usb_phy.c w/components/usb/usb_phy.c
index 4c1241b83d..cfad1a64fa 100644
--- i/components/usb/usb_phy.c
+++ w/components/usb/usb_phy.c
@@ -13,6 +13,7 @@
 #include "esp_private/periph_ctrl.h"
 #include "esp_private/usb_phy.h"
 #include "soc/usb_dwc_periph.h"
+#include "hal/usb_serial_jtag_hal.h"
 #include "hal/usb_wrap_hal.h"
 #include "hal/usb_utmi_hal.h"
 #include "esp_rom_gpio.h"
@@ -371,6 +372,13 @@ esp_err_t usb_new_phy(const usb_phy_config_t *config, usb_phy_handle_t *handle_r
         usb_wrap_hal_phy_set_external(&phy_context->wrap_hal, (phy_target == USB_PHY_TARGET_EXT));
 #endif
     }
+#if SOC_USB_SERIAL_JTAG_SUPPORTED
+    else if (config->controller == USB_PHY_CTRL_SERIAL_JTAG) {
+        usb_serial_jtag_hal_phy_set_external(NULL, (config->target == USB_PHY_TARGET_EXT));
+        phy_context->otg_mode = USB_OTG_MODE_DEVICE;
+        phy_context->otg_speed = USB_PHY_SPEED_FULL;
+    }
+#endif
 
     // For FSLS PHY that shares pads with GPIO peripheral, we must set drive capability to 3 (40mA)
     if (phy_target == USB_PHY_TARGET_INT) {
```

And on master it looks like this:

```diff
diff --git i/components/usb/usb_phy.c w/components/usb/usb_phy.c
index 83d1930040..3f456f9c7c 100644
--- i/components/usb/usb_phy.c
+++ w/components/usb/usb_phy.c
@@ -13,6 +13,7 @@
 #include "esp_private/usb_phy.h"
 #include "esp_private/critical_section.h"
 #include "soc/usb_dwc_periph.h"
+#include "hal/usb_serial_jtag_hal.h"
 #include "hal/usb_wrap_hal.h"
 #include "hal/usb_utmi_hal.h"
 #include "esp_rom_gpio.h"
@@ -347,6 +348,12 @@ esp_err_t usb_new_phy(const usb_phy_config_t *config, usb_phy_handle_t *handle_r
         usb_wrap_hal_phy_set_external(&phy_context->wrap_hal, (phy_target == USB_PHY_TARGET_EXT));
 #endif
     }
+#if SOC_USB_SERIAL_JTAG_SUPPORTED
+    else if (config->controller == USB_PHY_CTRL_SERIAL_JTAG) {
+        usb_serial_jtag_hal_phy_set_external(NULL, (config->target == USB_PHY_TARGET_EXT));
+        phy_context->otg_mode = USB_OTG_MODE_DEVICE;
+    }
+#endif
 
     // For FSLS PHY that shares pads with GPIO peripheral, we must set drive capability to 3 (40mA)
     if (phy_target == USB_PHY_TARGET_INT) {
```

Additionally it's possible to work around this issue in user code by calling `usb_serial_jtag_hal_phy_set_external(NULL, false);` after `usb_new_phy` when re-initializing the USB/JTAG device. I'll still open a PR since I believe this to be a regression since the `USB_PHY_CTRL_SERIAL_JTAG` is no longer used anywhere without this fix.

johnboiles, 2025-05-09:
PR https://github.com/espressif/esp-idf/pull/15932 is open!

closed 2025-05-15 by espressif-bot

johnboiles, 2025-05-16:
Woohoo thanks!

commit 25571321 (2025-05-27): fix(usb/phy): Fix ability to switch back to USB/JTAG after uninstalling TinyUSB

Fixes #15912. This is a revert of a change made in 005ae0554 that breaks the
ability to re-initalize the USB/JTAG device after uninstalling TinyUSB.

Closes https://github.com/espressif/esp-idf/issues/15912