espressif/esp-idf #15913: Connecting to remote Bluetooth devices takes too long time

iv-sys, 2025-05-07:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.4.1

### Espressif SoC revision.

ESP32 Pico, ESP32U4WDH

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-PICO-MINI02, also tested on ESP32U4WDH on custom board

### Power Supply used.

USB

### What is the expected behavior?

Connecting from ESP32 to a remote Bluetooth device, such as a third party headphones, will take about 1 second.

### What is the actual behavior?

Connecting from ESP32 to a remote Bluetooth headphones takes from 20 seconds to 1 minute.

### Steps to reproduce.

1. Find a third-party Bluetooth 5.0 or 5.1 audio sink device, such as Lenovo thinkplus headphones.
2. Use A2DP_SOURCE example provided with ESP IDF. 
3. Modify variable remote_device_name in this example to reflect your headphones device name.
4. Compile it for ESP32-PICO-MINI02 or similar development board.
5. Flash to the development board using USB connection to the board.
6. Start Monitoring your project.
7. Power ON your headphones.
8. Watch the time it will take for ESP32 board to connect to the headphones.
9. In the attached log, it took one minute and a half for ESP32 board to connect to the headphones. Once the connection is established, it works OK without issues, even if you separate ESP32 and headsets 10 meters apart.

### Debug Logs.

```plain
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6276
load:0x40078000,len:15748
load:0x40080400,len:4
--- 0x40080400: _init at ??:?

ho 8 tail 4 room 4
load:0x40080404,len:3860
entry 0x4008063c
I (32) boot: ESP-IDF v5.4.1 2nd stage bootloader
I (32) boot: compile time May  6 2025 14:35:49
I (32) boot: Multicore bootloader
I (33) boot: chip revision: v3.0
I (36) boot.esp32: SPI Speed      : 40MHz
I (40) boot.esp32: SPI Mode       : DIO
I (43) boot.esp32: SPI Flash Size : 2MB
I (47) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00177000
I (80) boot: End of partition table
I (83) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1d47ch (119932) map
I (132) esp_image: segment 1: paddr=0002d4a4 vaddr=3ff80000 size=0001ch (    28) load
I (132) esp_image: segment 2: paddr=0002d4c8 vaddr=3ffbdb60 size=02b50h ( 11088) load
I (140) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=79c78h (498808) map
I (313) esp_image: segment 4: paddr=000a9ca0 vaddr=3ffc06b0 size=02050h (  8272) load
I (317) esp_image: segment 5: paddr=000abcf8 vaddr=40080000 size=16c64h ( 93284) load
I (365) boot: Loaded app from partition at offset 0x10000
I (365) boot: Disabling RNG early entropy source...
I (375) cpu_start: Multicore app
I (384) cpu_start: Pro cpu start user code
I (384) cpu_start: cpu freq: 160000000 Hz
I (384) app_init: Application information:
I (384) app_init: Project name:     a2dp_source
I (388) app_init: App version:      1
I (391) app_init: Compile time:     May  6 2025 14:33:48
I (396) app_init: ELF file SHA256:  d6939a7eb...
I (401) app_init: ESP-IDF:          v5.4.1
I (405) efuse_init: Min chip rev:     v0.0
I (408) efuse_init: Max chip rev:     v3.99 
I (412) efuse_init: Chip rev:         v3.0
I (416) heap_init: Initializing. RAM available for dynamic allocation:
I (423) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (428) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (433) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (438) heap_init: At 3FFCA450 len 00015BB0 (86 KiB): DRAM
I (443) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (448) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (454) heap_init: At 40096C64 len 0000939C (36 KiB): IRAM
I (461) spi_flash: detected chip: generic
I (463) spi_flash: flash io: dio
W (466) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (479) coexist: coex firmware version: e727207
I (483) main_task: Started on CPU0
I (493) main_task: Calling app_main()
=== APP 1 ===
I (503) BTDM_INIT: BT controller compile version [dc1cd58]
I (503) BTDM_INIT: Bluetooth MAC: 4c:75:25:e4:ab:e2
I (513) phy_init: phy_version 4860,6b7a6e5,Feb  6 2025,14:47:07
=== APP 2 ===
=== APP 3 ===
=== APP STARTING ===
GAP CB: A, 00000000
GAP EIR: 4, 07050309
APP: Init starting
W (1143) BT_BTC: A2DP Enable without AVRC
72730413 A2D Init 1
GAP CB: A, 00000000
GAP EIR: 4, 07050309
Dev0: 45AECB83
Dev1: E3A373CC
Actual Size: 2 0
=== APP RUNNING ===
APP: Disc starting
GAP CB: 1, 00000001
GAP CB Disc State CHANGE to: 1
GAP CB: 0, E3A373CC
GAP CB Disc RESULT: E3A373CC, 3
GAP BDA: cc:73:a3:e3:b4:e3
I (13603) BT_AV: --Class of Device: 0x2360324
I (13603) BT_AV: --RSSI: 4294967246
I (13603) BT_AV: Cancel device discovery ...
GAP CB: 1, 00000000
GAP CB Disc State CHANGE to: 0
APP: Init OK
W (13623) BT_APPL: reset flags
2485E0F0 A2D connecting
W (18743) BT_SDP: SDP - Rcvd conn cnf with error: 0x4  CID 0x40

W (18743) BT_HCI: hcif conn complete: hdl 0xfff, st 0x4
GAP CB: 10, 00000104
GAP ACL: 260, E3A373CC
W (18753) BT_BTC: BTA_AV_OPEN_EVT::FAILED status: 2

4A31B504 --- A2D DISC --- 0
APP: Disc starting
GAP CB: 1, 00000001
GAP CB Disc State CHANGE to: 1
GAP CB: 1, 00000000
GAP CB Disc State CHANGE to: 0
APP: Disc starting
GAP CB: 1, 00000001
GAP CB Disc State CHANGE to: 1
GAP CB: 1, 00000000
GAP CB Disc State CHANGE to: 0
APP: Disc starting
GAP CB: 1, 00000001
GAP CB Disc State CHANGE to: 1
GAP CB: 1, 00000000
GAP CB Disc State CHANGE to: 0
APP: Disc starting
GAP CB: 1, 00000001
GAP CB Disc State CHANGE to: 1
GAP CB: 1, 00000000
GAP CB Disc State CHANGE to: 0
APP: Disc starting
GAP CB: 1, 00000001
GAP CB Disc State CHANGE to: 1
GAP CB: 1, 00000000
GAP CB Disc State CHANGE to: 0
APP: Disc starting
GAP CB: 1, 00000001
GAP CB Disc State CHANGE to: 1
GAP CB: 1, 00000000
GAP CB Disc State CHANGE to: 0
APP: Disc starting
GAP CB: 1, 00000001
GAP CB Disc State CHANGE to: 1
GAP CB: 0, E3A373CC
GAP CB Disc RESULT: E3A373CC, 3
GAP BDA: cc:73:a3:e3:b4:e3
I (98303) BT_AV: --Class of Device: 0x2360324
I (98303) BT_AV: --RSSI: 4294967230
I (98303) BT_AV: Cancel device discovery ...
GAP CB: 1, 00000000
GAP CB Disc State CHANGE to: 0
APP: Init OK
2485E0F0 A2D connecting
W (99863) BT_HCI: hcif conn complete: hdl 0x80, st 0x0
GAP CB: 10, 00000000
GAP ACL: 0, E3A373CC
GAP CB: 15, E3A373CC
I (100073) BT_AV: event: 21
W (100133) BT_APPL: new conn_srvc id:18, app_id:0
5AC4FF1B +++ A2D CONN +++
APP: CON OK
5FB71562 ACK_OK
APP: Media Ready
W (100153) BT_APPL: new conn_srvc id:18, app_id:1
I (100173) BT_LOG: bta_av_link_role_ok hndl:x41 role:0 conn_audio:x1 bits:1 features:x8400

4E02DC19 MEDIA STARTED
APP: Media Started
W (101153) BT_L2CAP: L2CAP - rcvd conn req for unknown PSM: 23
```

ESP-YTGerd, 2025-05-12:
Hi, @iv-sys 
We recommend performing device discovery before attempting to connect to the target device. Conducting an inquiry can significantly improve the success rate of paging the target device, thereby increasing the overall chance of a successful connection.

If the ESP32 still struggles to establish a connection even after discovering the remote device, you can try setting the ESP32 to non-discoverable and non-connectable mode before initiating the connection by calling:
`esp_bt_gap_set_scan_mode(ESP_BT_NON_CONNECTABLE, ESP_BT_NON_DISCOVERABLE)`
This might help improve the success rate.

Additionally, regarding the issue you mentioned,`Sometimes ESP32 locks in the discovery phase and no any discovery events come after the call to esp_bt_gap_start_discovery. But this is a separate issue with the GAP, which is of a less dramatical impact.` 
Trying reducing the inquiry duration instead of using the default value in the example (10 * 1.28s). A shorter duration may yield better results.

iv-sys, 2025-05-12:
Hi ESP-YTGerd,

**Regarding Discovery**:
If it is used by itself, without A2D, it works reasonably well. One trick you still have to use - do not let it to timeout by its own. I have to call `esp_bt_gap_cancel_discovery` manually, otherwise next attempt to discover will fail. It seems that discovery timeout does not release resources: if after the discovery times out I turn off the peer device, I will get a log message that the connection was closed.
But the main problem: if I use `esp_a2d_source_connect` and it fails, then all consecutive discoveries will slow down significantly. I tried to increase various numbers (ACL) connections in the config menu, but it did not help. 

**Regarding esp_a2d_source_connect**:
Yes, most certainly, we are using the `esp_bt_gap_set_scan_mode(ESP_BT_NON_CONNECTABLE, ESP_BT_NON_DISCOVERABLE)`. But nothing helps. `esp_a2d_source_connect` fails many times before it finally establishes the connection. We tried to call it immediately after the peer is discovered, after receiving the `ESP_BT_GAP_DISCOVERY_STOPPED` callback, after waiting for 10ms, etc - but we never could get a reliable connection with the remote audio sink within even 10 seconds.

Our customers keep our device powered off most of the time and turn it on only when they need it (its a medical device for specific situations). The maximum connection time they can tolerate is about 1-2 seconds - they say Widows devices can do it consistently. We will either loose this market or will have to find a reliable solution.

Best Regards,
_iv-sys_

ESP-YTGerd, 2025-05-14:
Hi, @iv-sys 
I tested both of the scenarios you mentioned using ESP32 with a Sony WH-1000XM4 headset and ESP32-to-ESP32 simulation locally, and I wasn’t able to reproduce the issue you're encountering.

I noticed from your logs that there's a `W (18743) BT_HCI: hcif conn complete: hdl 0xfff, st 0x4` message.  `0x4 = Page Timeout` ,The default page timeout is 5.12 seconds, which means if the connection isn't successfully established within that time, the event will be triggered and the connection will fail.

So, I’d like to know: how often does it take 5–10 seconds to successfully connect? In such cases, do you consistently see the page timeout event, and are multiple connection attempts being made as a result?

Additionally, does reducing the distance between the ESP32 and the headset improve the success rate?

If possible, could you also test locally using your current code and two ESP32 boards — one running A2DP Source and the other A2DP Sink — and check whether the connection success rate is similar to what you're seeing with the headset?

iv-sys, 2025-05-14:
Hi ESP-YTGerd,
Thank you for the investigation. I will prepare the detailed steps to reproduce, with the logs and exact timing. Will also use the boards and headphones that you have mentioned. This will take me about a week or two. Please provide any other details and scenarios that you would like to see. Meanwhile, let me describe what I already have:

1. There is a scenario where connection is established reasonably well. This is when I already have headphones ON, they are actively waiting for incoming connection and only then I turn the ESP32 board on. The discovery is done within a second and the esp_a2d_source_connect is also established within a second. No retries in 95% of cases.

2. Here where it takes 10 to 15 seconds at best: I power on the ESP32. It starts discovery. ESP32 keeps retrying discovery because headphones are off. Then I turn the headphones on. In such situation Windows reconnects to headphones within 1 second, if I have them paired before. But ESP32 keeps retrying to discovery from 10 to 20 seconds before it discovers them (there are no other sources that are trying to connect to them at this time). I set timeouts anywhere from 100 ms to 13 seconds, not much difference. Once discovered, esp_a2d_source_connect connects within 2-10 seconds, or from the first to third esp_a2d_source_connect attempt. It is always 0x4 page timeout event. Reducing esp_a2d_source_connect timeout from 5 seconds to 3 seconds makes things worse. The very first esp_a2d_source_connect is usually more lucky and connects in 60% cases. If it is a second reconnect cycle (as the result of phones power cycled after first connection), then the chances for esp_a2d_source_connect to succeed is less and it takes longer.
 
3. The worst case: I turn Off the phones while ESP32 tries to connect to them during the active esp_a2d_source_connect call. Then the esp_a2d_source_connect does not timeout as normal after 5 seconds. It timeouts after about 30 seconds or longer, and ignores esp_a2d_source_disconnect calls. There are no 0x4 page timeout events here.

4. Signal strength and interference. I have a separate environment with no other 2.5GHz transmitters. Normally, I place ESP32 0.8 meters (2-3 feet) away from the phones with RSSI at -50dB...-60dB. If I place them as close as possible with RSSI at -30dB, the results are exactly the same. Once the connection established, it works very well even at 10 meters and with multiple Bluetooth traffic around.

Again, please feel free to comment on specific scenarios that you would like me to include in the further testing. I have several ESP32-PICO development boards and multiple ESP32-U4WDH custom boards with 1PPM crystals and large well tuned antennas. Can add additional printouts or modifications to the bluedroid if you need.

Best Regards,
iv-sys

ESP-YTGerd, 2025-05-15:
Hi, @iv-sys 
I followed the steps you provided but still couldn’t reproduce the issue on my side. Since I’m not sure what modifications you’ve made based on our example, I’ve added some debug information based on v5.4.1. Could you please help reproduce the issue again and share the corresponding logs with me? That would help with further analysis and troubleshooting.

[debug.zip](https://github.com/user-attachments/files/20220691/debug.zip)

iv-sys, 2025-05-15:
Hi @ESP-YTGerd,
Thank you. It will take me few days. Will definitely let you know the results.
Best regards,
_iv-sys_

iv-sys, 2025-05-16:
Hi @ESP-YTGerd ,

This is what I found:

Modern headphones, when they are first used, will become discoverable right after you power them ON. But once they have been paired with an audio source like Windows desktop computer, they are no longer discoverable for the first 30-40 seconds after power ON. Instead, they are giving the priority to the previously paired device and actively attempting themselves to connect to it. Other platforms (Windows, iPhone, Qualcomm, Cypress, TI, Jeili, etc) are using this behavior and that's why they connect to the headphones within less than one second. ESP32 waits for the headphones to become discoverable, which headphones consider a part of low priority pairing process, and delay it for 30-40 seconds after the start up. Other platforms do not use the discovery after the pairing had been made.

I recorded 2 log files. First, [log.a2dp_source.txt](https://github.com/user-attachments/files/20236946/log.a2dp_source.txt), is the exact a2dp_source example application from 5.4.1, where the only change is the sink device name. It calls 
`esp_bt_gap_set_scan_mode(ESP_BT_NON_CONNECTABLE,ESP_BT_NON_DISCOVERABLE); `
You will see that it takes about 30 seconds for ESP32 board to connect to the headphones there.

The second log file, [log.a2dp_MySrc.txt](https://github.com/user-attachments/files/20236947/log.a2dp_MySrc.txt), is a simplified version of the above example project: it does not use the discovery and calls the `esp_a2d_source_connect` directly. What is important, it calls 
`esp_bt_gap_set_scan_mode( ESP_BT_CONNECTABLE, ESP_BT_NON_DISCOVERABLE); `
As the result, the ESP32 board receives an incoming ACL connection request from the headphones within 100 milliseconds after they are powered on! I think other platforms use it and convert somehow into a a2dp connection right away. If ESP32 could do something similar, that would be a perfect solution. 

Other attached files: 
- [MySrc.patch](https://github.com/user-attachments/files/20236949/MySrc.patch) - some extra log messages in the bludroid regarding role changes.
- [a2dp_MySrc.zip](https://github.com/user-attachments/files/20236950/a2dp_MySrc.zip) - source code for my simplified example project. 
- Five headphones that I tested. They all were bought within last few months. They all behave the same as described above. Here is their picture:

[image]

ESP-YTGerd, 2025-05-16:
Hi, @iv-sys 
Thank you for your testing and detailed log. Based on the information you provided, we have the following analysis:

- The headset appears to have the following behavior: during the first power-on and pairing, it stays in a waiting state, expecting other devices to initiate the connection.
- After a successful connection, on subsequent power cycles, the headset actively attempts to reconnect to the last connected device. During this phase, it may not be discoverable, as it is busy trying to initiate a connection.
- This explains why it took ESP32 around 30 seconds to discover the headset during inquiry — because the headset only became discoverable after it failed to reconnect to the previous device.
- In your first log, we can see that ESP32 took about 30 seconds`(1321 to 32581)` from inquiry start to discovering the headset, but only about 1.55 seconds `(32581 to 34131)` to establish the connection after that.

Therefore, this behavior is caused by the headset’s internal mechanism and not an issue on ESP32's side.

You may try the following test to confirm:

1. Factory reset the headset (clear pairing history).
2. Pair it with ESP32 as the first device.
3. Power it off and on again.
4. Then try to connect to it using Windows and observe the time it takes.

Regarding your second question — connecting without a discovery process — here's our analysis:

From the logs you provided, we can see that the connection attempt initiated by ESP32 failed. The error reported is a page timeout, which indicates that the headset did not respond to the page request.
Shortly after, the headset initiated a connection itself, and ESP32 accepted the request, resulting in a successful connection.
The reason the headset initiated this connection is likely due to ESP32 being the most recently connected device, so the headset attempted to automatically reconnect to it after power-on.
Therefore, the successful connection was not due to ESP32 skipping inquiry, but rather a result of the headset's auto-reconnection mechanism.

So to shorten the connection time between the ESP32 and the Bluetooth headset, you can consider the following methods:
1. Reset the headset before each connection
Resetting the headset clears any previous pairing records. Upon power-up, the headset will not attempt to auto-connect to previously paired devices and will instead enter a connectable and discoverable state immediately. This allows the ESP32 to quickly perform inquiry and initiate the connection.
2. Force the headset into "connectable + discoverable" mode
Some headsets allow you to hold the pairing button to manually enter this mode, even if they were previously paired. In this state, the headset will actively respond to inquiry and page requests from the ESP32, significantly reducing the time needed to establish the connection.

iv-sys, 2025-05-16:
Hi @ESP-YTGerd,

Regarding your first point

> Therefore, this behavior is caused by the headset’s internal mechanism and not an issue on ESP32's side.

The headsets' behavior is logical and expected. I am afraid that all modern headsets will behave the same. Imagine that your headset will become discoverable and connectable each time after you power it on. Then all smart phones within 100 meters radius will instantly see it and probably will try to pair with it or connect to it. It is very unsecure and will create unnecessary network traffic. Instead, the headsets do what they should do: effectively connect to the last paired device, without letting new devices to connect.

All headsets that I have even do not have a pair button. Neither they have a factory reset nor other admin options. Even if I would be able to reset the headphone, you will still have to convince to do the same 1000000 of our customers, which is impossible. No one will like to do it and they will simply turn away to many other options available on the market.

Regarding the second scenario when headset initiates the connection:

> Resetting the headset clears any previous pairing records. Upon power-up, the headset will not attempt to auto-connect to previously paired devices and will instead enter a connectable and discoverable state immediately.

Again, this is what we cannot do. The headset behaves exactly as it should. It helps ESP32 to quickly and securely establish the connection, which it should accept, not reject. It will be absolutely impossible to ask all our 1000000 and all your (even more) future customers to reset or manipulate their headsets. They will simply buy another solution. I don't even think that Bluetooth SIG will certify our device if we design it this way.

Best Regards,
_iv-sys_

iv-sys, 2025-05-22:
Hi @ESP-YTGerd ,

> In fact, when the ESP32 was the last device connected to the headset, the headset’s reconnection attempt was accepted, and the connection was established promptly.

Unfortunately, it is not the case. Even if the ESP32 was the one and only ever connected device to the headphone, it takes ESP32 30 seconds to connect. You can see it in the log. Headphone initiates a connection, ESP32 accepts it, but then closes it - instead of using it. Then it waits 30 seconds till headphone gives up to establish the connection. Then ESP32 initiates its own connection, which succeeds, 30 seconds later.

> The only scenario causing delays is when the last paired device is not the ESP32, and the headset is busy trying to reconnect to that other device. During that time, it is not discoverable, which causes the ESP32's inquiry process to take longer.

Again, this is not the case. Even if the ESP32 was the only ever connected device to the headphone, once the headphone decided that it is paired to the ESP32, it takes ESP32 30 seconds to connect. 

> This issue is not unique to the ESP32 — it would affect any device attempting to connect to the headset under the same circumstances.

No, this is not the case. The Bluetooth A2DP and Generic Audio/Video Distribution Profile (GAVDP) specification explicitly tells that headphone is allowed to initiate the connection and the source should accept it. Please see core specification sections:
> Generic Audio/Video Distribution Profile (GAVDP)
> 1.6 Configurations and Roles
> 3.1 Audio Streaming setup
> 3.1.1 Connection Establishment

I tested with several other microchip vendors and they work perfectly fine. Connection is always established within 1 second.

Other users are also complaining about this defect. Please see for example:
https://esp32.com/viewtopic.php?p=147936

sourxsunny, 2025-06-28:
I also face the same problem.
Even if the ESP32 was the one and only ever connected device to the headphone.
Headphone initiates a connection, ESP32 accepts it, but then closes it - instead of using it.

sourxsunny, 2025-07-02:
release/v5.4