espressif/esp-idf #15916: ESP-S3 tinyusb freertos os build error

Carlos1810044, 2025-05-07:
IDF version: 5.4.1

Operating System: Linux

How build project: VS Code IDE

Chip: Esp32-S3-DevKitC-1 

Expected behavior:
That the program builds successfully. It does build without the OS change, by using CFG_TUSB_OS OPT_OS_NONE. But i am planing to use freertos to expand the project CFG_TUSB_OS OPT_OS_FREERTOS.

Actual behavior:
Getting a build error when i change the OS to freertos.

Steps to reproduce: 
1. get the https://components.espressif.com/components/espressif/tinyusb/versions/0.18.0~2
2.  Started by using the tinyusb uac2 example :https://github.com/espressif/tinyusb/tree/4095aba50e97e84b944693b38ab58b54e14bd311/examples/device/uac2_headset
3. Made own files by coping the example over in own main directory and link those files to tinyusb
4. Changed CFG_TUSB_OS from none to OPT_OS_FREERTOS in tusb_config.h
5. added (to tusb_config.h)#define CFG_TUSB_OS_INC_PATH freertos/
6. build -> error...

Stack trace: 
[...]/USBAudio/main/usb_descriptors.c
In file included from [...]/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/portable.h:57,
                 from [...]/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:69,
                 from [...]/USBAudio/managed_components/espressif__tinyusb/src/osal/osal_freertos.h:31,
                 from [...]/USBAudio/managed_components/espressif__tinyusb/src/osal/osal.h:57,
                 from [...]/USBAudio/managed_components/espressif__tinyusb/src/tusb.h:38,
                 from [...]/USBAudio/main/usb_descriptors.c:1:
[...]/USBAudio/managed_components/espressif__tinyusb/src/osal/osal_freertos.h: In function 'osal_semaphore_post':
[...]/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:474:5: error: implicit declaration of function 'traceISR_EXIT_TO_SCHEDULER' [-Wimplicit-function-declaration]
  474 |     traceISR_EXIT_TO_SCHEDULER(); \
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
[...]/v5.4.1/esp-idf/components/esp_common/include/esp_macros.h:46:64: note: in expansion of macro 'portYIELD_FROM_ISR_NO_ARG'
   46 | #define CHOOSE_MACRO_VA_ARG_INN(one, two, MACRO1, MACRO2, ...) MACRO1
      |                                                                ^~~~~~
[...]/v5.4.1/esp-idf/components/esp_common/include/esp_macros.h:47:71: note: in expansion of macro 'CHOOSE_MACRO_VA_ARG_INN'
   47 | #define CHOOSE_MACRO_VA_ARG(MACRO_WITH_ARGS, MACRO_WITH_NO_ARGS, ...) CHOOSE_MACRO_VA_ARG_INN(0, ##__VA_ARGS__, MACRO_WITH_ARGS, MACRO_WITH_NO_ARGS, 0)
      |                                                                       ^~~~~~~~~~~~~~~~~~~~~~~
[...]/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:492:33: note: in expansion of macro 'CHOOSE_MACRO_VA_ARG'
  492 | #define portYIELD_FROM_ISR(...) CHOOSE_MACRO_VA_ARG(portYIELD_FROM_ISR_ARG, portYIELD_FROM_ISR_NO_ARG, ##__VA_ARGS__)(__VA_ARGS__)
      |                                 ^~~~~~~~~~~~~~~~~~~
[...]/USBAudio/managed_components/espressif__tinyusb/src/osal/osal_freertos.h:131:37: note: in expansion of macro 'portYIELD_FROM_ISR'
  131 |     if ( xHigherPriorityTaskWoken ) portYIELD_FROM_ISR();
      |                                     ^~~~~~~~~~~~~~~~~~
[4/7] Completed 'bootloader'
ninja: build stopped: subcommand failed.

 *  The terminal process "ninja" failed to launch (exit code: 1). 

Extra info: 
Found this in the stack trace. A not needed tag. From osal_freertos.h lines 122-138.
TU_ATTR_ALWAYS_INLINE static inline bool osal_semaphore_post(osal_semaphore_t sem_hdl, bool in_isr) {
  if ( !in_isr ) {
    return xSemaphoreGive(sem_hdl) != 0;
  } else {
    BaseType_t xHigherPriorityTaskWoken = pdFALSE;
    BaseType_t res = xSemaphoreGiveFromISR(sem_hdl, &xHigherPriorityTaskWoken);

#if CFG_TUSB_MCU == OPT_MCU_ESP32S2 || CFG_TUSB_MCU == OPT_MCU_ESP32S3
    // not needed after https://github.com/espressif/esp-idf/commit/c5fd79547ac9b7bae06fa660e9f814d18d3390b7
    if ( xHigherPriorityTaskWoken ) portYIELD_FROM_ISR();
#else
    portYIELD_FROM_ISR(xHigherPriorityTaskWoken);
#endif

    return res != 0;
  }
}

tore-espressif, 2025-05-07:
@Carlos1810044 thank you for the report.

I suspect there is something wrong with you project setup.

Your missing defintion of `traceISR_EXIT_TO_SCHEDULER` is [here](https://github.com/espressif/esp-idf/blob/master/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h#L1489), it is defined if `ESP_PLATFORM` is defined, which is default for esp-idf projects.

> added (to tusb_config.h)#define CFG_TUSB_OS_INC_PATH freertos/

If your projects is setup correctly, this is not needed, the include path is resolved by tinyusb https://github.com/espressif/tinyusb/blob/release/v0.18/src/common/tusb_mcu.h#L370

Could you pass you full build command and output?

Carlos1810044, 2025-05-07:
idf.py -v build
Checking Python dependencies...
Python requirements are satisfied.
Constraint file: [...]/.espressif/espidf.constraints.v5.4.txt
Requirement files:
 - [...]/v5.4.1/esp-idf/tools/requirements/requirements.core.txt
Python being checked: [...]/.espressif/python_env/idf5.4_py3.12_env/bin/python
Checking used Python interpreter...
Executing action: all (aliases: build)
Running ninja in directory [...]/USBAudio/build
Executing "ninja -v all"...
[1/7] cd [...]/USBAudio/build/bootloader && /usr/bin/cmake --build .
[1/1] cd [...]/USBAudio/build/bootloader/esp-idf/esptool_py && [...]/.espressif/python_env/idf5.4_py3.12_env/bin/python [...]/v5.4.1/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 [...]/USBAudio/build/bootloader/bootloader.bin
Bootloader binary size 0x49e0 bytes. 0x3620 bytes (42%) free.
[2/7] cd [...]/USBAudio/build/bootloader && /usr/bin/cmake -E echo_append
[3/7] [...]/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DMD5_ENABLED=1 -DSERIAL_FLASHER_BOOT_HOLD_TIME_MS=50 -DSERIAL_FLASHER_BOOT_INVERT=false -DSERIAL_FLASHER_DEBUG_TRACE=false -DSERIAL_FLASHER_INTERFACE_UART -DSERIAL_FLASHER_RESET_HOLD_TIME_MS=100 -DSERIAL_FLASHER_RESET_INVERT=false -DSERIAL_FLASHER_WRITE_BLOCK_RETRIES=3 -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -DUNITY_INCLUDE_CONFIG_H -I[...]/v5.4.1/esp-idf/components/xtensa/esp32s3/include -I[...]/v5.4.1/esp-idf/components/xtensa/include -I[...]/v5.4.1/esp-idf/components/xtensa/deprecated_include -I[...]/USBAudio/build/config -I[...]/v5.4.1/esp-idf/components/newlib/platform_include -I[...]/v5.4.1/esp-idf/components/freertos/config/include -I[...]/v5.4.1/esp-idf/components/freertos/config/include/freertos -I[...]/v5.4.1/esp-idf/components/freertos/config/xtensa/include -I[...]/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/include -I[...]/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I[...]/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -I[...]/v5.4.1/esp-idf/components/freertos/esp_additions/include -I[...]/v5.4.1/esp-idf/components/esp_hw_support/include -I[...]/v5.4.1/esp-idf/components/esp_hw_support/include/soc -I[...]/v5.4.1/esp-idf/components/esp_hw_support/include/soc/esp32s3 -I[...]/v5.4.1/esp-idf/components/esp_hw_support/dma/include -I[...]/v5.4.1/esp-idf/components/esp_hw_support/ldo/include -I[...]/v5.4.1/esp-idf/components/esp_hw_support/debug_probe/include -I[...]/v5.4.1/esp-idf/components/esp_hw_support/port/esp32s3/. -I[...]/v5.4.1/esp-idf/components/esp_hw_support/port/esp32s3/include -I[...]/v5.4.1/esp-idf/components/heap/include -I[...]/v5.4.1/esp-idf/components/heap/tlsf -I[...]/v5.4.1/esp-idf/components/log/include -I[...]/v5.4.1/esp-idf/components/soc/include -I[...]/v5.4.1/esp-idf/components/soc/esp32s3 -I[...]/v5.4.1/esp-idf/components/soc/esp32s3/include -I[...]/v5.4.1/esp-idf/components/soc/esp32s3/register -I[...]/v5.4.1/esp-idf/components/hal/platform_port/include -I[...]/v5.4.1/esp-idf/components/hal/esp32s3/include -I[...]/v5.4.1/esp-idf/components/hal/include -I[...]/v5.4.1/esp-idf/components/esp_rom/include -I[...]/v5.4.1/esp-idf/components/esp_rom/esp32s3/include -I[...]/v5.4.1/esp-idf/components/esp_rom/esp32s3/include/esp32s3 -I[...]/v5.4.1/esp-idf/components/esp_rom/esp32s3 -I[...]/v5.4.1/esp-idf/components/esp_common/include -I[...]/v5.4.1/esp-idf/components/esp_system/include -I[...]/v5.4.1/esp-idf/components/esp_system/port/soc -I[...]/v5.4.1/esp-idf/components/esp_system/port/include/private -I[...]/v5.4.1/esp-idf/components/lwip/include -I[...]/v5.4.1/esp-idf/components/lwip/include/apps -I[...]/v5.4.1/esp-idf/components/lwip/include/apps/sntp -I[...]/v5.4.1/esp-idf/components/lwip/lwip/src/include -I[...]/v5.4.1/esp-idf/components/lwip/port/include -I[...]/v5.4.1/esp-idf/components/lwip/port/freertos/include -I[...]/v5.4.1/esp-idf/components/lwip/port/esp32xx/include -I[...]/v5.4.1/esp-idf/components/lwip/port/esp32xx/include/arch -I[...]/v5.4.1/esp-idf/components/lwip/port/esp32xx/include/sys -I[...]/v5.4.1/esp-idf/components/esp_driver_gpio/include -I[...]/v5.4.1/esp-idf/components/esp_pm/include -I[...]/v5.4.1/esp-idf/components/mbedtls/port/include -I[...]/v5.4.1/esp-idf/components/mbedtls/mbedtls/include -I[...]/v5.4.1/esp-idf/components/mbedtls/mbedtls/library -I[...]/v5.4.1/esp-idf/components/mbedtls/esp_crt_bundle/include -I[...]/v5.4.1/esp-idf/components/mbedtls/mbedtls/3rdparty/everest/include -I[...]/v5.4.1/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m -I[...]/v5.4.1/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m/p256-m -I[...]/v5.4.1/esp-idf/components/esp_app_format/include -I[...]/v5.4.1/esp-idf/components/esp_bootloader_format/include -I[...]/v5.4.1/esp-idf/components/app_update/include -I[...]/v5.4.1/esp-idf/components/bootloader_support/include -I[...]/v5.4.1/esp-idf/components/bootloader_support/bootloader_flash/include -I[...]/v5.4.1/esp-idf/components/esp_partition/include -I[...]/v5.4.1/esp-idf/components/efuse/include -I[...]/v5.4.1/esp-idf/components/efuse/esp32s3/include -I[...]/v5.4.1/esp-idf/components/esp_mm/include -I[...]/v5.4.1/esp-idf/components/spi_flash/include -I[...]/v5.4.1/esp-idf/components/esp_security/include -I[...]/v5.4.1/esp-idf/components/pthread/include -I[...]/v5.4.1/esp-idf/components/esp_timer/include -I[...]/v5.4.1/esp-idf/components/e [... cut ...]
      |                                     ^~~~~~~~~~~~~~~~~~
[4/7] cd [...]/USBAudio/build && /usr/bin/cmake -E make_directory [...]/USBAudio/build/CMakeFiles && /usr/bin/cmake -E touch [...]/USBAudio/build/CMakeFiles/bootloader-complete && /usr/bin/cmake -E touch [...]/USBAudio/build/bootloader-prefix/src/bootloader-stamp/bootloader-done
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the [...]/USBAudio/build/log/idf_py_stderr_output_2352812 and [...]/USBAudio/build/log/idf_py_stdout_output_2352812

closed 2025-05-09 by Carlos1810044