espressif/esp-idf #15919: why dont ethernet over spi mention spi-port-3 in menuconfig, only 0,1,2

savenlid, 2025-05-07:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have latest master installed.
Using ESP32-S3

In various examples spi3 is not an option to select in menuconfig for SPI based ethernet.
Why is that ?

savenlid, 2025-05-07:
> [@savenlid](https://github.com/savenlid) I think it is cause IDX count from 0 0, 1, 2 = 3x 3x SPI ( SPI0, SPI1, SPI2)
> 
> uint8_t spi_port_nr[3]; spi_init(spi_port_nr[2]); // SPI port 3
> 
> does it help?

Index count, 0,1,2,3 versus 1,2,3,4 ?
It states 0,1,2 as option, but you say it translates into SPI1, GP-SPI2, GP-SPI3 ?

I forced it to use 3 by editing the range in the konfig file.
It compiled it without problem.
But....

E (312) spi: spi_bus_initialize(808): invalid host_id
E (312) example_eth_init: spi_bus_init(159): SPI host #3 init failed
E (312) example_eth_init: example_eth_init(267): SPI bus init failed

I am more leaning towards the GP-SPI3 lacks som DMA capabilities, and therefore its not used.
It the DMA capability is important or not I dont know.

from the esp32-s3 TRM
30.9 Differences Between GP-SPI2 and GP-SPI3
DMA-controlled configurable segmented transfer is only supported in GP-SPI2. Therefore, CONF state is
not used in GP-SPI3.

ps
GP = General Purpose
SPIO = FLASH ( not usable )
SPI1 = PSRAM ( not usable for us who have PSRAM atleast )
ds

wanckl, 2025-05-08:
@savenlid From SPI side, can refer to https://github.com/espressif/esp-idf/blob/aaebc374676621980878789c49d239232ea714c5/components/hal/include/hal/spi_types.h#L28
It shows SPI1/2/3 in index 0/1/2, The pysical SPI0 is invisible, it is what TRM says S3 has 4 SPI.

Actually SPI1(index 0) also not suggest using in application, when you set index 2 in Kconfig, should pysically GPSPI3.

@kostaond Dose it correct?

kostaond, 2025-05-08:
The numbering aligns with [`spi_host_device_t`](https://github.com/espressif/esp-idf/blob/aaebc374676621980878789c49d239232ea714c5/components/hal/include/hal/spi_types.h#L28) as mentioned by @wanckl. I'll consider updating the menuconfig to rather align with documentation. Thanks for pointing it out.