espressif/esp-idf #15920: Console prompt re-writes to the screen with every keypress

mbratch, 2025-05-07:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using the `console` component on UART0.
My prompt is set to ">" so when prompted for a commend I see: "> ".

When I start typing, say, "help", with each keypress the entire prompt starting from beginning of the line is rewritten. The command line works ok, but the rewriting of the line is very visible and distracting.

So type 'h', then the component rewrites, "> h".
Type 'e', then the component rewrites, "> he".
Etc...

Is this expected behavior or perhaps do I have something not setup correctlyi>

songruo, 2025-05-08:
I don't see the problem you described on our [console/basic](https://github.com/espressif/esp-idf/tree/master/examples/system/console/basic) example. Is your usage somehow different from the example?

mbratch, 2025-05-08:
Thank you @songruo . I need to try the sample project. In my project I just have the one UART (0) which is the main USB port for logging, etc. on my ESP32 dev kit. Maybe that affects the behavior.

I'll comment back after trying the basic sample on my platform.

igrr, 2025-05-08:
Just one additional question, what is your OS and the terminal application?

igrr, 2025-05-08:
This could be related to the fact that when the completion callback is set, we always refresh the entire line, even when the character was appended at the end

https://github.com/espressif/esp-idf/blob/aaebc374676621980878789c49d239232ea714c5/components/console/linenoise/linenoise.c#L706

mbratch, 2025-05-08:
@igrr thanks for the reply.
You asked a great question: what terminal application? I apologize I wasn't even thinking about this.

What I have been using is the terminal function inside VS Code. I just tried an external application (putty in this case) and it does not exhibit this behavior.

closed 2025-05-08 by mbratch