espressif/esp-idf #15921: Wake stub in ESP32S3 taking 10ms extra to boot

DarkZeros, 2025-05-07:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have a working code on ESP32-PICO-D4 with a wake up stub that access the SPI for updating an e-ink display, then goes back to sleep. Then wakes up again on EXT0 when display is not BUSY, puts the display to sleep, and goes back to deep sleep on a timer.

This sketch works on v5.4 on ESP-D4.

When porting to ESP32-S3 the system works as expected, SPI transfer, deep sleep, etc..
But I notice the power usage has gone UP from 100nAh/update to 280nAh/update.
When measuring this with a power meter (I can´t share the logs, or pictures sry...), I can see an extra 11ms in both deep sleep wake ups of the chip, using ~20mA. This is around 20mA*22ms = ~150nAh

I have enabled the "skip image checks", and as far as I know FLASH is off during the wake up.
The first thing the CPU should do is execute my code from RTC, but it seems is doing something else first for 10ms.

Can you guide me? I would really like to use ESP32-S3 instead of PICOD4, but this is another blocking issue for me.

Some information on my setup:
* Skip all image checks from deep sleep
* External Clock source, no calibration
* Automatic PM light sleep (on the normal part, non deep sleep stub)
* FLASH Octa mode (tried Quad mode and is same)

DarkZeros, 2025-05-08:
Added real data image.
[image]

esp-wzh, 2025-05-08:
Hi, @DarkZeros  I guess the extra time is printing the log in ROM.

You can call [`esp_deep_sleep_disable_rom_logging`](https://docs.espressif.com/projects/esp-idf/en/v5.4.1/esp32s3/api-reference/system/sleep_modes.html#_CPPv434esp_deep_sleep_disable_rom_loggingv) before sleep to disable the ROM boot printing, or permanently disable it via [menuconfig](https://docs.espressif.com/projects/esp-idf/en/v5.4.1/esp32s3/api-reference/kconfig.html#config-boot-rom-log-scheme).

I used the IDF [deepsleep wakestub example](https://github.com/espressif/esp-idf/tree/master/examples/system/deep_sleep_wake_stub), turned off ROM printing, and tested that it took 2.7ms from chip wakeup to executing the code in the stub.

[image]

DarkZeros, 2025-05-08:
@esp-wzh Thanks! Adding [esp_deep_sleep_disable_rom_logging](https://docs.espressif.com/projects/esp-idf/en/v5.4.1/esp32s3/api-reference/system/sleep_modes.html#_CPPv434esp_deep_sleep_disable_rom_loggingv) just before deep sleep fixed the issue!

However, I already had enabled the setting "CONFIG_BOOT_ROM_LOG_ALWAYS_OFF" and I could not see any log in the monitor. I guess the setting is broken? Or by enabling it, it removes the logs but it actually still spends time printing them?

DarkZeros, 2025-05-08:
Actually the issue is not fully solved. If i zoom in the boot of the esp32 (before the display responds to the command via spi and spikes the power). There is still an extra 3.6ms of time.
Both chips have a characteristic 2.8ms low current when waking. But the s3 has still an extra part.
[image]

esp-wzh, 2025-05-09:
ROM log will also print through USB_SERIAL_JTAG, and `CONFIG_BOOT_ROM_LOG_ALWAYS_OFF` will only disable UART printing, while `esp_deep_sleep_disable_rom_logging` controls both. You need to manually burn efuse to disable USB_SERIAL_JTAG printing.

`espefuse.py --chip esp32s3 -p /dev/ttyUSB0  burn_efuse DIS_USB_SERIAL_JTAG_ROM_PRINT 1`

esp-wzh, 2025-05-09:
Due to the different implementation of ROM boot code, the time taken by esp32s3 from chip wake-up to executing wake stub code increased to 2.75ms, while the time taken by esp32 was 1.33ms.

[image]

DarkZeros, 2025-05-10:
Thanks @esp-wzh ! I guess this clarifies the differences, my personal results are very similar to your results, roughly there is a 2x time increase in boot time and a 3x power usage (because of time spend in boot + extra average current during boot).

I will probably stay on the ESP32 D4 then, since my application spends around 20% of the total power usage in these wakeups + SPI updates.

Indastri, 2025-05-26:
Hi @DarkZeros, closing if you will have any further issues/questions feel free to reopen this or create a new issue.

Cheers!

closed 2025-05-26 by Indastri