espressif/esp-idf #15922: WSS client frequent disconnections

zekageri, 2025-05-08:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3.2.250210

### Espressif SoC revision.

v0.3

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-Wrover-E

### Power Supply used.

External 3.3V

### What is the expected behavior?

Consistent reliable ws connection

### What is the actual behavior?

Websocket frequently disconnects for now reason.

- Have plenty of ram
- Have plenty of stack
- Have plenty of time

### Steps to reproduce.

1. Connect to an external server as a client with wss
2. Have it sit there with only ping-pong
3. Watch as it disconnects after some time ( like a hour or two )

### Debug Logs.

```plain
E (2600790) transport_ws: Error read data
E (2600791) websocket_client: esp_transport_read() failed with -1, transport_error=ESP_ERR_ESP_TLS_TCP_CLOSED_FIN, tls_error_code=0, tls_flags=0, errno=119
E (2600798) Szerver híd: Got WS Error
E (2600803) websocket_client: Error receive data
I (2600810) websocket_client: Reconnect after 5000 ms

E (2600940) transport_ws: Error transport_poll_write
E (2600940) websocket_client: esp_transport_write() returned 0, transport_error=ESP_OK, tls_error_code=0, tls_flags=0, errno=0
E (2600946) Szerver híd: Got WS Error
I (2600951) websocket_client: Reconnect after 5000 ms
```

### Diagnostic report archive.

Disconnect -> Connect repeats over and over again.

### More Information.

Node server logs: 

```
WS_Controller WebSocket closed for device: FC6EF851FA29F9 Code: 1006 Reason:
```

nopnop2002, 2025-05-09:
In my environment, this sample works fine.

https://github.com/espressif/esp-protocols/tree/master/components/esp_websocket_client/examples/target

zekageri, 2025-05-21:
@nopnop2002 Can you suggest some debugging approaches? How can I determine what is causing the disconnections?
The esp32 is properly and reliable connected for hours after boot up, but it starts to disconnect more and more frequently as time goes on. Ram is totally ok, both internal and external. The continuous block is also plenty for both type of ram when these disconnections happen.

On the server side, I only see _Abnormal Closure_ (code 1006) and on the esp32 side it's just `0x65` which is coming from `esp_websocket_event_data_t`

This is my error handler function (sorry for the Hungarian sentences)

```cpp
void ServerBridge::logErrorIfNonzero(const char *message, int error_code){
    if (error_code == 0) { return; }
    ESP_LOGE(SERVER_BRIDGE_DEBUG_TAG, "Last error %s: 0x%x", message, error_code);
    notification.add(
        SERVER_BRIDGE_DEBUG_TAG,
        NotificationType::NOTY_TYPE_ERROR,
        "%s Hibakód: 0x%x", message, error_code
    );
}

void ServerBridge::logLastErrors(esp_websocket_event_data_t *data){
    logErrorIfNonzero(
        "A szerver elutasította a kapcsolódási kérést. Lehet, hogy nem elérhető, vagy a kapcsolat megszakadt.",
        data->error_handle.esp_ws_handshake_status_code
    );
    if (data->error_handle.error_type == WEBSOCKET_ERROR_TYPE_TCP_TRANSPORT) {
        logErrorIfNonzero(
            "Biztonságos kapcsolat (TLS) létesítése sikertelen. Kérjük, ellenőrizze az internetkapcsolatot.",
            data->error_handle.esp_tls_last_esp_err
        );
        logErrorIfNonzero(
            "A titkosított kapcsolat során váratlan hiba történt. A kapcsolat bontva.",
            data->error_handle.esp_tls_stack_err
        );
        logErrorIfNonzero(
            "Hálózati probléma lépett fel. Ellenőrizze az internetkapcsolatot.",
            data->error_handle.esp_transport_sock_errno
        );
    }
}
```

`logLastErrors` is called from a few places. 

- WEBSOCKET_EVENT_DISCONNECTED
- WEBSOCKET_EVENT_ERROR

```cpp
void ServerBridge::evtHandler(int32_t evtId, void *evtData) {
    esp_websocket_event_data_t *data = (esp_websocket_event_data_t *)evtData;
    if (evtId == WEBSOCKET_EVENT_CONNECTED) {
        handleConnect();
    } else if (evtId == WEBSOCKET_EVENT_DISCONNECTED) {
        handleDisconnect(data);
    } else if (evtId == WEBSOCKET_EVENT_DATA) {
        handleWebSocketData(data);
    } else if (evtId == WEBSOCKET_EVENT_CLOSED) {
        handleClose(data);
    } else if (evtId == WEBSOCKET_EVENT_ERROR) {
        handleError(data);
    } else if (evtId == WEBSOCKET_EVENT_FINISH) {
        handleFinish(data);
    } else {
        ESP_LOGW(SERVER_BRIDGE_DEBUG_TAG, "Unhandled ws event: %d", evtId);
    }
}
```

Also, there is a `WS_CLOSE_FRAME` checking in my code when I receive data frame.

`handleWebSocketData` is called from `WEBSOCKET_EVENT_DATA`

```cpp
void ServerBridge::handleWebSocketData(esp_websocket_event_data_t *data) {
    if (data->op_code == WS_TEXT_FRAME) {
        handleRequest(data);
    } else if (data->op_code == WS_CLOSE_FRAME) {
        handleCloseMessage(data);
    } else if (data->op_code == WS_PING_FRAME) {
        handlePingFrame(data);
    } else if (data->op_code == WS_PONG_FRAME) {
        handlePongFrame(data);
    } else {
        ESP_LOGW(SERVER_BRIDGE_DEBUG_TAG, "Got unknown frame: 0x%x", data->op_code);
    }
}
```

Since my esp32 is configured to use as much `PSRAM` as possible, and I have plenty of ram, I'm totally clueless where to search the source of the problem.

- Can task time affect the connection (does the WS task not get enough time in my system?)
- Can UART communication affect the WSS connection? (I have a continuous UART communication going on the other core)
- Can FLASH access affect it?

Thank you for the help!

nopnop2002, 2025-05-21:
The most likely cause is a memory leak.
Check for memory leaks by calling the following periodically:

```
void traceHeap() {
	static uint32_t _free_heap_size = 0;
	if (_free_heap_size == 0) _free_heap_size = esp_get_free_heap_size();

	int _diff_free_heap_size = _free_heap_size - esp_get_free_heap_size();
	ESP_LOGI(__FUNCTION__, "_diff_free_heap_size=%d", _diff_free_heap_size);

	printf("esp_get_free_heap_size() : %6"PRIu32"\n", esp_get_free_heap_size() );
#if 0
	printf("esp_get_minimum_free_heap_size() : %6"PRIu32"\n", esp_get_minimum_free_heap_size() );
	printf("xPortGetFreeHeapSize() : %6zd\n", xPortGetFreeHeapSize() );
	printf("xPortGetMinimumEverFreeHeapSize() : %6zd\n", xPortGetMinimumEverFreeHeapSize() );
	printf("heap_caps_get_free_size(MALLOC_CAP_32BIT) : %6d\n", heap_caps_get_free_size(MALLOC_CAP_32BIT) );
	// that is the amount of stack that remained unused when the task stack was at its greatest (deepest) value.
	printf("uxTaskGetStackHighWaterMark() : %6d\n", uxTaskGetStackHighWaterMark(NULL));
#endif
}
```

zekageri, 2025-05-21:
> The most likely cause is a memory leak.

- If that was the case, I would see increasing ram usage over time.
- I have a memory monitor task which logs to my server.
- I do not see any memory leak. The system is going on for 14 days straight, despite the disconnections and reconnections.
- The server is up 0-24 (it is a paid instance) and it has an industry proven WSS server implementation.

This is the memory usage of one of my systems.
This graph is based on the last 1000 data.
System information is sent every 30 sec.
There are definitely peaks, but it always recovers.

**External used**

- Average: 714.20 KB
- Maximum: 884.42 KB
- Minimum: 688.05 KB
- Current: 688.05 KB

**Internal used**

- Average: 242.98 KB
- Maximum: 244.58 KB
- Minimum: 242.69 KB
- Current: 242.96 KB

[image]

nopnop2002, 2025-05-21:
There doesn't seem to be a memory leak.
Are you fixing the web socket communication task to core 1?

zekageri, 2025-05-21:
My `espressif_esp_websocket_client` library is `1.4.0` and there is no way to alter the core it runs on. At least, I can't find a way to adjust it other than modify the lib.

In pseudocode

```cpp
#define DEFAULT_SERVER_BRIDGE_SKIP_CERT_COMMON_NAME_CHECK false
#define DEFAULT_SERVER_BRIDGE_TRANSPORT WEBSOCKET_TRANSPORT_OVER_SSL
#define DEFAULT_SERVER_BRIDGE_PING_INTERVAL_SEC 30
#define DEFAULT_SERVER_BRIDGE_NETWORK_TIMEOUT_MS 50000
#define DEFAULT_SERVER_BRIDGE_RECONNECT_TIMEOUT_MS 5000
#define DEFAULT_SERVER_BRIDGE_BUFFER_SIZE 10000
#define DEFAULT_SERVER_BRIDGE_TASK_PRIO 10
#define DEFAULT_KEEP_ALIVE_ENABLE true

void ini(){
  client = esp_websocket_client_init(&config);
  esp_err_t startErr = esp_websocket_client_start(client);
}
```

The `esp_websocket_client_config_t` does not allow me to set task core, just stack and priority.

zekageri, 2025-05-21:
I can try to modify `esp_websocket_client_start` so it creates a task pinned to a core.

```cpp
esp_err_t esp_websocket_client_start(esp_websocket_client_handle_t client)
{
    if (client == NULL) {
        return ESP_ERR_INVALID_ARG;
    }
    if (client->state >= WEBSOCKET_STATE_INIT) {
        ESP_LOGE(TAG, "The client has started");
        return ESP_FAIL;
    }

    client->transport = client->config->ext_transport;
    if (!client->transport) {
        if (esp_websocket_client_create_transport(client) != ESP_OK) {
            ESP_LOGE(TAG, "Failed to create websocket transport");
            return ESP_FAIL;
        }
    }

    if (xTaskCreate(esp_websocket_client_task, client->config->task_name ? client->config->task_name : "websocket_task",
                    client->config->task_stack, client, client->config->task_prio, &client->task_handle) != pdTRUE) {
        ESP_LOGE(TAG, "Error create websocket task");
        return ESP_FAIL;
    }
    xEventGroupClearBits(client->status_bits, STOPPED_BIT | CLOSE_FRAME_SENT_BIT);
    ESP_LOGI(TAG, "Started");
    return ESP_OK;
}
```

nopnop2002, 2025-05-21:
I am launching a task just for the WS client like this:

https://github.com/nopnop2002/esp-idf-mirf/blob/master/ws/main/main.c#L356

zekageri, 2025-05-21:
Oh, so you are initializing the WS client in a separate task which is pinned to a core. 
Will the handler task run on the core, it is initialized? So, if I initialize it on core 0 the `esp_websocket_client_task` will run on core 0 as well or is it undeterminable?

nopnop2002, 2025-05-21:
>Oh, so you are initializing the WS client in a separate task which is pinned to a core.

No.
xTaskCreate is a task launch that does not pin a core.

xTaskCreatePinnedToCore is a task launch that does pin a core.

zekageri, 2025-05-21:
> There is no point in pinning an application task to core 0.

I was forced to pin my flash task to core 0 and my UART task to core 1 because of interrupt disabling.
Flash access disables all interrupts on it's core, so UART could not receive any data while flash is written or read.
The same goes for sender and receiver. These tasks are using queues in PSRAM and since PSRAM also uses the same SPI as flash I had to pin these to the core 0 as well.

nopnop2002, 2025-05-21:
I think the UART task is harmless.

Would it be possible to disable the flash task (not start it) or replace it with STAB and see what happens?

zekageri, 2025-05-21:
> Since we have the source for esp_websocket_client_task, why not put the analysis code here?

Will do `esp_log_level_set("websocket_client", ESP_LOG_DEBUG);` so I can see more.

> Would it be possible to disable the flash task (not start it) or replace it with STAB and see what happens?

The interesting thing is that flash operation is really low. Flash is almost never written, just in special cases which are not at all that frequent as the WS disconnections. I can disable flash write / read completely, just to check it.

On boot up it is necessary because all the config is in flash. Even the WS configuration is in flash.
But I can disable further reading / writing after the system is up.

nopnop2002, 2025-05-21:
>Will do esp_log_level_set("websocket_client", ESP_LOG_DEBUG); so I can see more.

Good idea!!

zekageri, 2025-05-21:
> I think the UART task is harmless.

Even with queues and ask in PSRAM?
Almost all of my queue is in PSRAM and I try to put every other task stack into PSRAM as well.

**Example of WS queues**

```cpp
highPriorityQueue = xQueueCreateWithCaps(20, sizeof(WsMessage), MALLOC_CAP_SPIRAM);
lowPriorityQueue = xQueueCreateWithCaps(128, sizeof(WsMessage), MALLOC_CAP_SPIRAM);
requestQueue = xQueueCreateWithCaps(10, sizeof(std::string), MALLOC_CAP_SPIRAM);
```

nopnop2002, 2025-05-21:
>Even with queues and task in PSRAM?

I have never used it this way.

zekageri, 2025-05-21:
These are relatively simple tasks, just waiting on a queue.

```cpp
worker.offload(
  std::bind(&ServerBridge::bridgeReceiver_task, this),
  6048, 10, "BridgeReceiver", nullptr, 0
);
worker.offloadToPsRam(
  std::bind(&ServerBridge::bridgeSender_task, this),
  3848, 10, "BridgeSender", nullptr, 0
);
```

```cpp
void ServerBridge::bridgeReceiver_task() {
    char *rawMessage;
    while (1) {
        if (xQueueReceive(requestQueue, &rawMessage, portMAX_DELAY) == pdTRUE) {
            parseRequest(rawMessage);
            free(rawMessage);
        }
    }
}
```

nopnop2002, 2025-05-21:
There is something strange about the way xQueueReceive is being used.

```
    char *rawMessage;
    while (1) {
        if (xQueueReceive(requestQueue, &rawMessage, portMAX_DELAY) == pdTRUE) {
            parseRequest(rawMessage);
            free(rawMessage);
        }
    }
```

I always do it this way.
```
    char rawMessage[1024];
    while (1) {
        if (xQueueReceive(requestQueue, &rawMessage, portMAX_DELAY) == pdTRUE) {
            parseRequest(rawMessage);
            //free(rawMessage);
        }
    }
```

zekageri, 2025-05-22:
I have enabled `debug` logs on `websoket_client` but it made no difference. I still see the same logs.

nopnop2002, 2025-05-22:
These sources of esp_transport_read/esp_transport_ws_send_raw can be found here.   
You should put your debug code here.

```
esp-idf/components/tcp_transport/transport_ws.c:423:int esp_transport_ws_send_raw(esp_transport_handle_t t, ws_transport_opcodes_t opcode, const char *b, int len, int timeout_ms)

esp-idf/components/tcp_transport/transport.c:137:int esp_transport_read(esp_transport_handle_t t, char *buffer, int len, int timeout_ms)
```

nopnop2002, 2025-05-22:
Adding some logging to this part would make it even clearer.

```
        rlen = esp_transport_read(client->transport, client->rx_buffer, client->buffer_size, client->config->network_timeout_ms);
        if (rlen < 0) {
            // Add Logging
            esp_websocket_free_buf(client, false);
            esp_tls_error_handle_t error_handle = esp_transport_get_error_handle(client->transport);
            if (error_handle) {
                esp_websocket_client_error(client, "esp_transport_read() failed with %d, transport_error=%s, tls_error_code=%i, tls_flags=%i, errno=%d",
                                           rlen, esp_err_to_name(error_handle->last_error), error_handle->esp_tls_error_code,
                                           error_handle->esp_tls_flags, errno);
            } else {
                esp_websocket_client_error(client, "esp_transport_read() failed with %d, errno=%d", rlen, errno);
            }
            return ESP_FAIL;
        }

        // send with ws specific way and specific opcode
        wlen = esp_transport_ws_send_raw(client->transport, opcode, (char *)client->tx_buffer, need_write,
                                         (timeout == portMAX_DELAY) ? -1 : timeout * portTICK_PERIOD_MS);
        if (wlen < 0 || (wlen == 0 && need_write != 0)) {
            // Add Logging
            ret = wlen;
            esp_websocket_free_buf(client, true);
            esp_tls_error_handle_t error_handle = esp_transport_get_error_handle(client->transport);
            if (error_handle) {
                esp_websocket_client_error(client, "esp_transport_write() returned %d, transport_error=%s, tls_error_code=%i, tls_flags=%i, errno=%d",
                                           ret, esp_err_to_name(error_handle->last_error), error_handle->esp_tls_error_code,
                                           error_handle->esp_tls_flags, errno);
            } else {
                esp_websocket_client_error(client, "esp_transport_write() returned %d, errno=%d", ret, errno);
            }
            esp_websocket_client_abort_connection(client, WEBSOCKET_ERROR_TYPE_TCP_TRANSPORT);
            goto unlock_and_return;
        }
```

zekageri, 2025-05-22:
Maybe it is some menu config option misconfigured in the TCP/IP layer or somewhere deep?

nopnop2002, 2025-05-22:
This is probably a timing specific issue.

It's very difficult to pinpoint and troubleshoot a timing specific issue.

nopnop2002, 2025-05-22:
>client = esp_websocket_client_init(&config);

What parameters are you setting in the config?

zekageri, 2025-05-22:
Ohhh, yes, I have tasks which have bigger priority. Will lower them and test it!

zekageri, 2025-05-22:
Well, I lowered all of my tasks below 18 and still got an error almost immediately after boot up with the new changes.

```
E (98174) transport_ws: Error transport_poll_write
E (98174) websocket_client: esp_transport_write() returned 0, transport_error=ESP_OK, tls_error_code=0, tls_flags=0, errno=0
E (98180) Szerver híd: Got WS Error
I (98187) websocket_client: Reconnect after 5000 ms
W (98190) Szerver híd: Disconnected
W (103196) Szerver híd: Unhandled ws event: 5
I (103905) Szerver híd: Connected
```

zekageri, 2025-05-22:
No, it did not help.

| Name | Priority|
| -------- | ------- |
| ipc0  | 24    |
| ipc1 | 24     |
| esp_timer | 22    |
| sys_evt | 20 |
| arduino_events |  19 |
| tiT | 18 |

nopnop2002, 2025-05-22:
arduino_events

What is this??

I never see.

zekageri, 2025-05-22:
The app is "Arduino as a component of IDF".
Latest _IDF 5.3.2.250210_ and latest Arduino.

nopnop2002, 2025-05-22:
Have you disabled loop()?

```
void setup() {
  Serial.begin(115200);
  xTaskCreate(task0, "Task0", 4096, NULL, 1, NULL);
  xTaskCreate(task1, "Task1", 4096, NULL, 1, NULL);
  vTaskDelete( NULL );
}

void loop() {
}
```

zekageri, 2025-05-23:
- I have deleted the loop task ( vTaskDelete(null) in loop )
- I have disabled everything I could from arduino in the menuconfig
- I am using vTaskDelays to switch tasks

nopnop2002, 2025-05-23:
With this fix, the WebSocket Client should work stably, but if it does not, you will need to add further debug code here to investigate the cause.
```
int esp_transport_read(esp_transport_handle_t t, char *buffer, int len, int timeout_ms)
{
    if (t && t->_read) {
        return t->_read(t, buffer, len, timeout_ms);
    }
    return -1;
}

int esp_transport_ws_send_raw(esp_transport_handle_t t, ws_transport_opcodes_t opcode, const char *b, int len, int timeout_ms)
{
    uint8_t op_code = ws_get_bin_opcode(opcode);
    if (t == NULL) {
        ESP_LOGE(TAG, "Transport must be a valid ws handle");
        return ESP_ERR_INVALID_ARG;
    }
    ESP_LOGD(TAG, "Sending raw ws message with opcode %d", op_code);
    return _ws_write(t, op_code, WS_MASK, b, len, timeout_ms);
}
```

nopnop2002, 2025-05-26:
>I have reduced ping interval to 5 sec

This will achieve the same effect.

```
esp_websocket_client_config_t config = {};
config.ping_interval_sec = 5;
client = esp_websocket_client_init(&config);
```