espressif/esp-idf #15924: PBAC: Out of Service Records (3)

boborjan2, 2025-05-08:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

master

### Espressif SoC revision.

ESP32 v3.1

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

HFP_HF example shall download phonebook.

### What is the actual behavior?

I get the following (tried 2 phones, both androids though but different vendors and android versions):
```
I (960625) BT_HF: --connection state slc_connected, peer feats 0xfef, chld_feats 0x3f
I (960625) BT_HF: APP HFP event: INBAND_RING_TONE_EVT
I (960625) BT_HF: --inband ring state Provided
W (960685) BT_BTM: BTM_SEC_REG: Out of Service Records (3)

W (960685) BT_BTM: btm_sec_l2cap_access_req() PSM: 4100 no application registered

E (960695) OBEX_TL_L2CAP: l2cap_connect_cfm but no ccb found

W (967485) BT_HCI: hcif mode change: hdl 0x80, mode 2, intv 768, status 0x0
I (967485) BT_HF: ESP_BT_GAP_MODE_CHG_EVT mode:2
```

### Steps to reproduce.

I just compile and run stock hfp_hf example, trying to make pbac work.

### Debug Logs.

```plain

```

xiongweichao, 2025-05-12:
Hi @boborjan2 I'm sorry that I couldn't reproduce this issue. May I ask if the version you are using includes this [commit](https://github.com/espressif/esp-idf/commit/b2a5b967c91882721547523f7052c9227e0235ee)?

boborjan2, 2025-05-12:
Thanks for the feedback.
It does include that commit, yes. However I will remake the test with today's master HEAD.

xiongweichao, 2025-05-12:
If the problem still exists please let me know and provide the sdkconfig and commit id you used. Thanks！

boborjan2, 2025-06-24:
I confirm the issue is gone with todays's release/v5.5