espressif/esp-idf #15925: I2C transaction failed with ESP_IDF v5.4.0 using SSD1306 oled screen

electronicsguy, 2025-05-08:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.4.0

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Waveshare ESP32S3-0.91-amoled

### Power Supply used.

USB

### What is the expected behavior?

OLED screen connected over the i2c bus should work

### What is the actual behavior?

the i2c device cannot be initialized

### Steps to reproduce.

1. Clone this repo: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/i2c_oled
2. Setup esp_idf toolchain, vscode, esp_idf extension and open the above project
3. set the SDA & SCL pins correctly for the wiring used, within the file: i2c_oled_example_main.c
4. build, flash, monitor
...

### Debug Logs.

```plain
I (266) main_task: Started on CPU0
I (276) main_task: Calling app_main()
I (276) example: Initialize I2C bus
I (276) example: Install panel IO
I (276) example: Install SSD1306 panel driver
E (286) i2c.master: I2C hardware NACK detected
E (286) i2c.master: I2C transaction unexpected nack detected
E (296) i2c.master: s_i2c_synchronous_transaction(918): I2C transaction failed
E (296) i2c.master: i2c_master_multi_buffer_transmit(1180): I2C transaction failed
E (306) lcd_panel.io.i2c: panel_io_i2c_tx_buffer(193): i2c transaction failed
E (316) lcd_panel.ssd1306: panel_ssd1306_init(151): io tx param SSD1306_CMD_SET_MULTIPLEX failed
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x42009e4e
--- 0x42009e4e: app_main at /home/electronicsguy/programming/esp/esp32s3/esp_sdk/i2c_oled/main/i2c_oled_example_main.c:102 (discriminator 1)

file: "./main/i2c_oled_example_main.c" line 102
func: app_main
expression: esp_lcd_panel_init(panel_handle)

abort() was called at PC 0x4037a32b on core 0
--- 0x4037a32b: _esp_error_check_failed at /home/electronicsguy/programming/esp/esp-idf/components/esp_system/esp_err.c:49

Backtrace: 0x40375ab1:0x3fca1c30 0x4037a335:0x3fca1c50 0x40380a8d:0x3fca1c70 0x4037a32b:0x3fca1ce0 0x42009e4e:0x3fca1d10 0x4203112f:0x3fca1dd0 0x4037ad6d:0x3fca1e00
--- 0x40375ab1: panic_abort at /home/electronicsguy/programming/esp/esp-idf/components/esp_system/panic.c:454
--- 0x4037a335: esp_system_abort at /home/electronicsguy/programming/esp/esp-idf/components/esp_system/port/esp_system_chip.c:92
--- 0x40380a8d: abort at /home/electronicsguy/programming/esp/esp-idf/components/newlib/abort.c:38
--- 0x4037a32b: _esp_error_check_failed at /home/electronicsguy/programming/esp/esp-idf/components/esp_system/esp_err.c:49
--- 0x42009e4e: app_main at /home/electronicsguy/programming/esp/esp32s3/esp_sdk/i2c_oled/main/i2c_oled_example_main.c:102 (discriminator 1)
--- 0x4203112f: main_task at /home/electronicsguy/programming/esp/esp-idf/components/freertos/app_startup.c:208
--- 0x4037ad6d: vPortTaskWrapper at /home/electronicsguy/programming/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

ELF file SHA256: bfb3514d9

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x403759f0
```

wanckl, 2025-05-09:
@electronicsguy The log says `I2C transaction failed`

IIC connection need external pull up resister on SCL/SDA wire, If you are using internal pull up, it is very weak, can you slow down frequency and try again.

Do you mean it works on IDF old version and update 5.4 cause the issue?

electronicsguy, 2025-05-09:
ok, so I investigated further, following@mythbuster5 s comment:  

https://github.com/espressif/esp-idf/issues/15734#issuecomment-2814398980

Indeed, if you just remove the debug checks, you're good to go. It works fine if I change the lines to these:

```
    esp_lcd_panel_reset(panel_handle);
    esp_lcd_panel_init(panel_handle);
    esp_lcd_panel_disp_on_off(panel_handle, true);
```
Output:
```
I (266) main_task: Started on CPU0
I (276) main_task: Calling app_main()
I (276) example: Initialize I2C bus
I (276) example: Install panel IO
I (276) example: Install SSD1306 panel driver
I (386) example: Initialize LVGL
I (386) LVGL: Starting LVGL task
I (386) example: Display LVGL Scroll Text
I (386) main_task: Returned from app_main()
```

This works fine running on my ESP32S3 board. I can see the "Hello espressif, hello lvgl" scrolling line.

But if I connect VSCode monitor to it, it does error out after some time with this message:
```
I (426) example: Initialize LVGL
I (426) LVGL: Starting LVGL task
I (426) example: Display LVGL Scroll Text
I (426) main_task: Returned from app_main()
E (936) i2c.master: I2C hardware NACK detected
E (936) i2c.master: I2C transaction unexpected nack detected
E (936) i2c.master: s_i2c_synchronous_transaction(918): I2C transaction failed
E (936) i2c.master: i2c_master_multi_buffer_transmit(1180): I2C transaction failed
E (946) lcd_panel.io.i2c: panel_io_i2c_tx_buffer(193): i2c transaction failed
E (956) lcd_panel.ssd1306: panel_ssd1306_draw_bitmap(206): io tx color failed
E (5936) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (5936) task_wdt:  - IDLE0 (CPU 0)
E (5936) task_wdt: Tasks currently running:
E (5936) task_wdt: CPU 0: LVGL task
E (5936) task_wdt: CPU 1: IDLE1
E (5936) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x42020046:0x3FC93D10 0x4202045C:0x3FC93D30 0x4037734D:0x3FC93D60 0x42010402:0x3FCA44E0 0x42010D91:0x3FCA4500 0x420153DD:0x3FCA4530 0x42015487:0x3FCA4550 0x4200A5A0:0x3FCA4570 0x4037AD6D:0x3FCA4590
--- 0x42020046: task_wdt_timeout_handling at /home/electronicsguy/programming/esp/esp-idf/components/esp_system/task_wdt/task_wdt.c:434
--- 0x4202045c: task_wdt_isr at /home/electronicsguy/programming/esp/esp-idf/components/esp_system/task_wdt/task_wdt.c:507
--- 0x4037734d: _xt_lowint1 at /home/electronicsguy/programming/esp/esp-idf/components/xtensa/xtensa_vectors.S:1240
--- 0x42010402: draw_buf_flush at /home/electronicsguy/programming/esp/esp32s3/esp_sdk/i2c_oled/managed_components/lvgl__lvgl/src/core/lv_refr.c:1195
--- 0x42010d91: _lv_disp_refr_timer at /home/electronicsguy/programming/esp/esp32s3/esp_sdk/i2c_oled/managed_components/lvgl__lvgl/src/core/lv_refr.c:333
--- 0x420153dd: lv_timer_exec at /home/electronicsguy/programming/esp/esp32s3/esp_sdk/i2c_oled/managed_components/lvgl__lvgl/src/misc/lv_timer.c:313 (discriminator 2)
--- 0x42015487: lv_timer_handler at /home/electronicsguy/programming/esp/esp32s3/esp_sdk/i2c_oled/managed_components/lvgl__lvgl/src/misc/lv_timer.c:109
--- 0x4200a5a0: lvgl_port_task at /home/electronicsguy/programming/esp/esp32s3/esp_sdk/i2c_oled/managed_components/espressif__esp_lvgl_port/esp_lvgl_port.c:691
--- 0x4037ad6d: vPortTaskWrapper at /home/electronicsguy/programming/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

E (10936) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (10936) task_wdt:  - IDLE0 (CPU 0)
E (10936) task_wdt: Tasks currently running:
E (10936) task_wdt: CPU 0: LVGL task
E (10936) task_wdt: CPU 1: IDLE1
E (10936) task_wdt: Print CPU 0 (current core) backtrace
```

AFAIU, the standard example for SSD1306 is wrong. If an initial NACK is expected, why are we checking and aborting for it in the example? 
Could someone tell me which WDT to fire to make the above code reliable in the presence of connecting a monitor?

@wanckl Changing frequency doesn't seem to matter.

JoergSH, 2025-05-15:
use an older version 

change in platformio.ini

platform = espressif32@5.2.0

mythbuster5, 2025-05-16:
@electronicsguy As for idf example, `ESP_CHECK_ERROR` would assert if the return value is wrong. And the initial NACK is not expected.

Secondly, for watch dog, you can see it reports `i2c_ll_is_bus_busy`, so there is a fsm state wrong in master bus. It is usually caused by hardware connection, perhaps noise on bus or connection is very unstable by jump wire. May you please check the hardware connection? Thanks

Ropoperperithras, 2025-05-18:
> [@electronicsguy](https://github.com/electronicsguy) As for idf example, `ESP_CHECK_ERROR` would assert if the return value is wrong. And the initial NACK is not expected.
>
> Secondly, for watch dog, [...]

Had the same issue,  isolated the connections and got it fixed, for me the problem seemed to be the connections in the breadboard, not yet sure of the root of the problem.
version used 5.4.1

mythbuster5, 2025-05-20:
Yeah.. I tend to treat this question as connections, more updates or questions here?

mythbuster5, 2025-05-23:
Close the issue since there is no far more updates. Feel free to reopen if you have more clue. Thanks

closed 2025-05-23 by mythbuster5