espressif/esp-idf #15926: Guidance Needed: Implementing OTA Firmware Updates Over Thread Mesh Network Using ESP-IDF

kunj2308, 2025-05-09:
### Is your feature request related to a problem?

I'm developing an OTA update solution for Thread mesh networks using Espressif hardware and would appreciate expert advice. My current setup consists of:

Border Router Configuration:

ESP32 (Wi-Fi) running the [ot_br example](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_br)

Connected via UART to ESP32-C6 running the [ot_rcp example](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp)

Thread End Device:

ESP32-C6 running the [ot_cli example](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_cli)

Implementation Goal:

Download firmware binary on Border Router (via Wi-Fi/Internet)

Transfer firmware securely over Thread mesh network to end device

Implement reliable OTA update process using [ESP-IDF's OTA API](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/ota.html)

Technical Questions:

What modifications are needed to the OpenThread examples to support:

Firmware transfer over Thread

Chunking strategy for Thread's 127-byte MTU

Integrity verification (SHA256) and secure reboot

Are there known limitations with:

Thread network reliability for large binary transfers

ESP32-C6 flash write performance during OTA

Border router throughput constraints

Would alternative approaches be more suitable? For example:

Using CoAP for firmware transfer

Implementing a custom fragmentation protocol

Reference Materials:

[ESP-IDF OpenThread Guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-guides/openthread.html)

[Thread Network Specifications](https://www.threadgroup.org/ThreadSpec)

[OpenThread MTU Discussion](https://github.com/openthread/openthread/issues/7125)

Additional Request:
If anyone has implemented similar OTA-over-Thread solutions, I would greatly appreciate any code examples or design insights you could share.

tanyanquan, 2025-05-13:
Hi @kunj2308 thank you for your question. We currently don't have an example specifically designed for OTA updates using Thread on an end device, but the required components (such as the OTA API which you mentioned) are all available. 

This is because OTA updates are typically handled at the application layer, such as Matter-over-Thread or Rainmaker-over-Thread, which will be more convenient instead. Could you let us know a bit more details about your goals below so we can provide a more targeted solution?
- Are you planning to use any application-layer protocols / ecosystems for your solution?

If you wish to try out a simple solution for OTA-updating a Thread device, we already have an [example here for a Thread Border Router](https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/ota_update.html). If you connect another Thread BR (BR2) to the same network without connecting it to Wi-Fi, it will take on the role of end device and you can call the same command to start the OTA process for BR2 as well. Do note that depending on the stability of your Wi-Fi / Thread networks, this could take a long time and you may need to change some settings (e.g. http client timeout). 

We are also working on extending the [OTA API example](https://github.com/espressif/esp-idf/tree/master/examples/system/ota) to support Thread devices as well, which you can try in the near future!

kunj2308, 2025-05-14:
> Hi [@kunj2308](https://github.com/kunj2308) thank you for your question. We currently don't have an example specifically designed for OTA updates using Thread on an end device, but the required [...]

Thank you @tanyanquan for the detailed explanation and guidance!

To clarify my goal:

I am building a custom Thread mesh network using ESP32-C6 devices, where each device operates as an end node (e.g., lighting controller). My intent is to:

Control GPIOs on the end device (e.g., switch lights ON/OFF).

Deploy firmware OTA over the Thread network to these end devices.

Avoid using higher-level ecosystems like Matter or Rainmaker, and instead work directly with OpenThread and the ESP-IDF OTA API.

Key Questions:
Can you share a list of Thread-based lighting end devices (either evaluation boards or commercial products) that:

Support OTA natively over Thread (not via Matter).

Can be integrated or emulated using ESP32-C6 or OpenThread CLI?

In current market implementations (besides Matter), how do companies implement OTA updates over Thread? Are they using CoAP-based fragmentation or some proprietary protocol layered over Thread?

Does Espressif have internal or upcoming OTA-over-Thread examples specifically for ESP32-C6 devices running as Thread end devices (e.g., light nodes)?

I'm working directly at the application layer using OpenThread CLI and ESP-IDF, and would prefer not to integrate Matter at this stage.

Thank you again for your help! If there's any beta/example code for Thread-based OTA end device support, I would be happy to test and contribute feedback.

kunj2308, 2025-05-14:
Hi @tanyanquan ,

Thank you for the detailed clarifications and support.

We are from xyz Automations ,and we want to developing our own custom ecosystem based on a Thread mesh network using ESP32-C6 devices.

Our Goals:
We want to:

Build a minimal, lightweight system without Matter or Rainmaker

Control GPIOs on Thread end devices (e.g., switch lights ON/OFF)

Communicate directly via Thread protocol using OpenThread CLI and UDP messaging

Implement OTA firmware updates over the Thread mesh (without relying on Wi-Fi or external ecosystems)

We are intentionally avoiding Matter to keep the firmware lightweight and to maintain full control over the application layer. We understand this means we need to handle chunking, integrity checks (e.g., SHA256), and reliable transfer logic ourselves — and we are ready to do that.

Questions:
Is it mandatory to use Matter with Thread for real-world applications? Or is it fully acceptable to use Thread independently, with our own protocol for lighting control and OTA?

You mentioned that ESP-IDF has OTA examples like simple_ota_example, which works over Wi-Fi. Could you share a working (or in-development) example for OTA on Thread end devices (ESP32-C6 with ot_cli), where:

The firmware is downloaded on the Thread Border Router

It is sent to an end device over Thread (using chunked UDP or CoAP)

The ESP-IDF OTA API is used to write it to flash on the end device

If a working example doesn’t exist yet, is there a recommended way to extend the existing OpenThread CLI/ESP-IDF to enable this flow? Even an experimental branch would help — we’d be glad to test it and contribute feedback.

Thanks again for your support. A minimal OTA-over-Thread setup would greatly accelerate our development, and we are committed to building on Espressif’s stack for our production rollout.

tanyanquan, 2025-05-14:
Hi @kunj2308 thanks for your clarifications! We are happy to hear that you intend to implement your own application layer using Thread as the networking protocol and that is definitely possible. 

If you wish to use the simple_ota_example, we can start by extending it to include C6 (using Thread, so we'll need to turn off Wi-Fi). By default, this example uses HTTP (or HTTPS), so you can take a look at the set up instructions [for simple_ota example](https://github.com/espressif/esp-idf/tree/master/examples/system/ota).

We've prepared an esp-idf patch here which you can try: [thread_ota.txt](https://github.com/user-attachments/files/20203945/thread_ota.txt)

Please add the following config options to `examples/system/ota/simple_ota_example/sdkconfig.defaults` (remember to change `CONFIG_EXAMPLE_FIRMWARE_UPGRADE_URL`):

```
CONFIG_EXAMPLE_FIRMWARE_UPGRADE_URL="https://192.168.0.123:8070/blink.bin"
CONFIG_EXAMPLE_SKIP_COMMON_NAME_CHECK=y

CONFIG_OPENTHREAD_ENABLED=y
CONFIG_EXAMPLE_CONNECT_WIFI=n
CONFIG_EXAMPLE_CONNECT_ETHERNET=n
CONFIG_EXAMPLE_CONNECT_THREAD=y
CONFIG_MBEDTLS_CMAC_C=y
```

You may also need to change the default Thread network configurations in `idf.py menuconfig`:

```
Component config → OpenThread → OpenThread → Thread Core Features → Thread Operational Dataset
```

We've tried this using a HTTP server on a Linux host machine + Thread BR (both connected to the same Wi-Fi network). Do let us know how it goes 😄

Alvin1Zhang, 2025-06-30:
Thanks for reporting, feel free to reopen.

closed 2025-06-30 by Alvin1Zhang

txinto, 2025-07-15:
Hi, thanks for researching on this.
I've trying to replicate the steps but I arrive to the following situation:

`I (1907) simple_ota_example: Starting OTA example task
I (1917) simple_ota_example: Converted URL: http://[fd94:f364:8b18:2:0:0:c0a8:103]:8000/ota_example.bin
I (1927) simple_ota_example: Attempting to download update from http://[fd94:f364:8b18:2:0:0:c0a8:103]:8000/ota_example.bin
I (1907) main_task: Returned from app_main()
W(4307) OPENTHREAD:[W] DuaManager----: Failed to perform next registration: NotFound
E (6957) esp-tls: [sock=54] select() timeout
E (6957) transport_base: Failed to open a new connection: 32774
E (6957) HTTP_CLIENT: Connection failed, sock < 0
E (6967) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (6967) esp_https_ota: Failed to establish HTTP connection
E (6977) simple_ota_example: Firmware upgrade failed
I(74857) OPENTHREAD:[N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:48, src:8a1b0f65460c0f85, dst:0xffff, sec:yes, tag:1294, offset:128, dglen:176
W(77297) OPENTHREAD:[W] DuaManager----: Failed to perform next registration: NotFound`

I have manually patched the v5.4.1, because I did not find any reference of the codebase to patch (main branch? 5.4.2?).

I am using the ESP-C6-DevKit-M1 and my BR is the basic_border_router from espressif.

I can ping from the thread node to the Linux with the HTTP server, I can not test curl because curl is not active in this example (I'll try to activate it).

Thanks, feedback is welcome.