espressif/esp-idf #15927: ESP32-S2 : ADC unit can not be controlled by the ULP RISCV (ADC_ULP_MODE_RISCV)

DascaluDorian, 2025-05-09:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF Release v5.4.1

### Espressif SoC revision.

ESP32-S2FNR2 (revision v1.0) (ESP32-S2-MINI-2)

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

ULP RISCV ADC unit initialized with _ADC_ULP_MODE_RISCV_ should provide fresh values.

### What is the actual behavior?

ULP RISCV ADC initialized with _ADC_ULP_MODE_RISCV_ can not read any value and always returns 8191

### Steps to reproduce.

ULP RISCV can not control the ADC unit (GPIO9-ADC1_CHN8) when configured with  _ADC_ULP_MODE_RISCV_ option. With this configuration, a value of 8191 is always returned.

We can see fresh and new values if the ULP RISCV ADC unit is configured with _ADC_ULP_MODE_DISABLE_.

Unfortunately, the _ADC_ULP_MODE_DISABLE_ configuration does not work in light sleep (also deep sleep). ULP RISCV configured with this option reads only the last available ADC value, and then it is stale. This is a critical problem for the Power Management project, including automatic light sleep, making the ADC read in ULP impossible.

1. None of the ESP IDF ULP examples are working
 - https://github.com/espressif/esp-idf/blob/master/components/ulp/test_apps/ulp_riscv/main/test_ulp_riscv.c
 - https://github.com/espressif/esp-idf/tree/master/examples/system/ulp/ulp_riscv/adc
2. There are previous issues, not solved but with some workarounds, but they are not solving this problem
- https://github.com/espressif/esp-idf/issues/12426
4. Tried even the ADC Legacy example code
5. Tried different configurations, compiler options, power modes/domains, and reduced the code to a minimum size.

I can confirm that the ULP is started and running correctly, and no errors are reported by the LOG.

### Debug Logs.

```plain
ULP RISCV configured with _ADC_ULP_MODE_RISCV_

![Image](https://github.com/user-attachments/assets/80a97c4a-d94a-4df3-95cd-27b5abf0892e)

ULP RISCV configured with _ADC_ULP_MODE_DISABLE_

![Image](https://github.com/user-attachments/assets/cef2fb28-745f-4511-a0aa-15661b72a371)
```

KonstantinKondrashov, 2025-05-09:
Hi @DascaluDorian! 
Thank you for reporting this issue. I will initially check this issue and come back with a solution.

DascaluDorian, 2025-05-30:
Hi @KonstantinKondrashov 

Wow, that seems to work. Thank you a lot.

I have already tried to take some parts of the master branch related to some power domain options, but I have not taken the entire branch. Currently, I am using version release/v5.4, and everything seems to be working again.

I reverted back to 5.4.1, and it was not working again, so something is broken inside, but glad it is already fixed on the master branch.

Thank you again :)

closed 2025-05-30 by DascaluDorian