espressif/esp-idf #15929: RMT Start receive status

janav1, 2025-05-09:
### Is your feature request related to a problem?

It would be great if the RMT library could provide the state of the rmt peripheral - specifically whether data is already being received. Right now I can just call start and when the data is received, an interrupt will occur. I use the RMT peripheral to receive long data (several seconds) and it would be good to know that the data is already being received. Either by some status bit or by using an interrupt.

### Describe alternatives you've considered.

I am currently using an RMT peripheral to receive the data and an interrupt from the same pin to signal the start of data reception.

Kainarx, 2025-05-12:
Hi @janav1 
After calling `rmt_receive`, reception starts on the first signal edge (level transition).
You can use a **GPIO interrupt**: configure a GPIO interrupt on the same pin used for RMT reception. This way, you can detect when data transmission begins, as it will trigger an interrupt on the first level change.

P.S. I'm curious—what kind of application needs to know the exact moment reception starts? If you're willing to share, I'd love to hear more about it.

Kainarx, 2025-05-13:
> I already use the solution you suggested.    👍

 

Thank you for your feedback. This is indeed a point that deserves improvement. We will consider improvements in the subsequent hardware design.

closed 2025-05-15 by espressif-bot