espressif/esp-idf #15935: Touch FSM measures different values on deep sleep on ESP32

DarkZeros, 2025-05-10:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have a piece of code fully working to set up the Touch Wakeup from deep sleep.
The code works if the threshold values are hardcoded manually, however when trying to read the sensor and setting it up that way, the value is completely different from the deep sleep one.

For example:
 * The values using touchRead() for the 4 pads:
   * 50,45,57,57 
   * When touching them: 30, 25, 37, 35
 * The values read by using touch_ll_read_raw_data() just after deep sleep:
   * 333, 257, 382, 416
   * When touching them: 160, 100, 150, 130

The actual threshold values that wake up the device from deep sleep are approximately 200.

The device uses CONFIG_RTC_CLK_SRC_EXT_OSC=y, my theory is that the FSM timer CLK is different while the chip is on, or is in deep sleep, and because of that it takes a different amount of time, and measures a different value. 

Example code I am using:
```
  touch_pad_init();
  touch_pad_set_voltage(TOUCH_HVOLT_2V4, TOUCH_LVOLT_0V5, TOUCH_HVOLT_ATTEN_1V);
  touch_ll_set_meas_time(mSettings.mCycles[0] * 1024);
  touch_pad_set_measurement_interval((8 * 1024));
  esp_sleep_enable_touchpad_wakeup();
  touch_pad_set_fsm_mode(TOUCH_FSM_MODE_TIMER);

  auto setTouchPad = [&](auto&& v) {
    auto pad = (touch_pad_t)HW::Touch::Pad[mSettings.mMap[v]];
    auto& m = mSettings.mMaxMeasures[mSettings.mMap[v]];

    touch_pad_config(pad, m - 100);
  };

// For all touch pads
setTouchPad(...);

...
deepsleep()
```

DarkZeros, 2025-06-02:
I am 99% sure this issue is due to a change in the reference clock during deep sleep.

I set my code to do:
```
    // ..... wakeup
    LOG(mTouch.readAll());
    delay(3000); // The rate of measurement of the FSM is set to 1s, therefore a new measurement should be ready
    LOG(mTouch.readAll());
    // Set timer for deep sleep for 1s (no touch wakeup)
    deep_sleep();
```
Therefore the device is deep sleeping "1s", Light sleeping "3s" and active around a few milliseconds.

I can see this graph when pluging an oscilloscope to the Touch pad:
[image]
This clearly shows 3 fast readings, and 1 slow reading. The slow reading is exactly 31ms.
That is the value I set in the touch (1024 / 32.768 Khz = 1/32 s = 31.25ms).

[image]
 (And it matches the 100-300ish values, cycles)

But the other readings are much much faster:
[image]

Around 0.125ms, which would match with a clock of (1024 / 8MHz  = 0.125ms).

Looks to me that the touch pad is using 8MHz RTC_FAST_CLK not in deep sleep and 32.768 KHz (external clock reference from my RTC clock) the rest of the time. How could I fix this?

DarkZeros, 2025-06-02:
I found this in the file a line that says the touch uses RTC_SLOW_CLK, this contradicts the ESP32 technical document saying that it uses RTC_FAST_CLK (8MHz):
/components/hal/esp32/include/hal/touch_sensor_ll.h
```
/**
 * Set touch sensor measurement time.
 *
 * @param meas_time The duration of the touch sensor measurement.
 *                  t_meas = meas_time / (8MHz), the maximum measure time is 0xffff / 8M = 8.19 ms.
 */
static inline void touch_ll_set_meas_time(uint16_t meas_time)
{
    //touch sensor measure time= meas_cycle / 8Mhz
    HAL_FORCE_MODIFY_U32_REG_FIELD(SENS.sar_touch_ctrl1, touch_meas_delay, meas_time);
    //the waiting cycles (in 8MHz) between TOUCH_START and TOUCH_XPD
    HAL_FORCE_MODIFY_U32_REG_FIELD(SENS.sar_touch_ctrl1, touch_xpd_wait, TOUCH_LL_PAD_MEASURE_WAIT_MAX);
}

...

/**
 * Set touch sensor sleep time (interval of measurement).
 *
 * @param sleep_time  The touch sensor will sleep after each measurement.
 *                    sleep_cycle decide the interval between each measurement.
 *                    t_sleep = sleep_cycle / (RTC_SLOW_CLK frequency).
 *                    The approximate frequency value of RTC_SLOW_CLK can be obtained using `rtc_clk_slow_freq_get_hz` function.
 */
static inline void touch_ll_set_sleep_time(uint16_t sleep_time)
{
    //touch sensor sleep cycle Time = sleep_cycle / RTC_SLOW_CLK( can be 150k or 32k depending on the options)
    HAL_FORCE_MODIFY_U32_REG_FIELD(SENS.sar_touch_ctrl2, touch_sleep_cycles, sleep_time);
}
```

DarkZeros, 2025-06-05:
I downgraded to v5.1 (I was originally on v5.4) and now everything is working fine, I will try to pinpoint the offending commit.

DarkZeros, 2025-06-05:
@esp-wzh Can i draw your attention to this?

I am working on my own mirror of IDF, so i can cherry pick the revert of the commit, but lest see if we can properly fix it.

esp-wzh, 2025-06-06:
Maybe this [errata](https://docs.espressif.com/projects/esp-chip-errata/en/latest/esp32/03-errata-description/esp32/ulp-and-touch-sensors-cannot-be-used.html#ulp-ulp-coprocessor-and-touch-sensors-can-not-be-used-in-deep-sleep-mode-if-rtc-periph-power-domain-is-powered-up) is related to the problem you are encountering, FYI.

esp-wzh, 2025-06-18:
Can you add print `esp_rom_printf("pd_flags:0x%08x\n",pd_flags);` at [esp_sleep_start](https://github.com/espressif/esp-idf/blob/67c1de1eebe095d554d281952fde63c16ee2dca0/components/esp_hw_support/sleep_modes.c#L778)  to check if RTC_PERIPH is really powered down during sleep? @DarkZeros

DarkZeros, 2025-06-18:
> Can you add print `esp_rom_printf("pd_flags:0x%08x\n",pd_flags);` at [...]

I get 0x00000180 with my revert of the commit applied (everything working fine).
-> REG_SET_FIELD(RTC_CNTL_TIMER1_REG, RTC_CNTL_CK8M_WAIT, RTC_CNTL_CK8M_WAIT_SLP_CYCLES);

When I go back to the master/release branches I get:  0x00040180 
Then, the issue starts.

esp-wzh, 2025-06-19:
@DarkZeros  Thanks for feedback,  [BIT(1)](https://github.com/espressif/esp-idf/blob/67c1de1eebe095d554d281952fde63c16ee2dca0/components/esp_hw_support/port/esp32/include/soc/rtc.h#L507) is the control bit for power-down RTC_PERIPH power domain, so in both of your versions the RTC_PERIPH power domain is not powered down, you can trace back to see if there is code or driver in your project that requests to keep the RTC_PERIPH power domain during sleep via `esp_sleep_pd_config`, or you can try to set BIT(1) of `pd_flags` directly to force it to power off.

DarkZeros, 2025-06-24:
Thanks for all the help provided to pinpoint the issue. @esp-wzh @L-KAYA 

I think the original confusion comes from the fact that setting RTC_PERIPH pd_option OFF does not turn it off.
It has to be set as AUTO in order to be turned OFF. 
**Is this intended?**

To give more context:
In my project i am using GPIO wakeup (for light sleep) and TOUCH wakeup (from deep sleep), but i was not modifying the pd_option manually. 

It might be the RTC_PERIPH is left in a wrong state (Force ON) after the GPIO wakeup is used once.

And even though i was setting it manually to OFF , still was on in the end (since AUTO was needed).