espressif/esp-idf #15937: A compilation error message for gen_gdbinit.py

ustccw, 2025-05-12:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-1386-g8ad0d3d8f2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

The compilation should not display any error messages.

### What is the actual behavior?

```
Traceback (most recent call last):
  File "/home/runner/work/esp-at/esp-at/esp-idf/components/esp_rom/gen_gdbinit.py", line 72, in <module>
    gdbinit_lines = generate_gdbinit_rom_add_symbols(target)
  File "/home/runner/work/esp-at/esp-at/esp-idf/components/esp_rom/gen_gdbinit.py", line 32, in generate_gdbinit_rom_add_symbols
    raise EnvironmentError(
OSError: ESP_ROM_ELF_DIR environment variable is not defined. Please try to run IDF "install" and "export" scripts.
CMake Warning at /home/runner/work/esp-at/esp-at/esp-idf/tools/cmake/gdbinit.cmake:52 (message):
  Error while generating esp_rom gdbinit
Call Stack (most recent call first):
  /home/runner/work/esp-at/esp-at/esp-idf/tools/cmake/project.cmake:356 (__generate_gdbinit)
  /home/runner/work/esp-at/esp-at/esp-idf/tools/cmake/project.cmake:924 (__project_info)
  CMakeLists.txt:67 (project)
```

### Steps to reproduce.

See GitHub Actions:
https://github.com/espressif/esp-at/actions/runs/14926351432/job/42025822921#step:6:1103

### Build or installation Logs.

```plain

```

igrr, 2025-05-12:
The issue is that in esp-at project, the build script doesn't export all the environment variables which `idf_tools.py export` needs to export:

https://github.com/espressif/esp-at/blob/a016d33ee5a7f8fac4561c4fd588c2f158dd0aeb/build.py#L400-L413

The code only sets PATH and IDF_PYTHON_ENV_PATH. Other paths, like ESP_ROM_ELF_DIR, are not set in the environment, resulting in the above error message.

A future-proof fix would be to export all the variables which `idf_tools.py export` lists.

ustccw, 2025-05-12:
@igrr thanks a lot!

closed 2025-05-13 by igrr