espressif/esp-idf #15938: Getting GET_SHORT_DEV_DESC fail on USB Flash drive

Asanga-SPLat, 2025-05-12:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-DevKitC-1 v1.0

### Power Supply used.

USB

### What is the expected behavior?

Mounting USB flash drive no ESP32-S3 USB host without any issue.

### What is the actual behavior?

I am experiencing an issue failing at the GET_SHORT_DEV_DESC for my one of USB Flash drives. It is imported from china with custom labelled. I am using idf5.4.1, ESP32-S3-DevKitC-V1.0 board with OTG cable and usb/host/msc example for testing. When I am plugin the custom made USB stick, some times it works fully and some times ESP stucks at ENUM: Bad transfer status 1: CHECK_SHORT_DEV_DESC. I checked with commercial available other USB sticks, and they didn't fail at any time.

I changed debug output into Debug and here is the output when plug the USB stick and when it fails.

    I (612) main_task: Calling app_main()
    D (612) intr_alloc: Connected src 38 to int 5 (cpu 0)
    D (612) EXT_PORT: Driver installed
    D (622) EXT_HUB: Driver installed
    I (622) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2
    D (632) intr_alloc: Connected src 16 to int 8 (cpu 0)
    I (632) example: Waiting for USB flash drive to be connected
    D (652) USB_MSC: USB MSC handling start
    D (4882) HUB: Root port reset
    D (4882) HUB: Device tree node (uid=1): new
    D (4882) ENUM: [0:0] Start processing, device address 0
    E (4882) USBH: Dev 0 EP 0 Error
    D (4882) ENUM: [0:0] GET_SHORT_DEV_DESC OK
    D (4882) USBH: Processing actions 0xe
    D (4892) USBH: Default pipe device 0
    E (4892) ENUM: Bad transfer status 1: CHECK_SHORT_DEV_DESC
    E (4892) ENUM: [0:0] CHECK_SHORT_DEV_DESC FAILED
    D (4902) ENUM: [0:0] CANCEL OK
    D (4902) HUB: Disabling root port

Here is the debug output when success on plugging the USB stick (Custom labelled)

    D (612) EXT_PORT: Driver installed
    D (622) EXT_HUB: Driver installed
    I (622) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2
    D (632) intr_alloc: Connected src 16 to int 8 (cpu 0)
[... 147 lines cut ...]
    D (612) intr_alloc: Connected src 38 to int 5 (cpu 0)
    D (612) EXT_PORT: Driver installed
    D (622) EXT_HUB: Driver installed
    I (622) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2
    D (632) intr_alloc: Connected src 16 to int 8 (cpu 0)
    I (632) example: Waiting for USB flash drive to be connected
    D (652) USB_MSC: USB MSC handling start
    D (962) HUB: Root port reset
    D (962) HUB: Device tree node (uid=1): new
    D (962) ENUM: [0:0] Start processing, device address 0
    D (962) ENUM: [0:0] GET_SHORT_DEV_DESC OK

I have an application to start logging data into USB stick. According to my application, there is nobody to unplug the USB stick and plug it back after turning on the machine. So, mounting the USB stick file system on power up it important. Like I said, this is only happens with my custom logo USB stick, which is working fine in my Win PC without any driver issue.

Thanks.

### Steps to reproduce.

Unplug the flash drive and plug it back.

### Debug Logs.

```plain

```

roma-jam, 2025-05-12:
Hi @Asanga-SPLat, 

thanks a lot for creating a separate issue for this problem. 

It seems that device doesn't response and we have unfinished ongoing transaction. 
Unfortunately, there is no available timeout mechanism (https://github.com/espressif/esp-idf/issues/10454) so we will never receive the error in such case. 

The best option would be to implement this timeout, but we didn't plan it yet.

There is a possible workaround at the current moment: you can try to reset the port if the application is not able to find the drive and/or not able to mount the file system for the logging. You can do it with USB Host library public API: 

```
                usb_host_lib_set_root_port_power(false);
                vTaskDelay(10); // Give USB Host task some time so it can handle the disconnection
                usb_host_lib_set_root_port_power(true);  
```

Let me know if this can help or you need any other solution. 
Sorry for the inconvenience.

Asanga-SPLat, 2025-05-12:
Thanks @roma-jam. Little help? I don't know where to put above code lines??

igi540, 2025-05-13:
Hi @Asanga-SPLat ,
Just to share a quick workaround idea that seems to work well in this case.

Instead of resetting the USB root port blindly at startup, we kept the initialization of `usb_host` and `msc_host` unchanged and only modified the main loop in `app_main()`.

Specifically, we changed the line:
`xQueueReceive(app_queue, &msg, portMAX_DELAY);`
to
`if(xQueueReceive(app_queue, &msg, pdMS_TO_TICKS(3000)))`

The original code inside the `if` block remains unchanged (handling of `APP_DEVICE_CONNECTED`, `APP_DEVICE_DISCONNECTED`, etc.).

Then, we added the following `else` branch to handle the timeout case:
else {
if (!dev_present) {
usb_host_lib_set_root_port_power(false);
vTaskDelay(pdMS_TO_TICKS(10));
usb_host_lib_set_root_port_power(true);
vTaskDelay(pdMS_TO_TICKS(100));
}
}
so the whole app_main in the example looks like:
`void app_main(void)
{
    // Create FreeRTOS primitives
    app_queue = xQueueCreate(5, sizeof(app_message_t));
    assert(app_queue);

    BaseType_t task_created = xTaskCreate(usb_task, "usb_task", 4096, NULL, 2, NULL);
    assert(task_created);

    // Init BOOT button: Pressing the button simulates app request to exit
    // It will disconnect the USB device and uninstall the MSC driver and USB Host Lib
    const gpio_config_t input_pin = {
        .pin_bit_mask = BIT64(APP_QUIT_PIN),
        .mode = GPIO_MODE_INPUT,
        .pull_up_en = GPIO_PULLUP_ENABLE,
        .intr_type = GPIO_INTR_NEGEDGE,
    };
    ESP_ERROR_CHECK(gpio_config(&input_pin));
    ESP_ERROR_CHECK(gpio_install_isr_service(ESP_INTR_FLAG_LEVEL1));
    ESP_ERROR_CHECK(gpio_isr_handler_add(APP_QUIT_PIN, gpio_cb, NULL));

    ESP_LOGI(TAG, "Waiting for USB flash drive to be connected");
    msc_host_device_handle_t msc_device = NULL;
    msc_host_vfs_handle_t vfs_handle = NULL;

    // Perform all example operations in a loop to allow USB reconnections
    while (1)
    {
        app_message_t msg;
        if (xQueueReceive(app_queue, &msg, pdMS_TO_TICKS(3000)))
        {

            if (msg.id == APP_DEVICE_CONNECTED)
            {
                if (dev_present)
                {
                    ESP_LOGW(TAG, "MSC Example handles only one device at a time");
                }
                else
                {
                    // 0. Change flag
                    dev_present = true;
                    // 1. MSC flash drive connected. Open it and map it to Virtual File System
                    ESP_ERROR_CHECK(msc_host_install_device(msg.data.new_dev_address, &msc_device));
                    const esp_vfs_fat_mount_config_t mount_config = {
                        .format_if_mount_failed = false,
                        .max_files = 3,
                        .allocation_unit_size = 8192,
                    };
                    ESP_ERROR_CHECK(msc_host_vfs_register(msc_device, MNT_PATH, &mount_config, &vfs_handle));

                    // 2. Print information about the connected disk
                    msc_host_device_info_t info;
                    ESP_ERROR_CHECK(msc_host_get_device_info(msc_device, &info));
                    msc_host_print_descriptors(msc_device);
                    print_device_info(&info);

                    // 3. List all the files in root directory
                    ESP_LOGI(TAG, "ls command output:");
                    struct dirent *d;
                    DIR *dh = opendir(MNT_PATH);
                    assert(dh);
                    while ((d = readdir(dh)) != NULL)
                    {
                        printf("%s\n", d->d_name);
                    }
                    closedir(dh);

                    // 4. The disk is mounted to Virtual File System, perform some basic demo file operation
                    file_operations();

                    // 5. Perform speed test
                    speed_test();

                    ESP_LOGI(TAG, "Example finished, you can disconnect the USB flash drive");
                }
            }
            if ((msg.id == APP_DEVICE_DISCONNECTED) || (msg.id == APP_QUIT))
            {
                if (dev_present)
                {
                    dev_present = false;
                    if (vfs_handle)
                    {
                        ESP_ERROR_CHECK(msc_host_vfs_unregister(vfs_handle));
                        vfs_handle = NULL;
                    }
                    if (msc_device)
                    {
                        ESP_ERROR_CHECK(msc_host_uninstall_device(msc_device));
                        msc_device = NULL;
                    }
                }
                if (msg.id == APP_QUIT)
                {
                    // This will cause the usb_task to exit
                    ESP_ERROR_CHECK(msc_host_uninstall());
                    break;
                }
            }
        }
        else
        {
            if (!dev_present)
            {
                ESP_LOGW(TAG, "No device detected, resetting USB root port...");
                usb_host_lib_set_root_port_power(false);
                vTaskDelay(pdMS_TO_TICKS(10));
                usb_host_lib_set_root_port_power(true);
                vTaskDelay(pdMS_TO_TICKS(100));
            }
        }
    }

    ESP_LOGI(TAG, "Done");
    gpio_isr_handler_remove(APP_QUIT_PIN);
    vQueueDelete(app_queue);
}
`

After the reset, the loop continues and waits again.

This helps recover from cases where the USB device is physically connected at power-up but doesn’t respond properly (e.g. gets stuck at `GET_SHORT_DEV_DESC`). With this logic, the device eventually enumerates correctly after one retry.

Let me know if this makes sense or if there’s a better place where the reset should be handled.

Asanga-SPLat, 2025-05-13:
Thanks @igi540. 

It worked for me even from system reset. At error or unexpected holds dues to not having SHORT_DEV_DESC, the USB device resets and USB drive mounts perfectly.

igi540, 2025-06-03:
Closing the issue as completed, feel free to reopen it if the issue persits.

closed 2025-06-03 by tore-espressif