espressif/esp-idf #15940: UART: When using serial port 0 at a baud rate of 3M, transmitting garbled files larger than 5M will cause the serial port to fail

736676938, 2025-05-13:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32

### Power Supply used.

USB

### What is the expected behavior?

Background: ESP32 enables AP, creates a TCP server, and then the computer connects to the server. Files are sent via TCP, and ESP32 writes the received TCP characters intact using uart_write.The serial port started IRAM
Pass the data received by the TCP server into the serial port intact。

### What is the actual behavior?

Packet loss + garbled text。
100% reproduction

### Steps to reproduce.

1、At a baud rate of 3M, normal log files (13M in size) were transferred. After the transfer was completed, no packets were lost and the data remained stable。
2、The environment remained unchanged. When I converted the file into a compiled elf file, I found that the transmission was over 5M, the serial port lost packets, and all subsequent logs were garbled. After comparison, the garbled data was the same as the elf file I sent with lost packets.
3、The fifo, interrupt threshold and memory application have been modified, but none of them worked

### Debug Logs.

```plain
no
```

songruo, 2025-05-15:
@736676938 Long UART transmissions at high speeds are vulnerable to clock drift, cumulative timing drift could cause the receiver to sample at the wrong points. If you experience packet loss or garbling over large files, especially with binary data, it's very likely due to baud rate mismatch and lack of idle time for realignment.

Can you call `uart_get_baudrate` to check what is the actual baud rate set for UART port 0? It might be slight off from 3M. And try to use the exact baud rate value to decode. If this doesn't help, then you probably need to send shorter frames, with short gaps or checksums to let the receiver realign.

736676938, 2025-06-05:
> If it works with ASCII/text data but loses chunks of binary, the problem may be in the way you transfer data from TCP to the UART; specifically when determining the length of data to copy over.
I have tested that the character length received by TCP is the same as the character length I sent. The problem I can confirm now is that there is packet loss when calling the esp_uart_write() interface. Doubt whether it is related to DMA

736676938, 2025-06-09:
> I haven't heard of any transmission packet loss issue before. Can you check the TX line with a logic analyzer, see if it is really missing a chunk of data? Maybe you can simplify the usage, and create a minimal reproducible program? The TCP part can be eliminated, the compiled elf file can be an embedded file in the chip, and use UART to send directly. I would like to see how you initialize the UART driver, and whether you are using TX ring buffer? And what if you set TX ring buffer size to 0 to use FIFO only, will there be any packet loss?
Ok. I'll spare some time to try and see if it can be reproduced

736676938, 2025-06-10:
> I haven't heard of any transmission packet loss issue before. Can you check the TX line with a logic analyzer, see if it is really missing a chunk of data? Maybe you can simplify the usage, and [...]
When sending 30K data at a time, a packet loss occurred. The source code is as follows：
[uart_echo.zip](https://github.com/user-attachments/files/20663796/uart_echo.zip)

736676938, 2025-06-17:
> I haven't heard of any transmission packet loss issue before. Can you check the TX line with a logic analyzer, see if it is really missing a chunk of data? Maybe you can simplify the usage, and [...]

hello, could you help take a look at this problem?