espressif/esp-idf #15943: `wifi_prov_mgr` example crashes after starting provisioning on ESP32-C2

iamiha, 2025-05-13:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**Description**:

I'm running the `wifi_prov_mgr` example from ESP-IDF v5.2.3 on an ESP32-C2 board, and the device crashes right after the provisioning starts. Here's what I observe:

* Monitor is started using:

  ```bash
  idf.py -p /dev/ttyACM0 -b 74880 monitor
  ```
* Log output ends with:

  ```
  I (575) app: Starting provisioning
  --- Error: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
  --- Waiting for the device to reconnect
  ```
* Log output :

  ```
  I (368) cpu_start: Pro cpu start user code
  I (368) cpu_start: cpu freq: 120000000 Hz
  I (368) cpu_start: Application information:
  I (371) cpu_start: Project name:     wifi_prov_mgr
  I (377) cpu_start: App version:      1
  I (381) cpu_start: Compile time:     May 13 2025 12:01:40
  I (387) cpu_start: ELF file SHA256:  904b2e5e8...
  I (393) cpu_start: ESP-IDF:          v5.2.3-dirty
  I (398) cpu_start: Min chip rev:     v1.0
  I (403) cpu_start: Max chip rev:     v1.99 
  I (407) cpu_start: Chip rev:         v1.0
  I (412) heap_init: Initializing. RAM available for dynamic allocation:
  I (419) heap_init: At 3FCBCB80 len 0001FFF0 (127 KiB): RAM
  I (425) heap_init: At 3FCDCB70 len 0000294C (10 KiB): RAM
  I (433) spi_flash: detected chip: generic
  I (436) spi_flash: flash io: dio
  I (441) sleep: Configure to isolate all GPIO pins in sleep state
  I (447) sleep: Enable automatic switching of GPIO sleep configuration
  I (454) coexist: coex firmware version: 27d8387
  I (459) coexist: coexist rom version de8c800
  I (465) main_task: Started on CPU0
  I (465) main_task: Calling app_main()
  I (475) pp: pp rom version: de8c800
  I (475) net80211: net80211 rom version: de8c800
  I (485) wifi:wifi driver task: 3fcc50ac, prio:23, stack:6144, core=0
  I (485) wifi:wifi firmware version: 82f54d2
  I (485) wifi:wifi certification version: v7.0
  I (495) wifi:config NVS flash: enabled
  I (495) wifi:config nano formating: enabled
  I (495) wifi:Init data frame dynamic rx buffer num: 32
  I (505) wifi:Init static rx mgmt buffer num: 5
  I (505) wifi:Init management short buffer num: 32
  I (515) wifi:Init dynamic tx buffer num: 32
  I (515) wifi:Init static tx FG buffer num: 2
  I (525) wifi:Init static rx buffer size: 1600
  I (525) wifi:Init static rx buffer num: 10
  I (525) wifi:Init dynamic rx buffer num: 32
  I (535) wifi_init: rx ba win: 6
  I (535) wifi_init: tcpip mbox: 32
  I (535) wifi_init: udp mbox: 6
  I (545) wifi_init: tcp mbox: 6
  I (545) wifi_init: tcp tx win: 5760
  I (555) wifi_init: tcp rx win: 5760
  I (555) wifi_init: tcp mss: 1440
  I (555) wifi_init: WiFi IRAM OP enabled
  I (565) wifi_init: WiFi RX IRAM OP enabled
  I (565) wifi_prov_scheme_ble: BT memory released
  I (575) app: Starting provisioning
  I (575) phy_init: phy_version 340,e255ce0,Jun  4 2024,16:44:11
  --- Error: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
  --- Waiting for the device to reconnect 

  ```

* After this point, the device repeatedly resets and shows the same logs again.

**Steps to reproduce**:

1. Clone ESP-IDF v5.2.3 and set it up for `esp32c2`.

2. Navigate to `examples/provisioning/wifi_prov_mgr/`.

3. Run:

   ```bash
   idf.py set-target esp32c2
   idf.py menuconfig  # make any required changes
   idf.py build
   idf.py flash monitor
   ```

4. Observe the crash after `Starting provisioning`.

**Expected behavior**:

Provisioning should begin normally and BLE should become available for Wi-Fi configuration.

**Observed behavior**:

Device crashes and monitor shows “device disconnected” error, then keeps rebooting.

**Target**: ESP32-C2
**ESP-IDF Version**: v5.2.3
**Baud rate**: 74880
**Connection**: `/dev/ttyACM0`

**Additional Info**:

* Attached my sdkconfig file below
* Crash seems to occur right after:

  ```text
  I (575) app: Starting provisioning
  ```

Please let me know if additional logs or memory dumps are needed.

[sdkconfig.txt](https://github.com/user-attachments/files/20182611/sdkconfig.txt)

hrushikesh430, 2025-05-19:
Hello @iamiha , I tried to reproduce your issue with your instructions and sdkcofing, but seems that no issue. Cloud you please mention other details or patch file of app if you have many any changes.

hrushikesh430, 2025-05-29:
Hi @iamiha, Closing this issue due to no response. Please feel free to reopen the issue, if you have any queries.