espressif/esp-idf #15949: About WIFI Provisioning.

williamleeag, 2025-05-14:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

If it possible to run the SmartConfig, SoftAp and BLE provisioning at the same time? Before I try to implement all these together, I want to know if there are some hardware or deep mechanism that make it impossible?

williamleeag, 2025-05-16:
Hi, I am trying to make the BLE AND SOFTAP Provisioning coexist, but it seems that when I use the official SoftAp provisioning app, after I connect to the ESP32C5 AP and start to ask it to scan WIFI, after a few seconds, the phone disconnect with the ESP32C5 AP. If it is a hardware problem?

Espressif-liuuuu, 2025-05-16:
> Hi, I am trying to make the BLE AND SOFTAP Provisioning coexist, but it seems that when I use the official SoftAp provisioning app, after I connect to the ESP32C5 AP and start to ask it to scan WIFI, after a few seconds, the phone disconnect with the ESP32C5 AP. If it is a hardware problem?

Hi @williamleeag, thanks for your report.
From hardware aspect, there is only one RF for ESP32C5 and only one module can work at the time. Basically we need to apply TDD to make softAP works with BLE as stable as possbile, though it cant be too well, due to there is no protocol sleep mechanism for softAP.
For current sdk, the TDD is still not fully implemented, but there is one in internal test, I can provide you with the demo if you are glad to have a try.  

And back to your report, do you mean that even without BLE, your phone disconnected from Wi-Fi connection when you trying to order softAP (actually ap-station mode) to do Wi-Fi scan? If so, its possible due to the long process of 5G scan, which is only related with firmware.  
Meanwhile, if bt controller is enabled, the scan time can be longer, which can be easier to trigger phone disconnection in this scenario.  
A simple way to check if its the scenario above
* try select 2G only mode by call `esp_wifi_set_band`.
* try disable full bt stack, no initialization.  

We will fix the issue by updating connection timeout and supporting home channel return at coexistence scenarios.

Espressif-liuuuu, 2025-05-19:
> > > Hi, I am trying to make the BLE AND SOFTAP Provisioning coexist, but it seems that when I use the official SoftAp provisioning app, after I connect to the ESP32C5 AP and start to ask it to scan WIFI, after a few seconds, the phone disconnect with the ESP32C5 AP. If it is a hardware problem?
> > 
> > 
> > Hi [@williamleeag](https://github.com/williamleeag), thanks for your report. From hardware aspect, there is only one RF for ESP32C5 and only one module can work at the time. Basically we need to apply TDD to make softAP works with BLE as stable as possbile, though it cant be too well, due to there is no protocol sleep mechanism for softAP. For current sdk, the TDD is still not fully implemented, but there is one in internal test, I can provide you with the demo if you are glad to have a try.
> > And back to your report, do you mean that even without BLE, your phone disconnected from Wi-Fi connection when you trying to order softAP (actually ap-station mode) to do Wi-Fi scan? If so, its possible due to the long process of 5G scan, which is only related with firmware. Meanwhile, if bt controller is enabled, the scan time can be longer, which can be easier to trigger phone disconnection in this scenario. A simple way to check if its the scenario above
> > 
> > * try select 2G only mode by call `esp_wifi_set_band`.
> > * try disable full bt stack, no initialization.
> > 
> > We will fix the issue by updating connection timeout and supporting home channel return at coexistence scenarios.
> 
> Hi, Thanks for your reply, I am using the wifi_provisioning component, I tried not to call wifi_prov_scheme_ble.prov_start, then the scan is done successfully, so it should be possible to disable the other two provisioning modes when I am using one of them(1.Connected with the AP, 2. Connected with the BLE, 3. SmartConfig Scan Done).

Yes that it is, it seems that BLE coexistence does make the scan process longer and trigger to disconnection. And yes, if you can disable BLE provision when using softAP provision, it shall be OK.