espressif/esp-idf #15950: ESP MESH esp_mesh_get_subnet_nodes_num and esp_mesh_get_subnet_nodes_list error

mgatocarmona, 2025-05-14:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When I use these functions with nodes in an ESP Wi-Fi Mesh, the esp_mesh_get_subnet_nodes_num function always returns 0 or an error for nodes in layers greater than 3, even though I know these nodes have children.

Additionally, when I use esp_mesh_get_subnet_nodes_list, I sometimes get the list in an incorrect order—nodes from higher layers appear before those from lower layers.

MaxwellAlan, 2025-05-15:
Hi @mgatocarmona 

Plz provide some more detail info for this issue,

- Which IDF version or commit?
- Which chips?
- IDF example or only using your demo code can reproduce this issue?

Thanks.

mgatocarmona, 2025-05-16:
At the moment I am using IDF v5.2 on the HEAD.
I am using ESP32c6 for the childs and the root node is always ESP32. The ESP32 is the one using the functions esp_mesh_get_subnet_nodes_num and esp_mesh_get_subnet_nodes_list.
I have not tried IDF examples for this, I have only used my own code.

closed 2025-06-30 by Alvin1Zhang