espressif/esp-idf #15955: 烧录时i2c报错

wyp1207, 2025-05-15:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF 5.3 

### Espressif SoC revision.

ESP32-S3 N16R8

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32 S3

### Power Supply used.

USB

### What is the expected behavior?

正常下载，能正常使用

### What is the actual behavior?

相反，编译没有问题，下载时，i2c传输失败

[image]

### Steps to reproduce.

源码：
https://codeload.github.com/78/xiaozhi-esp32/zip/refs/heads/main
搭建环境：
https://icnynnzcwou8.feishu.cn/wiki/JEYDwTTALi5s2zkGlFGcDiRknXf
最后一步烧录报错

### Debug Logs.

```plain

```

### Diagnostic report archive.

mythbuster5, 2025-05-15:
Then you need to check 1. whether i2c address is correct. 2. hardware is connected correctly. 3. Check online data if there is any glitch or noise. I don't think this is related to an idf issue. Please check it carefully, thanks!

mythbuster5, 2025-05-15:
Ask related question in xiaozhi-esp32 if you use their code or hardware. I'm going to close this issue soon. Please feel free to reopen if you have clearer clue for debugging, thanks!

closed 2025-05-15 by mythbuster5