espressif/esp-idf #15956: The IEEE 802.15.4 communication reports an "Instruction access fault" error.

Weimj11, 2025-05-15:
I ported the IEEE 802.15.4 communication routine to the RT-Thread operating system and used RTT's semaphore in the receive callback function to notify the processing of received data frames. The sender transmits data every few seconds, and then I found that the receiver may report an "Instruction access fault" error and restart after receiving data a few times. The error and restart always occur in debug_print_packet. What could be the reason?
The error is as follows:
[10:23:22.552]收←◆Guru Meditation Error: Core 0 panic'ed (Instruction access fault). Exception was unhandled.

Core 0 register dump:
MEPC : 0x00000080 RA : 0x40022d60 SP : 0x408157c0 GP : 0x40804dd0
TP : 0xdeadbeef T0 : 0x00000080 T1 : 0x10000000 T2 : 0xdeadbeef
S0/FP : 0x40835464 S1 : 0x00000041 A0 : 0x00000000 A1 : 0x00000035
A2 : 0x40835424 A3 : 0x00000041 A4 : 0x00000001 A5 : 0x00000000
A6 : 0xa0000000 A7 : 0x0000000a S2 : 0x40835465 S3 : 0x40835424
S4 : 0x40815960 S5 : 0x00000001 S6 : 0x42041a3f S7 : 0x00000000
S8 : 0x00000001 S9 : 0x00000000 S10 : 0x40815974 S11 : 0x00000002
T3 : 0x00000000 T4 : 0x00006f48 T5 : 0x00000050 T6 : 0xdeadbeef
MSTATUS : 0x00001880 MTVEC : 0x40800001 MCAUSE : 0x00000001 MTVAL : 0x00000080
MHARTID : 0x00000000

The code is as follows:

#include <string.h>
#include <esp_ieee802154.h>
#include <esp_log.h>
#include "rtthread.h"
#include "802154_proto.h"

#define RADIO_TAG "main"

#define PANID 0x4242
#define CHANNEL 19

#define SHORT_NOT_CONFIGURED 0xFFFE
#define SHORT_TEST_RECEIVER 0x2222

static rt_thread_t tid1 = RT_NULL;

static struct rt_semaphore sem;
static uint8_t shared_frame[257]; // 共享缓冲区，用于存储接收到的帧

static void debug_handler_task(void *parameter);
static void debug_print_packet(uint8_t *packet, uint8_t packet_length);

void esp_ieee802154_receive_done(uint8_t *frame, esp_ieee802154_frame_info_t *frame_info)
{
ESP_EARLY_LOGI(RADIO_TAG, "rx OK, received %d bytes", frame[0]);

for (uint8_t idx = 1; idx < frame[0]; idx += 8)
{
ESP_EARLY_LOGI(RADIO_TAG, "%02x %02x %02x %02x %02x %02x %02x %02x",
frame[idx], frame[idx + 1], frame[idx + 2], frame[idx + 3],
frame[idx + 4], frame[idx + 5], frame[idx + 6], frame[idx + 7]);
}

// 将接收到的帧存储到共享缓冲区
memcpy(shared_frame, frame, frame[0] + 1);
// 通过信号量通知线程处理数据
rt_sem_release(&sem);

esp_ieee802154_receive_handle_done(frame);
}

void esp_ieee802154_receive_failed(uint16_t error)
{
ESP_EARLY_LOGI(RADIO_TAG, "rx failed, error %d", error);
}

void esp_ieee802154_receive_sfd_done(void)
{
ESP_EARLY_LOGI(RADIO_TAG, "rx sfd done, Radio state: %d", esp_ieee802154_get_state());
}

int main(void)
[... 206 lines cut ...]
ESP_LOGI(RADIO_TAG, "PAN %04x S %04x %02X:%02X:%02X:%02X:%02X:%02X:%02X:%02X to %04x %02X:%02X:%02X:%02X:%02X:%02X:%02X:%02X %s", pan_id,
short_src_addr, src_addr[0], src_addr[1], src_addr[2], src_addr[3], src_addr[4], src_addr[5], src_addr[6], src_addr[7],
short_dst_addr, dst_addr[0], dst_addr[1], dst_addr[2], dst_addr[3], dst_addr[4], dst_addr[5], dst_addr[6], dst_addr[7],
broadcast ? "BROADCAST" : "");

if (broadcast)
for (uint8_t idx = 0; idx < 8; idx++)
dst_addr[idx] = 0xFF;

break;
}
case FRAME_TYPE_ACK:
{
uint8_t sequence_number = packet[position++];
ESP_LOGI(RADIO_TAG, "Ack (%u)", sequence_number);
break;
}
default:
{
ESP_LOGE(RADIO_TAG, "Packet ignored because of frame type (%u)", fcs->frameType);
break;
}
}
ESP_LOGI(RADIO_TAG, "-----------------------");
}

chshu, 2025-05-15:
Please try out the following:

1. `esp_ieee802154_receive_done()` and `esp_ieee802154_receive_sfd_done()` are all in ISR context, please remove the `ESP_EARLY_LOGI()` to ensure the ISR functions could finish faster
2. comment the frame parsing code from `debug_print_packet()`, only print the received frame

If it works without any issue, you can add the parsing code gradually.

Weimj11, 2025-05-15:
I removed ESP_EARLY_LOGI() from both esp_ieee802154_receive_done() and esp_ieee802154_receive_sfd_done(), and debug_print_packet() only implements printing the received data frames. However, it still reports an error "Instruction access fault", but before the error occurs, the number of communication attempts has increased significantly.

void debug_print_packet(uint8_t *packet, uint8_t packet_length)
{
if (packet_length < sizeof(mac_fcs_t))
        return; // Can't be a packet if it's shorter than the frame control field

    ESP_LOGI(TAG, "rx OK, received %d bytes", packet_length);
    for (uint8_t idx = 1; idx < packet_length; idx += 8)
    {

        ESP_LOGI(TAG, "%02x %02x %02x %02x %02x %02x %02x %02x",
                       packet[idx], packet[idx + 1], packet[idx + 2], packet[idx + 3],
                       packet[idx + 4], packet[idx + 5], packet[idx + 6], packet[idx + 7]);
    }

}

chshu, 2025-05-16:
@Weimj11 The simplified `debug_print_packet()` looks good, it should not cause the issue.

Could you please share other related code and the whole log? Is it possible to reproduce the issue based on the basic IDF example: https://github.com/espressif/esp-idf/tree/master/examples/ieee802154/ieee802154_cli, want exclude the issue from RT-Thread porting.

Alvin1Zhang, 2025-05-29:
@Weimj11 Thanks for reporting, can you please help share if any further updates for the issue? Thanks.

Alvin1Zhang, 2025-07-02:
Thanks for reporting, will close due to inactivity, feel free to reopen.

closed 2025-07-02 by Alvin1Zhang