espressif/esp-idf #15957: HTTP POST requests issue

NirajSiroya, 2025-05-15:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello team , please help for the below problem.

We are using the **ESP32-C6** along with the **ESP-IDF Eclipse plugin**.

**Certificate Handling Summary:**
We have stored the client certificate and key in SPIFFS memory.
After reading them from SPIFFS, the parsing completes successfully,

Our controller acts as an HTTP client. For HTTP POST requests, we are including a client certificate. We are also using the same certificate for WebSocket connections, and it works correctly there.

However, when we use the same certificate with an HTTP POST request, we receive a 400 Bad Request response from the server, along with the error message: "400 No required SSL certificate was sent" . The server requires the Content-Type header to be set to application/json, and we are already sending that header in our HTTP POST request.

Please find the configuration as below.

esp_http_client_config_t config_post = {
.port = 443,
.method = HTTP_METHOD_POST,
.timeout_ms = 6000,
.skip_cert_common_name_check = true,
.cert_pem = NULL,
.client_cert_pem = (char*)client_cert_data,
.client_key_pem = (char*)client_key_data,
.transport_type = HTTP_TRANSPORT_OVER_SSL,
.event_handler = _http_event_handler,
};
config_post.url = Cloudurl;

Point to be noted : On the server side, SSL/TLS client certificate verification is mandatory.
On the client side, we do not require server certificate verification.

Thanks in advance.

nopnop2002, 2025-05-19:
Can your HTTPS server be accessed from Windows or Linux?

If so, there might be a problem with the client certificate.

nileshkale123, 2025-05-23:
Hello @NirajSiroya ,

Please check as mentioned in above suggestion:

> Can your HTTPS server be accessed from Windows or Linux?
> 
> If so, there may be a problem with the client certificate in SPIFFS memory.

You may try testing the server using the curl tool, or by embedding your server URL into the esp_http_client example provided in ESP-IDF.

If the issue persists, please share the relevant debug logs or a minimal reproducible example so we can assist you further in identifying the root cause.