espressif/esp-idf #15958: [5.3][MESH] esp_mesh_scan_get_ap_ie_len returns 0 for 4.4 mesh access point

KonssnoK, 2025-05-15:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi @zhangyanjiaoesp 
We are currently testing cross-compatibility between 4.4 and 5.3 to prepare for rollout of the updated SDK on the field.

we follow the code of ip_internal_network for handling the scan results in fixed root:

```
    for (int i = 0; i < num; i++) {
        // Get each single WiFi record in our cache
        esp_mesh_scan_get_ap_ie_len(&ie_len);
        esp_mesh_scan_get_ap_record(&record, &assoc);

        if (ie_len == sizeof(assoc)) {
```

We have a 4.4 mesh AP active with LTE connection, and 3 5.3 devices in fixed root scanning.

This is what we see from the device:

```
I (421787) mesh: 4026<scan>self-organized is disabled, users shall read out scan results by themselves.
I (421787) mesh_main: <MESH_EVENT_SCAN_DONE>number:39
W (421787) wifi:Haven't to connect to a suitable AP now!
I (421817) mesh_hand: [3], ac:17:54:00:0a:b1, channel:1, rssi:-33, ie_len 0 assoc 58
I (421827) mesh_hand: [4]Samsung Washer_E50AJTWF179A, bc:10:2f:66:ab:52, channel:1, rssi:-84, ie_len 0 assoc 58
W (421837) mesh_hand: <MESH>7c:df:a1:e0:74:41 is idle
W (421847) mesh_hand: <MESH>7c:df:a1:e2:7b:85 is idle

```

I would expect the mesh device ac:17:54:00:0a:b1, to send and ie_len of 58, or, if anything changed from 4.4 to 5.3, of a different size of 58, but not 0.

Any idea what could be happening?

Thanks

KonssnoK, 2025-05-15:
nevermind, the issue is this :

```
#if !MESH_IE_ENCRYPTED
    cfg.crypto_funcs = NULL;
#endif
```

which probably should be
```
#if !CONFIG_MESH_IE_ENCRYPTED
    cfg.crypto_funcs = NULL;
#endif
```

@zhangyanjiaoesp if you can update the example

KonssnoK, 2025-05-27:
@zhangyanjiaoesp 
my idea was "if the children do not encrypt they are not able to read the IE because it's encrypted".

According to what you wrote apparently this is not the case..

In that case then i have no idea, once i added back the encryption functions they started communicating with my other device..

It should be straightforward to test
- fixed root
- one root device with CONFIG_MESH_IE_ENCRYPTED set
- one child device with CONFIG_MESH_IE_ENCRYPTED not set
the child should not see the root as a mesh device 🤷‍♂

zhangyanjiaoesp, 2025-05-28:
@KonssnoK 
In one mesh network, the `CONFIG_MESH_IE_ENCRYPTED ` should be same, both enabled or disabled.

KonssnoK, 2025-05-29:
hi @zhangyanjiaoesp 
i did a rapid test with ip_internal_network from code 
https://github.com/KonssnoK/esp-idf/commit/851cc7d48397121591d9c0a4c9819d0c26b50424

two devkits with
`#if !CONFIG_MESH_IE_ENCRYPTED`

7B84
A304

one devkit with
`#if !MESH_IE_ENCRYPTED`

7440

Uncomment the code block 
mesh_handover.c 455
```

else {

            ESP_LOGI(TAG, "[%d]%s, "MACSTR", channel:%u, rssi:%d, ie_len %d assoc %d", i,
                record.ssid, MAC2STR(record.bssid), record.primary,
                record.rssi,
                ie_len, sizeof(assoc)
            );
        }
```

NO wifi available.

And this is what i see:

A304 reports
```
I (871969) mesh_hand: [7], 7c:df:a1:e0:74:41, channel:1, rssi:-18, ie_len 0 assoc 58
W (871999) mesh_hand: <MESH>7c:df:a1:e2:7b:85 is idle
```

7440 reports
```
I (391819) mesh_hand: [3], 7c:df:a1:e2:a3:05, channel:1, rssi:-7, ie_len 0 assoc 58
I (391859) mesh_hand: [7], 7c:df:a1:e2:7b:85, channel:1, rssi:-15, ie_len 0 assoc 58
```

So i confirm the bug is there.

logs attached

[devkit_A304.txt](https://github.com/user-attachments/files/20504308/devkit_A304.txt)
[devkit_7440.txt](https://github.com/user-attachments/files/20504309/devkit_7440.txt)

zhangyanjiaoesp, 2025-06-05:
@KonssnoK 
When you call `esp_mesh_scan_get_ap_ie_len()`, does it return OK? If it returns FAIL, when the len will be 0.  If the API returns fail, it means there are no APs in the saved AP list of the node.

KonssnoK, 2025-06-05:
@zhangyanjiaoesp i do not check the return code, but i do have values for all the nominal wifi stations (as can be seen in the logs)
I can try to add the result to the example

KonssnoK, 2025-06-27:
sorry i had no time to reproduce and i will not work on this topic anymore...

I guess at some point someone else will incur into this issue :)

closed 2025-06-27 by KonssnoK