espressif/esp-idf #15959: Secure Boot Verification failed

aygh4266, 2025-05-15:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello everyone,

I have enabled SBv2 and double-checked that the digest matches the public key.
Unfortunately, I’m still getting a "verification failed" message.

So I don't know what is the problem 
Here are my logs:
```cpp
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
Valid secure boot key blocks: 0
No signature block magic byte found at signature sector (found 0x61 not 0xe7). Image not V2 signed?
secure boot verification failed
ets_main.c 329
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
```

Logs from espsecure.py
```cpp
espsecure.py verify_signature --version 2 --keyfile sbv2_private_dev.pem build/app-template.bin
espsecure.py v4.8.1
Signature block 0 is valid (RSA).
Signature block 0 verification successful using the supplied key (RSA).
Signature block 1 invalid. Skipping.
Signature block 2 invalid. Skipping.
```

another summary log
```cpp
>espefuse.py --chip esp32s3 -p COM18 summary
espefuse.py v4.8.1
Connecting....

A fatal error occurred: Secure Download Mode is enabled. The tool can not read eFuses.
```

mahavirj, 2025-05-16:
@aygh4266 You have flashed signed bootloader in the flash? Verification command looks like for your application, how about the bootloader? Please share detailed steps for your setup along with the EFuse summary.

aygh4266, 2025-05-27:
> [@aygh4266](https://github.com/aygh4266) You have flashed signed bootloader in the flash? Verification command looks like for your application, how about the bootloader? Please share detailed steps for your setup along with the EFuse summary.

I have tried to flash the bootloader, partition table and the binary image to the corresponding addresses manually and it works using esptool.py. 

An automatic flashing isn't working using the button in esp-idf extension in VSCode