espressif/esp-idf #15960: Using SPI makes it easy to crash

coollofty, 2025-05-15:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3.2

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

LCD SPI is used normally

### What is the actual behavior?

LCD SPI often crashes

### Steps to reproduce.

No pattern has been found yet

### Debug Logs.

```plain
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed).

Core  0 register dump:
PC      : 0x421b6bb6  PS      : 0x00060034  A0      : 0x8037ae76  A1      : 0x3fca5f60
--- 0x421b6bb6: lcd_spi_post_trans_color_cb at E:/esp/esp-adf/esp-idf/components/esp_lcd/spi/esp_lcd_panel_io_spi.c:413

A2      : 0x3fcc7b94  A3      : 0x3fca90bc  A4      : 0x80376ef2  A5      : 0x3fcaf970
A6      : 0x3fca0704  A7      : 0x3c5744b0  A8      : 0x8037ac22  A9      : 0x3fca5f40
A10     : 0x0000cdcd  A11     : 0x0000cdcd  A12     : 0x80382831  A13     : 0x3fcaf960
A14     : 0x00060f23  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x421b6bb3:0x3fca5f60 0x4037ae73:0x3fca5f80 0x40378839:0x3fca5fb0 0x40379015:0x3fcaf9b0 0x403764e1:0x3fcaf9d0 0x4038245d:0x3fcaf9f0
--- 0x421b6bb3: spi_lcd_prepare_param_buffer at esp_lcd_panel_io_spi.c:?
0x4037ae73: spi_intr at E:/esp/esp-adf/esp-idf/components/esp_driver_spi/src/gpspi/spi_master.c:938
0x40378839: _xt_lowint1 at E:/esp/esp-adf/esp-idf/components/xtensa/xtensa_vectors.S:1240
0x40379015: spi_flash_op_block_func at E:/esp/esp-adf/esp-idf/components/spi_flash/cache_utils.c:141 (discriminator 1)
0x403764e1: ipc_task at E:/esp/esp-adf/esp-idf/components/esp_system/esp_ipc.c:65
0x4038245d: vPortTaskWrapper at E:/esp/esp-adf/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

### Diagnostic report archive.

[diag-a08e6bf7-1b1a-4735-80fa-546f79468a58.zip](https://github.com/user-attachments/files/20227716/diag-a08e6bf7-1b1a-4735-80fa-546f79468a58.zip)

### More Information.

My device uses two SPI devices, LCD and TF-CARD, which use SPI2HOST and SPI3HOST respectively. CPU0 is responsible for updating the LCD, at least when I call functions related to the esp_1cd prefix, it must be on CPU0. But it's always easy for SPI LCD to crash suddenly at some point. There is no pattern, the same action may have a problem this time and may not have a problem the next time

suda-morris, 2025-05-16:
How did you initialize the SPI bus in `spi_bus_config_t`?

Kainarx, 2025-05-16:
Hi @coollofty Could you try this patch and see whether the problem is fixed?

[0001-fix-spi_lcd-fix-spi_lcd-crash-when-cache-disabled.Patch](https://github.com/user-attachments/files/20238260/0001-fix-spi_lcd-fix-spi_lcd-crash-when-cache-disabled.Patch)

coollofty, 2025-05-17:
> How did you initialize the SPI bus in `spi_bus_config_t`?

```
        spi_bus_config_t bus =
            {
                .mosi_io_num = configs.lcd.mosi,
                .miso_io_num = GPIO_NUM_NC,                
                .sclk_io_num = configs.lcd.clock,
                .quadwp_io_num = -1,
                .quadhd_io_num = -1,
                .max_transfer_sz = (int)(w * 80 * sizeof(uint16_t)),
                .flags = SPICOMMON_BUSFLAG_SCLK | SPICOMMON_BUSFLAG_MOSI | SPICOMMON_BUSFLAG_MASTER,
                .intr_flags = ESP_INTR_FLAG_LOWMED | ESP_INTR_FLAG_IRAM
            };
        ESP_ERROR_CHECK(spi_bus_initialize(SPI2_HOST, &bus, SPI_DMA_CH_AUTO));
```

coollofty, 2025-05-17:
> Hi [@coollofty](https://github.com/coollofty) Could you try this patch and see whether the problem is fixed?
> 
> [0001-fix-spi_lcd-fix-spi_lcd-crash-when-cache-disabled.Patch](https://github.com/user-attachments/files/20238260/0001-fix-spi_lcd-fix-spi_lcd-crash-when-cache-disabled.Patch)

Okay, I'll test it as soon as possible and come over for feedback

suda-morris, 2025-05-19:
> ```
>         spi_bus_config_t bus =
>             {
>                 .mosi_io_num = configs.lcd.mosi,
>                 .miso_io_num = GPIO_NUM_NC,
>                 .sclk_io_num = [...]

Do you have a special reason to enable the `ESP_INTR_FLAG_IRAM` flag?

closed 2025-08-22 by suda-morris