espressif/esp-idf #15961: esp32 ble_ancs_demo

Whiteeeey, 2025-05-15:
The esp ble_ancs_demo advertises and connects to an iphone with no problems. However, when monitoring the notificaton output it appears that the esp32 is only receiving incoming call notifications. Is this due to the set up of the demo or should it be receiving all notifications. I have tried to look through the code to determine this but as far as I can tell all notifications are described and therefore I would have thought that all notifications should be received. Please let me know if you have any suggestions as to how I can get all notifications from my iphone.

Whiteeeey, 2025-05-27:
Does anyone have an answer to this question?

esp-zhp, 2025-06-04:
Hi,

We’ve verified that BLE ANCS is working correctly. The ESP32 **can successfully receive SMS notifications from the iPhone**, as shown in the log below:

```
EventID: New message  
CategoryID: Social  
Identifier: com.apple.Mobile / SMS  
Title: 自己 Halley  
Message: Mini
```

This confirms that the notification has been received properly.

We suggest checking your iPhone settings to ensure notifications from other apps are allowed to be forwarded via Bluetooth. Please try the following steps:

1. Go to **Settings > Bluetooth** on your iPhone.
2. Tap the connected ESP32 device.
3. Make sure **“Show Notifications”** or **“Share System Notifications”** is enabled.

Our test was conducted using **iPhone 18 with iOS 18.1**, which you can use as a reference.

[image]

[image]

esp-zhp, 2025-06-05:
Hi @Whiteeeey，

Thanks for the update.

We suggest you try testing with a newer iPhone model if available. Sometimes, older devices like the iPhone 6s may behave differently in terms of ANCS support or Bluetooth notification forwarding, which could lead to inconsistent results.

In addition, you can enable and capture the BLE HCI logs on the ESP32 side to verify whether the iPhone is actually sending the notifications over the air. This will help determine whether the issue is on the iPhone's side or within the ESP32 ANCS processing.

You can follow the instructions in the link below to enable and analyze HCI logs:
👉 [[ESP-IDF Bluetooth HCI log guide](https://github.com/espressif/esp-idf/tree/master/tools/bt)](https://github.com/espressif/esp-idf/tree/master/tools/bt)

If the HCI log confirms that the notification is already sent by the phone but not processed correctly on ESP32, we may consider making adjustments (within the protocol's allowance) to ensure the notification is reported properly.

Let us know what you find from the HCI log. We’re here to help further if needed.

Whiteeeey, 2025-06-07:
Thanks, I will try that.