espressif/esp-idf #15962: Zigbee: Assertion failed in zcl_general_commands.c:612 when calling esp_zb_zcl_report_attr_cmd_req on ESP32-H2 (IDF v5.4.1)

julosnacar, 2025-05-15:
Target:
ESP32-H2 (Rev v0.1)

Problem Description:
I am developing a Zigbee Router application on an ESP32-H2. The device successfully joins a Zigbee network managed by a Bellows/zigpy coordinator (Sonoff ZBDongle-E). The ESP32-H2 exposes a custom cluster (0xFC01) with three SinglePrecisionFloat attributes, and also standard clusters like Basic and Identify.
The application crashes with an assertion failure: Zigbee stack assertion failed zcl/zcl_general_commands.c:612 when I attempt to send an explicit attribute report using esp_zb_zcl_report_attr_cmd_req.

This crash occurs:
When trying to report an attribute from my custom cluster (0xFC01).
Crucially, it also occurs when trying to report a standard attribute from a standard cluster, such as ZCLVersion (ID 0x0000) from the Basic cluster (ID 0x0000).
This suggests the issue might be with the esp_zb_zcl_report_attr_cmd_req function itself or its prerequisites in the current SDK version for ESP32-H2, rather than being specific to my custom cluster.
Steps to Reproduce:
Initialize ESP32-H2 as a Zigbee Router.
Define and register an endpoint with standard clusters (Basic, Identify) and a custom cluster (0xFC01) with float attributes. (Details of custom cluster creation below).
Join a Zigbee network.
After successfully joining, attempt to send an explicit attribute report using esp_zb_zcl_report_attr_cmd_req.
Code Snippet (Simplified send_explicit_report causing the crash):

#include "esp_zigbee_core.h"
#include "esp_zigbee_zcl_command.h" // For esp_zb_zcl_report_attr_cmd_req and definitions
// ... other necessary includes ...

#define MY_ENDPOINT 1 // Example endpoint

// Function called after network is joined
static void attempt_explicit_report_basic_cluster() {
    esp_zb_zcl_report_attr_cmd_t report_cmd;
    memset(&report_cmd, 0, sizeof(esp_zb_zcl_report_attr_cmd_t));

    report_cmd.zcl_basic_cmd.dst_addr_u.addr_short = 0x0000; // Coordinator
    report_cmd.zcl_basic_cmd.dst_endpoint = MY_ENDPOINT;     // Assuming coordinator listens on a similar EP or this is EP of device itself for ZCL
    report_cmd.zcl_basic_cmd.src_endpoint = MY_ENDPOINT;
    report_cmd.address_mode = ESP_ZB_APS_ADDR_MODE_16_ENDP_PRESENT;
    
    // Attempting to report a standard attribute from Basic Cluster
    report_cmd.clusterID = ESP_ZB_ZCL_CLUSTER_ID_BASIC;         // 0x0000
    report_cmd.attributeID = ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID; // 0x0000

    // manuf_specific, manuf_code, direction, dis_default_resp are 0 due to memset.
    // report_cmd.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV; (implicitly 0)

    ESP_LOGI("TEST", "Attempting explicit report for Basic Cluster ZCLVersion...");
    esp_err_t ret = esp_zb_zcl_report_attr_cmd_req(&report_cmd); // <-- CRASHES HERE

    if (ret != ESP_OK) {
        ESP_LOGE("TEST", "Error sending explicit report for Basic ZCLVersion: %s", esp_err_to_name(ret));
    } else {
        ESP_LOGI("TEST", "Explicit report for Basic ZCLVersion sent (or attempted).");
    }
}

// Custom Cluster Creation (for context, though crash also happens with Basic Cluster):
// esp_zb_attribute_list_t *custom_cluster_attr_list = esp_zb_zcl_attr_list_create(0xFC01);
// float initial_val = 0.0f;
// esp_zb_cluster_add_attr(custom_cluster_attr_list, 0xFC01, 0x0001, ESP_ZB_ZCL_ATTR_TYPE_SINGLE, ESP_ZB_ZCL_ATTR_ACCESS_READ_ONLY | ESP_ZB_ZCL_ATTR_ACCESS_REPORTING, &initial_val);
// ... (add other attributes) ...
// esp_zb_cluster_list_add_custom_cluster(cluster_list, custom_cluster_attr_list, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
// ... (endpoint registration) ...

Expected Behavior:
esp_zb_zcl_report_attr_cmd_req should successfully send (or at least attempt to send without crashing) an attribute report to the coordinator.
Actual Behavior:
The ESP32-H2 firmware crashes with an assertion:
Zigbee stack assertion failed zcl/zcl_general_commands.c:612
(Followed by a backtrace pointing to esp_zb_zcl_report_attr_cmd_req).
Debug Logs (Backtrace

--- Waiting for the device to reconnect...........
I (3750) ZIGBEE_SENSOR_ROUTER: Cambiando estado del LED de 1 a 3
I (3800) ZIGBEE_SENSOR_ROUTER: Timer de parpadeo detenido para cambio de estado.
[... 51 lines cut ...]
40823230: 0x60cf37f6 0x3551ec30 0xe066634b 0xbd52e3ac 0xc5dbf21b 0xd9d941f6 0xbbb3a1f3 0xb512e6ff
40823250: 0x53b40d1c 0x14d2cb08 0x26de6164 0xb2db3af4 0x593ef97b 0x703f5141 0x0f67e1bf 0x8376f57a
40823270: 0x94cec66e 0x5d430fa5 0x977f4629 0xe9efc8ca 0x9a73b559 0xa3bcef70 0x10da84cb 0x00fbf316
40823290: 0xac025ad5 0x7a96e46e 0x4b091f52 0x237dcd43 0xd65d9f27 0x5a1b2eeb 0x47000066 0xf754e7b3
408232b0: 0x029524a7 0x10b0d670 0xd9364ba0 0xec6c6e5f 0x6952419f 0x4a285414 0xe0f9209a 0xf36531ee
408232d0: 0x56e189db 0xda47d10c 0xb6398eb1 0x23c46b51 0xab119635 0x6a04e4f3 0x957ddaa3 0x83242c1d
408232f0: 0x147f19fd 0x4eb7be8b 0x0e5eb621 0xa692c982 0x1d9b3150 0xbf4be989 0x92becdc7 0x973b6fb6
40823310: 0x79022b80 0x1283d029 0x50775ef6 0xcdc34737 0x2d5b9535 0x1ad82ba3 0x95fe7f7f 0x887ed091
40823330: 0x6baa6998 0xb5b596a7 0x2a1dfa72 0xe7a3b8c6 0x40f990fc 0x4f1f5b24 0x03333ed6 0x6a9103fa
40823350: 0x1b2187eb 0x4c24d798 0x8f59fd77 0x9d652f48 0x215251bb 0x6f7d55da 0x59b4bc75 0x0081da6a
40823370: 0x15ed2bbd 0x7ce34238 0xb6af5c69 0xe1037f9b 0x0800478c 0x191f0e76 0xacc3c0f2 0x8b173b78
40823390: 0x24d5d9f5 0xb0f7591d 0x9444259d 0xc5a1a714 0x2d779444 0x086b2e47 0xa467f5a0 0x8991d6e4
408233b0: 0x8c46831d 0xa4b08706 0x7c3436ff 0x94453321 0x92784851 0x295a5843 0x5fe9308e 0x8e814fb3
408233d0: 0x364bf47b 0x2a3b109e 0xaa70acea 0xc20ed5fb 0x9f8af1b3 0xc344ea86 0x4cf254c3 0xaaec1aeb

Additional Information:
The coordinator (Python script using bellows and zigpy) can successfully send a "Configure Reporting" command to the ESP32-H2 for the custom cluster attributes, and the ESP32-H2 responds with Status.SUCCESS.
However, automatic reports are also not being received by the coordinator, even after successful configuration. This might be related to the same underlying issue that causes esp_zb_zcl_report_attr_cmd_req to crash.
The ESP32-H2 is configured with ESP_ZB_DEVICE_TYPE_ROUTER.
Device Profile ID: ESP_ZB_AF_HA_PROFILE_ID (0x0104).
Device ID: ESP_ZB_HA_SIMPLE_SENSOR_DEVICE_ID (0x000C).
Question:
Could there be an issue with esp_zb_zcl_report_attr_cmd_req on ESP32-H2 in this IDF version, or specific prerequisites for its use (especially regarding attribute descriptors or internal state) that are not being met?
Why might automatic reporting also fail despite a successful "Configure Reporting" response?
Any guidance or insights would be greatly appreciated.

xieqinan, 2025-05-16:
@julosnacar ，

According to the ZCL specification, the `ZCLVersion` attribute of the Basic cluster does **not** have report access, meaning it cannot be reported using `esp_zb_zcl_report_attr_cmd_req()`.  

Additionally, the assert you're encountering may be caused by a missing Basic cluster client. This is because the report command is configured with:  
```c
report_cmd.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV;
```
This setting requires that the corresponding cluster **client** is registered. Please check and ensure that the Basic cluster client has been properly added on the sender side.

julosnacar, 2025-05-17:
Subject: Re: [espressif/esp-idf] Zigbee: Assertion failed in zcl/zcl_general_commands.c:612 when calling esp_zb_zcl_report_attr_cmd_req on ESP32-H2 (IDF v5.4.1)

Hi @xieqinan 

Thank you so much for your quick and insightful response regarding the ZCLVersion attribute and the potential need for a local cluster client.
Update based on your suggestions:
Reporting ZCLVersion: You were absolutely correct. I've confirmed that ZCLVersion is not a reportable attribute, so my test attempting to report it using esp_zb_zcl_report_attr_cmd_req was indeed invalid. This explains the failure in that specific test case.
Adding Local Cluster Clients (Server-Side):
Following your advice, I modified my ESP32-H2 firmware (main.c) to register both SERVER and CLIENT roles for the clusters involved on the device's endpoint. Specifically, I added client-role placeholders for:
Basic Cluster (0x0000)
Identify Cluster (0x0003)
My Custom Cluster (0xFC01)
The relevant part of my esp_zb_task now looks like this (simplified):
// In esp_zb_task on ESP32-H2
// ...
esp_zb_cluster_list_t *cluster_list = esp_zb_zcl_cluster_list_create();

// Basic Cluster (Server with attributes + Client placeholder)
esp_zb_attribute_list_t *basic_server_attrs = esp_zb_basic_cluster_create(...);
// (add manuf_name, model_id to basic_server_attrs)
esp_zb_cluster_list_add_basic_cluster(cluster_list, basic_server_attrs, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
esp_zb_attribute_list_t *basic_client_attrs = esp_zb_basic_cluster_create(...); // Default/empty config
esp_zb_cluster_list_add_basic_cluster(cluster_list, basic_client_attrs, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE);

// Identify Cluster (Server + Client placeholder)
esp_zb_attribute_list_t *identify_server_attrs = esp_zb_identify_cluster_create(...);
esp_zb_cluster_list_add_identify_cluster(cluster_list, identify_server_attrs, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
esp_zb_attribute_list_t *identify_client_attrs = esp_zb_identify_cluster_create(...);
esp_zb_cluster_list_add_identify_cluster(cluster_list, identify_client_attrs, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE);

// Custom Cluster 0xFC01 (Server with attributes + Client placeholder)
esp_zb_attribute_list_t *custom_server_attrs = esp_zb_custom_cluster_create(...); // My helper
esp_zb_cluster_list_add_custom_cluster(cluster_list, custom_server_attrs, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
esp_zb_attribute_list_t *custom_client_attrs = esp_zb_zcl_attr_list_create(0xFC01); // Empty attr list for client
esp_zb_cluster_list_add_custom_cluster(cluster_list, custom_client_attrs, ESP_ZB_ZCL_CLUSTER_CLIENT_ROLE);

// ... add clusters to endpoint, register device ...
Use code [with caution](https://support.google.com/legal/answer/13505487).
C
Outcome of these changes:
Positive for esp_zb_zcl_report_attr_cmd_req: After adding the client-role clusters on the ESP32-H2, the assertion failure at zcl/zcl_general_commands.c:612 NO LONGER OCCURS when calling esp_zb_zcl_report_attr_cmd_req (with report_cmd.direction = ESP_ZB_ZCL_CMD_DIRECTION_TO_SRV;). This is a significant improvement!
The ESP32 now logs that it attempts to send the explicit report, and esp_zb_zcl_report_attr_cmd_req returns ESP_OK.
// Log from ESP32 when send_explicit_report calls esp_zb_zcl_report_attr_cmd_req:
// I (timestamp) ZIGBEE_SENSOR_ROUTER: Intentando reporte explícito para Cluster 0xFC01, AttrID 0x0001 (Direction TO_SRV)
// I (timestamp) ZIGBEE_SENSOR_ROUTER: Reporte explícito enviado (o intento) para AttrID 0x0001
Use code [with caution](https://support.google.com/legal/answer/13505487).
C
Remaining Issue: Reports Not Received by Coordinator
Despite the ESP32-H2 now seemingly able to attempt sending explicit reports without crashing (and esp_zb_zcl_set_attribute_val working correctly to update local attribute values), I am still not receiving any attribute reports (neither automatic nor the explicit ones) on my Python-based coordinator (using zigpy with a Sonoff ZBDongle-E).
Coordinator-Side Configuration:
My Python script successfully sends "Configure Reporting" commands to the ESP32-H2 for each attribute in my custom cluster (0xFC01).
The ESP32-H2 responds to these "Configure Reporting" requests with Status.SUCCESS.
I have also implemented an explicit ZDO Bind_req from the coordinator to the ESP32's custom cluster server, which also reports Status.SUCCESS from the ESP32.
ESP32-H2 Behavior (Current State):
Local attributes are updated via esp_zb_zcl_set_attribute_val().
Explicit reports via esp_zb_zcl_report_attr_cmd_req() now return ESP_OK (no crash, no ESP_ERR_NOT_FOUND).
However, no data packets (reports) seem to arrive at the coordinator/Python application.
Current Hypothesis / Question:
Given that:
The "Configure Reporting" from the coordinator to the ESP32 is successful.
The explicit ZDO Binding from the coordinator to the ESP32's cluster is successful.
The ESP32's call to esp_zb_zcl_report_attr_cmd_req now returns ESP_OK.
...it seems the ESP32 should be sending reports. The fact that they are not being received by the coordinator (and automatic reports also don't arrive) suggests there might still be an issue within the ESP-IDF Zigbee stack's report generation/dispatch mechanism for custom clusters, or perhaps a more subtle binding/routing issue that isn't immediately apparent.
Could there be any other prerequisite or stack configuration on the ESP32-H2 side that is necessary for either automatic or explicit reports from a custom cluster (with float attributes) to be correctly dispatched to a bound coordinator, even if esp_zb_zcl_report_attr_cmd_req returns ESP_OK?

I am using ESP-IDF v5.4.1

Thank you again for your help!

xieqinan, 2025-05-19:
Hi @julosnacar ，

**Regarding this issue, you can follow the steps below to debug:**

* If possible, please provide a minimal code example that reproduces the issue. I suspect some details might be missing.
* Do you have a sniffer? If not, you can follow the [documentation](https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/developing.html#sniffer-and-wireshark) to install one. Use it to capture whether the `attribute report` is actually being sent.
* You can also refer to the official usage of `esp_zb_zcl_report_attr_cmd_req()` to compare it with your implementation.
* Enabling Zigbee debug mode in `menuconfig` and calling `esp_zb_set_trace_level_mask()` can help provide more debug information.

xieqinan, 2025-05-19:
The issue: https://github.com/espressif/esp-zigbee-sdk/issues/641 is the same with you, which is used as the reference.