espressif/esp-idf #15965: Debug serial port UART 0 to UART 2 issue

alexsvn, 2025-05-16:
### Is your feature request related to a problem?

ESP32 After changing to UART2, the application log can only print a portion and is no longer output. But the app_main function starts normally
How to troubleshoot? Currently, I have integrated Blueroid and Wi-Fi on the ESP32 side and can start normally

The output log is only partial, and it is not output after the Bluetooth controller log is output

<details>
<summary>log</summary>
  I (1200) BT_BLE_COEX: UART Pins: Tx:1 Rx:3 RTS:4294967295 CTS:4294967295
  I (1200) BTDM_INIT: BT controller compile version [2a2631f]
  I (1210) BTDM_INIT: Bluetooth MAC: ec:c9:ff:81:e4:0a
  I (1220) phy_init: phy_version 4840,02e0d70,Sep  2 2024,19:39:07
</details>

nopnop2002, 2025-05-19:
>The output log is only partial, and it is not output after the Bluetooth controller log is output

Are you using this to limit log bells somewhere?
esp_log_level_set("*", ESP_LOG_ERROR);

nopnop2002, 2025-05-20:
Which of the following is the logging output to UART2?

Output directly to UART2 in the callback function of esp_log_set_vprintf.

Write to MessageBuffer or RingBuffer in the callback function of esp_log_set_vprintf, and have the UART2 output task read from there.

nopnop2002, 2025-05-20:
Do you use esp_log_set_vprintf to output logs to UART2?

alexsvn, 2025-05-21:
SDKCONFIG
<details>
<summary>sdkconfig uart0 switch uart2</summary>

```conf
-CONFIG_CONSOLE_UART_DEFAULT=y
-# CONFIG_CONSOLE_UART_CUSTOM is not set
+# CONFIG_CONSOLE_UART_DEFAULT is not set
+CONFIG_CONSOLE_UART_CUSTOM=y
 # CONFIG_CONSOLE_UART_NONE is not set
 # CONFIG_ESP_CONSOLE_UART_NONE is not set
 CONFIG_CONSOLE_UART=y
-CONFIG_CONSOLE_UART_NUM=0
+# CONFIG_CONSOLE_UART_CUSTOM_NUM_0 is not set
+# CONFIG_CONSOLE_UART_CUSTOM_NUM_1 is not set
+CONFIG_CONSOLE_UART_NUM=2
+CONFIG_CONSOLE_UART_TX_GPIO=17
+CONFIG_CONSOLE_UART_RX_GPIO=16
 CONFIG_CONSOLE_UART_BAUDRATE=115200
```

nopnop2002, 2025-05-21:
From here.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig-reference.html#config-esp-console-uart-custom

Due to an ESP32 ROM bug, UART2 is not supported for console output via esp_rom_printf.

o-marshmallow, 2025-05-23:
As @nopnop2002 said, UART2 cannot be used as console output, so the option `CONFIG_ESP_CONSOLE_UART_CUSTOM_NUM_2` is not available, yet I see in your `sdkconfig` that you defined `CONFIG_ESP_CONSOLE_UART_CUSTOM_NUM_2`. 

If you are using the latest version of IDF, you shouldn't be able to select the UART2 as a custom console via the `menuconfig`.

closed 2025-06-03 by o-marshmallow