espressif/esp-idf #15970: `mbedtls_ssl_handshake returned -0x2700` on self signed certificate

mbastida123, 2025-05-16:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, we are integrating an HTTPS connection in production. We have created the .key and .crt files for the server as so:
```
openssl req -nodes -x509 -sha256 -newkey rsa:4096 \
  -keyout test.key \
  -out test.crt \
  -days 9125 \
  -subj "/C=ES/ST=Barcelona/L=city/O=enterprise SL/OU=test/CN=subdomain.domain.es"  \
  -addext "subjectAltName = URI.1:subdomain.domain.es" 
```
After the files are uploaded to the server the .pem file is retrieved using:
`echo | openssl s_client -connect subdomain.domain.es:443 -showcerts | awk '/BEGIN CERTIFICATE/,/END CERTIFICATE/ { print }' > ca_cert.pem`

This .pem file is attached to the firmware using:

```
extern const unsigned char pem_cert_dev_start[] asm("_binary_ca_cert_pem_start");
extern const unsigned char pem_cert_dev_end[]   asm("_binary_ca_cert_pem_end");
```

And configured in the HTTP client as so:
```
esp_http_client_config_t config = {
        .url=base_url,
        #ifndef CONFIG_ESP_TLS_SKIP_SERVER_CERT_VERIFY
        .cert_pem=(char *)pem_cert_dev_start,
        .cert_len=(size_t)(pem_cert_dev_end - pem_cert_dev_start),
        #endif
        .event_handler = _http_event_handler,
        .buffer_size_tx = 2048,
        .user_data = response, // Pass address of local buffer to get response
        #ifdef CONFIG_ESP_TLS_SKIP_SERVER_CERT_VERIFY
        .skip_cert_common_name_check = true,
        #endif
    };
```

The output of said code is the following:

> W (16:01:41.202) HTTPS: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT on attempt 2
> D (16:01:41.212) esp-tls: host:subdomain.domain.es: strlen 37
> D (16:01:41.219) esp-tls: [sock=54] Resolved IPv4 address: [some_ip]
> D (16:01:41.226) esp-tls: [sock=54] Connecting to server. HOST: subdomain.domain.es, Port: 443
> D (16:01:41.302) esp-tls: handshake in progress...
> E (16:01:41.372) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
> I (16:01:41.373) esp-tls-mbedtls: Failed to verify peer certificate!
> D (16:01:41.376) esp-tls-mbedtls: verification info:   ! The certificate Common Name (CN) does not match with the expected CN

Aparently the CN doesn't match. I find it dificult to believe since the .pem file was obtained directly from the server.... Any help?

nopnop2002, 2025-05-19:
The ca_cert.pem file may not have been created correctly.
Try this.
```
openssl s_client -showcerts -connect subdomain.domain.es:443 </dev/null >hoge

start=`grep -e "-----BEGIN CERTIFICATE-----" -n hoge | sed -e 's/:.*//g' | tail -n 1`

last=`grep -e "-----END CERTIFICATE-----" -n hoge | sed -e 's/:.*//g' | tail -n 1`

sed -n ${start},${last}p hoge > ca_cert.pem
```

mbastida123, 2025-05-19:
> The ca_cert.pem file may not have been created correctly.
> 
> What will happen if You create ca_cert.pem in this way?
> 
> ```
> openssl s_client -showcerts -connect subdomain.domain.es:443 </dev/null >hoge
> 
> start=`grep -e "-----BEGIN CERTIFICATE-----" -n hoge | sed -e 's/:.*//g' | tail -n 1`
> 
> last=`grep -e "-----END CERTIFICATE-----" -n hoge | sed -e 's/:.*//g' | tail -n 1`
> 
> sed -n ${start},${last}p hoge > ca_cert.pem
> ```
> 
> The CA root cert is the **last cert** given in the chain of certs.

Hi @nopnop2002 and thanks for the help.

This is the output for the commands you attached:
> $ sh generate_pem.sh 
> depth=0 C = ES, ST = Barcelona, L = city, O = company SL, OU = company, CN = subdomain.domain.es
> verify error:num=18:self signed certificate
> verify return:1
> depth=0 C = ES, ST = Barcelona, L = city, O = company SL, OU = company, CN = subdomain.domain.es
> verify return:1
> DONE

The .pem file is generated correctly, but I still get the same error:

> E (07:42:00.921) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
> I (07:42:00.921) esp-tls-mbedtls: Failed to verify peer certificate!
> D (07:42:00.925) esp-tls-mbedtls: verification info:   ! The certificate Common Name (CN) does not match with the expected CN

One thing to note in case it helps. The address is a little bit more complex than what I wrote as example. It is something in the lines of `this-that-those-v2.subdomain.domain.es`
I don't think it should have an impact, but since tls is complaining about a mismatch in the CN....

mbastida123, 2025-05-19:
Update: I have also tried with a certificate whose domain is for a URL with the form subsubdomain.subdomain.domain.es and it still gives the same error

mbastida123, 2025-05-21:
Any ideas?

mahavirj, 2025-05-21:
@mbastida123 

Few pointers:

- You may check following code (add some debug prints) on what hostname is getting passed to TLS stack: https://github.com/espressif/esp-idf/blob/27d68f57e6bdd3842cd263585c2c352698a9eda2/components/esp-tls/esp_tls_mbedtls.c#L809
- You may use `common_name` field from HTTP client configuration to set the correct CN: https://github.com/espressif/esp-idf/blob/27d68f57e6bdd3842cd263585c2c352698a9eda2/components/esp_http_client/include/esp_http_client.h#L190
- Consider enabling mbedTLS debug logs to get more information

mbastida123, 2025-05-23:
It's resolved.

`common_name` seems to be the key. However, I had to update idf to 5.x to have this feature. Thank you for the support!

closed 2025-05-23 by mbastida123