espressif/esp-idf #15971: FreeRTOS: Allow to include a custom #include on

KammutierSpule, 2025-05-16:
### Is your feature request related to a problem?

On "FreeRTOS.h" is includes a custom

```
/* Application specific configuration options. */
#include "FreeRTOSConfig.h"
```
Which on esp-idf is its own FreeRTOS port 
`/opt/esp/idf/components/freertos/config/include/freertos/FreeRTOSConfig.h`

which then includes
`#include "sdkconfig.h"`

that is a "Automatically generated file" and placed on build folder.

### Describe the solution you'd like.

A way to include our custom include on some particular project.

### Describe alternatives you've considered.

I considered to append an #include to the automatica generated .h file on the build, but that is not elegant.

### Additional context.

This allows to take more advanced use and extend of the FreeRTOS. Eg. setup trace defines.

KammutierSpule, 2025-05-17:
An ugly workaround

```
add_custom_target(append_include ALL)
add_dependencies(${CMAKE_PROJECT_NAME}.elf append_include)
add_custom_command(
    TARGET append_include
    POST_BUILD
    COMMAND printf "\"%s%s\\n\" \"#include\" \"<${CMAKE_CURRENT_LIST_DIR}/full local path.to.file>\"" >>sdkconfig.h
    WORKING_DIRECTORY ${CMAKE_BINARY_DIR}/config
    COMMENT "Append include to sdkconfig.h"
)
```

The include must be full workspace path.

because, even if I try to include a path:

```
idf_build_set_property(INCLUDE_DIRECTORIES ${CMAKE_CURRENT_LIST_DIR}/my path APPEND)
target_include_directories(${CMAKE_PROJECT_NAME}.elf PUBLIC ${CMAKE_CURRENT_LIST_DIR}/my path)
```

it wont work, as build wont be able to see these include paths.

KammutierSpule, 2025-05-17:
Related discussion on forum:
https://esp32.com/viewtopic.php?p=147750

SoucheSouche, 2025-05-19:
Hi @KammutierSpule, there is currently no proper way to provide your own definition of `traceTASK_CREATE`. We already have a menuconfig option for [application level tracing](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/app_trace.html). The definition of all the traceXX macros are provided in `SEGGER_SYSVIEW_FreeRTOS.h` included in `FreeRTOSConfig.h`.

Theoretically, we could add a configuration to allow users to provide a header file with the definition of all those tracing macros very much like it is currently done in `SEGGER_SYSVIEW_FreeRTOS.h`.

What is the your use case for this macro? I will discuss it internally to weigh in the pros and cons of adding support for this and get back to you.

SoucheSouche, 2025-05-19:
@KammutierSpule, thanks for clarifying. I will talk about this feature request internally on Wednesday and I will come back to you after we reached a conclusion concerning whether we should support it or not.

igrr, 2025-05-19:
Related to this... @gerekon would having a way to include a custom trace header help for https://github.com/percepio/TraceRecorderSource/pull/17/?

SoucheSouche, 2025-05-21:
Hello @KammutierSpule,

After discussing the feature request, we agreed that it could be useful to support user custom app tracing.

I will forward this feature request to the team in charge of maintaining the app_trace component.

In the mean time, I can offer you an alternative solution to allow you to continue with your development.

You can create a specific header including the definition of all the trace macros in e.g., main/include folder of your project:
```c
#pragma once

#ifdef __cplusplus
extern "C" {
#endif

#include "sdkconfig.h"

#ifdef traceTASK_CREATE
#undef traceTASK_CREATE
#endif
#define traceTASK_CREATE(pxNewTCB) <your implementation of the macro>

#ifdef __cplusplus
}
#endif

``` 

and you can force include it into all source files by modifying the main/CMakeLists.txt with the added command:

```cmake
idf_build_set_property(COMPILE_OPTIONS "-include<your_header_directory>/<your_header_name>" APPEND)
```

This is a bit hacky and you should be careful what you are defining in the header file but it should solve your issue for now.

KammutierSpule, 2025-05-21:
Thanks @SoucheSouche for consideration! I will also try your suggestion.

gerekon, 2025-05-22:
@KammutierSpule 

Yes, Looks like workaround suggested should work. But not sure because all trace macros should be defined before they are defined as NOP in [FreeRTOS.h](https://github.com/espressif/esp-idf/blob/27d68f57e6bdd3842cd263585c2c352698a9eda2/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h#L629).
If suggested approach with header does not work for you you can define all those macros in command line
```cmake
idf_build_set_property(COMPILE_OPTIONS "-DtraceTASK_CREATE=custom_traceTASK_CREATE" APPEND)
```

For SystemView and Percepio (not yet commited) we hard-code in "FreeRTOSConfig.h"
```c
    #if CONFIG_SYSVIEW_ENABLE
        #include "SEGGER_SYSVIEW_FreeRTOS.h"
        #undef INLINE /* to avoid redefinition */
    #endif /* CONFIG_SYSVIEW_ENABLE */

  #ifdef CONFIG_PERCEPIO_TRACERECORDER_ENABLED
      #include "trcRecorder.h"
  #endif
```

Plan to make this more flexible, so user will be able to provide it own "FreeRTOSTraceConfig.h" with macros.

SoucheSouche, 2025-05-23:
@KammutierSpule, I suggest closing the issue unless you have other issues you would like to discuss related to this ticket.

KammutierSpule, 2025-05-23:
> I will forward this feature request to the team in charge of maintaining the app_trace component.

> I suggest closing the issue unless you have other issues you would like to discuss related to this ticket.

Do you have or could you later come here paste the related ticket to the feature implementation? When do I know that it will be merged?

gerekon, 2025-05-27:
Looks like a good idea to keep this ticket until solution is implemented.

MrMabulous, 2025-07-20:
Has there been an update to this? Is idf_build_set_property(COMPILE_OPTIONS "-include<your_header_directory>/<your_header_name>" APPEND) still the cleanest way to do this?