espressif/esp-idf #15975: Can't flash ESP32-C5 -  chip revision in range [v0.0 - v0.99] (this chip is revision v1.0)

grigio, 2025-05-17:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.4.1

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I'd like to flash the blink example

### What is the actual behavior?

A fatal error occurred: bootloader/bootloader.bin requires chip revision in range [v0.0 - v0.99] (this chip is revision v1.0). Use --force to flash anyway.

Maybe related to https://esp32.com/viewtopic.php?p=147571&hilit=0x5fd1406f#p147571

### Steps to reproduce.

```
Executing task: /home/grigio/.espressif/python_env/idf5.4_py3.9_env/bin/python /home/grigio/esp/v5.4.1/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset --chip esp32c5 write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x2000 bootloader/bootloader.bin 0x10000 blink.bin 0x8000 partition_table/partition-table.bin 

esptool.py v4.8.1
Serial port /dev/ttyUSB0
Connecting....
WARNING: This chip doesn't appear to be a ESP32-C5 (chip magic value 0x5fd1406f). Probably it is unsupported by this version of esptool.
Chip is ESP32-C5 (revision v1.0)
Features: WiFi 6, BT 5, IEEE802.15.4
Crystal is 48MHz
MAC: ---
BASE MAC: ---
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...

A fatal error occurred: bootloader/bootloader.bin requires chip revision in range [v0.0 - v0.99] (this chip is revision v1.0). Use --force to flash anyway.
```

### Build or installation Logs.

```plain

```

jaylikesbunda, 2025-05-17:
if you clone master esp-idf it should work

grigio, 2025-05-17:
thanks, with `master` it worked

closed 2025-05-17 by grigio