espressif/esp-idf #15976: ULP RISC V i2c does not work after ULP core halts and restarts

0xPiyush, 2025-05-17:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Seeed Studio Xiao ESP32-S3

### Power Supply used.

USB

### What is the expected behavior?

ULP RISC V reads ADS1115 connected to i2c every 3 seconds while the Main core is in deep sleep. 

### What is the actual behavior?

i2c from RTC domain works as expected in the main core but in the ULP RISC V core i2c operations only work properly at the first boot of ULP RISC V, it does not work on consecutive restarts using ulp_set_wakeup_period. As the logs show the main core reads the ADS1115 value from i2c as expected but after the ULP RISC V core is started, only the first i2c reading is valid, all the consecutive attempts fail, each attempt is done at an interval of 3 seconds (This is the ulp core wake up time period), I have verified that the i2c is connect properly to the default RTC i2c pins i.e. GPIO 2 for SCL and GPIO 3 for SDA.

### Steps to reproduce.

1. Main core code ```main/main.c```:
```
#include <string.h>

#include "ads1115_defs.h"
#include "esp_log.h"
#include "esp_sleep.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "ulp_main.h"
#include "ulp_riscv.h"
#include "ulp_riscv_i2c.h"
#define TAG "main"

extern const uint8_t ulp_main_bin_start[] asm("_binary_ulp_main_bin_start");
extern const uint8_t ulp_main_bin_end[] asm("_binary_ulp_main_bin_end");

static void init_ulp_program(uint32_t wakeup_interval_sec) {
    esp_err_t err = ulp_riscv_load_binary(
        ulp_main_bin_start, (ulp_main_bin_end - ulp_main_bin_start));
    ESP_ERROR_CHECK(err);

    /* Set the wakeup interval for the ULP timer */
    if (wakeup_interval_sec == 0) {
        ESP_LOGW(TAG, "ULP wakeup interval is 0, setting to 1 second minimum.");
        wakeup_interval_sec = 1;
[... 505 lines cut ...]
I (1202) main: ADS1115 value: 31894, error: 0
I (1202) main: ADS1115 value: 0, error: -3
I (1202) main: ADS1115 value: 0, error: -3
I (1202) main: ADS1115 value: 0, error: -3
I (1212) main: ADS1115 value: 0, error: -3
I (1212) main: ADS1115 value: 0, error: -3
I (1222) main: ADS1115 value: 0, error: -3
I (1222) main: ADS1115 value: 0, error: -3
I (1222) main: ADS1115 value: 0, error: -3
I (1232) main: ADS1115 value: 0, error: -3
--- Error: ClearCommError failed (PermissionError(13, 'The device does not recognize the command.', None, 22))
--- Waiting for the device to reconnect............................................................
I (1202) main: Wakeup cause: 4
I (1202) main: Waking up from deep sleep after 30 seconds
I (1202) main: ADS1115 value: 0, error: -3
I (1202) main: ADS1115 value: 0, error: -3
I (1202) main: ADS1115 value: 0, error: -3
I (1202) main: ADS1115 value: 0, error: -3
I (1212) main: ADS1115 value: 0, error: -3
I (1212) main: ADS1115 value: 0, error: -3
I (1212) main: ADS1115 value: 0, error: -3
I (1222) main: ADS1115 value: 0, error: -3
I (1222) main: ADS1115 value: 0, error: -3
I (1232) main: ADS1115 value: 0, error: -3
```

0xPiyush, 2025-05-19:
Could this possibly resolve the errors I'm seeing?
https://github.com/espressif/esp-idf/issues/14571

SoucheSouche, 2025-05-20:
Hi @0xPiyush, I will take a look at your code and get back to you after.

0xPiyush, 2025-05-20:
That was it, thank you so much @SoucheSouche , I have no idea how I missed it, ULP i2c works as expected now. I'm going to close this issue now, thank you for helping out.
Cheers.