espressif/esp-idf #15977: ESP-IDF v5.3 Only Shows Secure Boot V1 in menuconfig (ESP32-D0WD-V3, Rev 3.1)

Vaibhav0216, 2025-05-17:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi everyone,
I'm using ESP-IDF v5.3 on Windows (PowerShell), and my chip info is below:
```
PS C:\Users\Admin\Desktop\std_gw> esptool.py read_mac
esptool.py v4.8.1
Found 2 serial ports
Serial port COM10
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: cc:db:a7:2e:e8:44
Uploading stub...
Running stub...
Stub running...
MAC: cc:db:a7:2e:e8:44
Hard resetting via RTS pin...
```
According to the documentation, this chip supports Secure Boot V2.
However, when I go to:
```
idf.py menuconfig → Security Features → Enable hardware Secure Boot in bootloader → Select secure boot version
```
I only see Secure Boot V1, and there's no option to select V2.

I have already confirmed my ESP-IDF version:
```
ESP-IDF v5.3-dirty
```

What could be causing this? Do I need to configure something extra to enable Secure Boot V2 in menuconfig?

mahavirj, 2025-05-19:
@Vaibhav0216 Secure boot V2 is supported starting ESP32 ECO3 revision. Please select the revision from "Hardware Settings" section and that shall enable the secure boot V2 menu:

[image]

AdityaHPatwardhan, 2025-05-26:
@Vaibhav0216 The same issue is tracked here (https://github.com/espressif/esp-idf/issues/15992)
Since, the main issue in this case (Secure Boot V1) is resolved.
I will go ahead and close this one.
We can follow up on the other issue separately.

closed 2025-05-26 by AdityaHPatwardhan