espressif/esp-idf #15979: program crashes while using i2s and sdspi at the same time

Terrazed, 2025-05-17:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1

### Espressif SoC revision.

ESP32-C3FH4 v0.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

create a task that read the content of a file in a sdcard via the sdspi API and write it to a buffer while having an other task that reads that buffer to send it over i2s. the tasks works with 2 buffer in a ping-pong manner. the loop goes like : (sd fill buffer0 and i2s read buffer1) -> (sd fills buffer1 and i2s read buffer0) and so on.

### What is the actual behavior?

it works sometimes but crashes unexpectedly after some time. when it crashed it doesn't always do the same things, it sometimes reboots directly, sometimes get stuck, sometimes gives a crash report...

### Steps to reproduce.

launch the function audio_pipeline_play_file("your/path");
the code is available on my github repo [here](https://github.com/Terrazed/random_sampler-code)

### Debug Logs.

```plain
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1400
load:0x403cc710,len:0xbe8
load:0x403ce710,len:0x3128
entry 0x403cc71a
I (30) boot: ESP-IDF v5.4.1 2nd stage bootloader
I (30) boot: compile time May 17 2025 18:53:30
I (30) boot: chip revision: v0.4
I (30) boot: efuse block revision: v1.3
I (34) boot.esp32c3: SPI Speed      : 80MHz
I (38) boot.esp32c3: SPI Mode       : DIO
I (41) boot.esp32c3: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
[... 161 lines cut ...]
64	}
#0  esp_cpu_wait_for_intr () at /home/trz/esp/esp-idf/components/esp_hw_support/cpu.c:64
#1  0x420113cc in esp_vApplicationIdleHook () at /home/trz/esp/esp-idf/components/esp_system/freertos_hooks.c:58
#2  0x40388700 in prvIdleTask (pvParameters=<optimized out>) at /home/trz/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4353
#3  0x40386eea in vPortTaskWrapper (pxCode=0x1, pvParameters=0x0) at /home/trz/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:255

ELF file SHA256: 38139f02b

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038072e
--- 0x4038072e: esp_restart_noos at /home/trz/esp/esp-idf/components/esp_system/port/soc/esp32c3/system_internal.c:112
```

### Diagnostic report archive.

[diag-00d355fe-1c63-40b5-8fd9-28acccf05641.zip](https://github.com/user-attachments/files/20266029/diag-00d355fe-1c63-40b5-8fd9-28acccf05641.zip)

### More Information.

In the debug logs it shows that a complete loop happened with no problem once but the second one crashed in the middle. most of the time it happens directly in the first loop. but i wanted to show that it works sometimes so i reseted the board until it happen in the way i wanted to show

i'm wondering if this is due to collisions from the APIs wanting to use the dma at the same time.

L-KAYA, 2025-05-19:
Hi @Terrazed , Thanks for reporting! It looks like something wrong during the interrupt. May I know if you registered any callbacks and what the callback will do? It would be great if you can provide your callback function code~

Terrazed, 2025-05-19:
Hi @L-KAYA, Thank you for helping me :-) 

I am not directly using interrupts, but i think freeRTOS does. I'm guessing tasks register as callbacks. But i don't think they are executed as an interrupt (not sure tho).

I have two tasks working in parallel that are sychronized by message queues. 
Both tasks can be found here [https://github.com/Terrazed/random_sampler-code/blob/main/main/audio_pipeline.c](https://github.com/Terrazed/random_sampler-code/blob/main/main/audio_pipeline.c)

The task called `void audio_pipeline_i2s_task(void* sem_end_task)` is the one used for sending the buffer to the i2s and notify the other task to write in the next buffer.

The task called `void audio_pipeline_sd_task(void* sem_end_task)` is the one used for reading the sd card and writing in the buffer and then signal the i2s task that the buffer is ready to play.

ginkgm, 2025-05-20:
Hi @L-KAYA , @Terrazed ,

One thing for you to notice. For SDHC and SDXC cards, there may be a delay between the read command and the data returned, which is 100ms at max. (See SD spec 3.01 version, section 4.12.4 table 4-51.) We have observed delay of about 20ms in our previous work.

Please consider this delay in the driver or the implementation and avoid possible issue when the data can't return in time.

ginkgm, 2025-05-20:
Another possible problem about the delay is the file system and SDSPI. SDSPI itself is not very fast. And when you are also using the file system, please make sure the read start point and size is multplie of 512 so that we can use longer command and smaller buffer.

ginkgm, 2025-05-20:
So much about the performance. But the Interrupt WDT is the direct issue. Maybe we can start from figuring out what causes the IWDT...

Terrazed, 2025-05-20:
Hi @ginkgm, thank you for making me notice these optimizations !

> Hi [@L-KAYA](https://github.com/L-KAYA) , [@Terrazed](https://github.com/Terrazed) ,
>
> One thing for you to notice. For SDHC and SDXC cards, there may be a delay between the read command and the data [...]

For this possible issue, correct me if I'm wrong, but I think my code already waits until the transaction with the sd card is complete, because I'm using a message queue to notify the i2s task that the buffer is ready to play and I'm only sending that signal after reading the sd card with `fread()` which I believe is a blocking function. so the i2s task should already be waiting for the sdcard to respond.

> Another possible problem about the delay is the file system and SDSPI. SDSPI itself is not very fast. And when you are also using the file system, please make sure the read start point and size is multplie of 512 so that we can use longer command and smaller buffer.

I hadn't really thought about that, but as a good practice, I try to always round my numbers to the closest power of 2. As the sample rate for the .wav files I use is 16000, I rounded the buffer to 16384 so that it contains around 0.5sec of data.

Terrazed, 2025-05-20:
From what i understood here :
[https://docs.espressif.com/projects/esp-techpedia/en/latest/esp-friends/advanced-development/debugging/guru-meditation.html#interrupt-wdt-timeout-on-cpu0-cpu1](https://docs.espressif.com/projects/esp-techpedia/en/latest/esp-friends/advanced-development/debugging/guru-meditation.html#interrupt-wdt-timeout-on-cpu0-cpu1)
The real issue is freeRTOS or hidden behind freeRTOS. I don't really know how to debug it...
because it seems like freeRTOS is doing one of my task in an ISR but that is not how it should behave. Right ?

L-KAYA, 2025-05-21:
Maybe it's a concurrency issue. The buffers are not fully protected by the semaphore locks when assign or read it's values. Could you try to move the lock like this:

[image]

[image]

Terrazed, 2025-05-21:
@L-KAYA, You are right that the buffers were not fully covered by the semaphores.
I moved the semaphores but unfortunately the issue is still there...

Terrazed, 2025-05-26:
re-flowing the board seems to have solved the issue! I believed something with the crystal was not soldered correctly and caused clock instabilities which messed with the timings...

closed 2025-05-26 by Terrazed