espressif/esp-idf #15981: esp_image: invalid segment length 0x71ebb4ba

zzzh, 2025-05-18:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.2

### Espressif SoC revision.

ESP32-C3 (revision 3)

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

CORE-ESP32

### Power Supply used.

USB

### What is the expected behavior?

esp_image_verify passed in esp_ota_end to Finish OTA.

### What is the actual behavior?

esp_image: invalid segment length 0x71ebb4ba

### Steps to reproduce.

can't provide the image, so can't reproduce.

### Debug Logs.

```plain
esp_image: invalid segment length 0x71ebb4ba
```

### Diagnostic report archive.

command not found: esp-idf-diag

### More Information.

the image is good to run, but can't pass esp_image_verify, tried to increase MAIN_TASK_STACK_SIZE but useless.

mahavirj, 2025-05-19:
@zzzh 

Could you share the log for `esptool.py image_info <file>` command here? Are you able to recreate the problem with any of the IDF provided examples? It is difficult to comment on your issue with the information you have provided.

zzzh, 2025-05-21:
Hi, @mahavirj 

I realized that it's not an esp-idf bug, thanks for the help.