espressif/esp-idf #15982: [Bug] DM9051 MAC driver missing TCR register write, packets not transmitted

castle2370, 2025-05-18:
## Problem Description

While using the ESP-IDF EMAC driver for the DM9051 chip (`components/esp_eth/src/spi/dm9051/esp_eth_mac_dm9051.c`), I found a critical issue in the `transmit()` function.

After writing the TX buffer via `dm9051_memory_write()` and setting the TX length using `TXPLL` / `TXPLH`, the driver **does not set the TXREQ bit in the TCR register (`dm9051_register_write(TCR, 0x01)`), so the DM9051 does not actually transmit the packet**.

This results in:

- `esp_eth_transmit()` returning success
- Debug logs showing `send ok`
- But the packet never appears on the wire (confirmed via Wireshark)

## Steps to Reproduce

1. Use ESP32 + DM9051 over SPI (e.g., 8 MHz or 16 MHz)
2. Construct a valid Ethernet or ARP reply packet
3. Call `esp_eth_transmit()`
4. Observe that `dm9051_memory_write()` is called
5. But no packet is captured by Wireshark, and `arp -a` on the PC does not show the ESP32 MAC

## Root Cause

According to the DM9051 datasheet:

> "To initiate transmission, the host must write the TX length to TXPLL/H, write the TX data into FIFO, and **then set TXREQ (bit 0) in the TCR register**."

The current `transmit()` implementation is missing the last step.

## Suggested Fix

Add the following after `dm9051_memory_write()` in `dm9051_transmit()`:

```c
dm9051_register_write(emac, DM9051_TCR, 0x01);  // Set TXREQ to start transmission

castle2370, 2025-05-18:
Additional info:

- ESP-IDF version: v5.4.1 
- Chip: ESP32-S3
- DM9051 via SPI (tested with 8 MHz and 16 MHz)
- Development OS: Ubuntu 24.04.2  LTS
- Toolchain: xtensa-esp32s3-elf (ESP-IDF default)
- Flash method: UART (using `idf.py flash`)
- Monitor: `idf.py monitor`

kostaond, 2025-05-19:
@castle2370 thanks for reporting potential issue with the DM9051. However, the issue must be somewhere else... The DM9051 driver is regularly tested (using ESP32) and we haven't observed any issues with transmissions. In addition, auto-transmit is enabled. This feature allows the packet to be transmitted automatically once the length is set and the packet data is written into the chip memory. Therefore, there is no need to use the following command to initiate packet transmission. This comes directly from Davicom's driver update proposal. However, I understand your confusion. DM9051 datasheet is not that detailed in some matters.

Have you tried Ethernet basic example? Check the SPI once again, try to enable debug logs and observe if DM9051 is able to receive any data.

kostaond, 2026-04-07:
@atlaste could you please describe the reproducible steps in more details? Is it possible to reproduce with any of IDF Ethernet examples?

atlaste, 2026-10-06:
@kostaond I honestly wouldn't know. I added the DM9051 as part of FluidNC, which utilizes quite a bit of CPU. During testing, I regularly keep these devices on for hours, even days. The code isn't too complex I think; the whole Ethernet codethat I use for this is basically here: https://github.com/atlaste/FluidNC/blob/MyWorks2/FluidNC/src/WebUI/EthernetConfig.cpp ; the rest is just plain old LWIP