espressif/esp-idf #15983: ESP32-C5-DevKitC-1 串口下载固件失败。

lab37, 2025-05-18:
错误：esp32c5 Packet content transfer stopped (received 4072 bytes)
idf版本：5.4.1与master都试过

lab37, 2025-05-19:
> We want to know the efuse information:
> 
> ```
> espefuse.py summary --chip esp32c5 --port /dev/ttyUSB0
> ```

thank you 
this is the efuse info:
`
PS E:\code_src\esp32c5-nnnxxx-cap> espefuse.py summary --port COM90
espefuse.py v4.8.1
Connecting....
Detecting chip type... ESP32-C5

=== Run "summary" command ===
EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Config fuses:
WR_DIS (BLOCK0)                                    Disable programming of individual eFuses           = 0 R/W (0x00000000)
RD_DIS (BLOCK0)                                    Disable reading from BlOCK4-10                     = 0 R/W (0b0000000)
DIS_ICACHE (BLOCK0)                                Represents whether icache is disabled or enabled.\ = False R/W (0b0)
                                                   \ 1: disabled\\ 0: enabled\\
DIS_TWAI (BLOCK0)                                  Represents whether TWAI function is disabled or en = False R/W (0b0)
                                                   abled.\\ 1: disabled\\ 0: enabled\\
KM_DISABLE_DEPLOY_MODE (BLOCK0)                    Represents whether the deploy mode of key manager  = 0 R/W (0x0)
                                                   is disable or not. \\ 1: disabled \\ 0: enabled.\\
KM_RND_SWITCH_CYCLE (BLOCK0)                       Set the bits to control key manager random number  = 0 R/W (0b00)
                                                   switch cycle. 0: control by register. 1: 8 km clk
                                                   cycles. 2: 16 km cycles. 3: 32 km cycles
KM_DEPLOY_ONLY_ONCE (BLOCK0)                       Set each bit to control whether corresponding key  = 0 R/W (0x0)
                                                   can only be deployed once. 1 is true; 0 is false.
                                                   bit 0: ecsda; bit 1: xts; bit2: hmac; bit3: ds
DIS_DIRECT_BOOT (BLOCK0)                           Represents whether direct boot mode is disabled or = False R/W (0b0)
                                                    enabled.\\ 1: disabled\\ 0: enabled\\
UART_PRINT_CONTROL (BLOCK0)                        Set the default UARTboot message output mode       = Enable R/W (0b00)
HYS_EN_PAD (BLOCK0)                                Represents whether the hysteresis function of corr = False R/W (0b0)
                                                   esponding PAD is enabled.\\ 1: enabled\\ 0:disable
                                                   d\\
HUK_GEN_STATE (BLOCK0)                             Set the bits to control validation of HUK generate = 0 R/W (0b000000000)
                                                    mode.\\ Odd of 1 is invalid.\\ Even of 1 is valid
                                                   .\\
XTAL_48M_SEL (BLOCK0)                              Represents whether XTAL frequency is 48MHz or not. = 1 R/W (0b001)
                                                    If not; 40MHz XTAL will be used. If this field co
                                                   ntains Odd number bit 1: Enable 48MHz XTAL\ Even n
                                                   umber bit 1: Enable 40MHz XTAL
XTAL_48M_SEL_MODE (BLOCK0)                         Specify the XTAL frequency selection is decided by = True R/W (0b1)
                                                    eFuse or strapping-PAD-state. 1: eFuse\\ 0: strap
                                                   ping-PAD-state
ECC_FORCE_CONST_TIME (BLOCK0)                      Represents whether to force ecc to use const-time  = False R/W (0b0)
                                                   calculation mode. \\ 1: Enable. \\ 0: Disable
PSRAM_CAP (BLOCK1)                                 Psram capacity                                     = 1 R/W (0b001)
PSRAM_VENDOR (BLOCK1)                              Psram vendor                                       = 1 R/W (0b01)
TEMP (BLOCK1)                                      Temp (die embedded inside)                         = 0 R/W (0b00)
TRIM_N_BIAS (BLOCK1)                               PADC CAL N bias                                    = 5 R/W (0b00101)
TRIM_P_BIAS (BLOCK1)                               PADC CAL P bias                                    = 31 R/W (0b11111)
BLOCK_USR_DATA (BLOCK3)                            User data
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLOCK_SYS_DATA2 (BLOCK10)                          System data part 2 (reserved)
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Flash fuses:
FLASH_TPUW (BLOCK0)                                Represents the flash waiting time after power-up;  = 0 R/W (0x0)
                                                   in unit of ms. When the value less than 15; the wa
                                                   iting time is the programmed value. Otherwise; the
                                                    waiting time is 2 times the programmed value
FORCE_SEND_RESUME (BLOCK0)                         Represents whether ROM code is forced to send a re = False R/W (0b0)
                                                   sume command during SPI boot.\\ 1: forced\\ 0:not
                                                   forced\\
[... cut ...]
WDT_DELAY_SEL (BLOCK0)                             Represents the threshold level of the RTC watchdog = 0 R/W (0b00)
                                                    STG0 timeout.\\ 0: Original threshold configurati
                                                   on value of STG0 *2 \\1: Original threshold config
                                                   uration value of STG0 *4 \\2: Original threshold c
                                                   onfiguration value of STG0 *8 \\3: Original thresh
                                                   old configuration value of STG0 *16 \\`

suda-morris, 2025-05-19:
Please also use `idf.py diag` to create a diagnostic report. We need it to investigate the issue.

lab37, 2025-05-19:
> Please also use `idf.py diag` to create a diagnostic report. We need it to investigate the issue.
> 
> https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-guides/tools/idf-diag.html

> Please also use `idf.py diag` to create a diagnostic report. We need it to investigate the issue.
> 
> https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-guides/tools/idf-diag.html

this the diag file
`https://www.nnnxxx.cn/diag.7z`

suda-morris, 2025-05-19:
I see this `# CONFIG_ESPTOOLPY_NO_STUB is not set` in your project.

Flash stub is not supported on esp32-c5 yet: https://github.com/espressif/esp-idf/blob/master/Kconfig#L121

Also please don't use idf 5.4 to test esp32c5. Please try release/v5.5 or master branch. It will enable the `CONFIG_ESPTOOLPY_NO_STUB` automatically for you.

lab37, 2025-05-19:
> I see this `# CONFIG_ESPTOOLPY_NO_STUB is not set` in your project.
>
> Flash stub is not supported on esp32-c5 yet: https://github.com/espressif/esp-idf/blob/master/Kconfig#L121
>
> Also please don't use [...]

thank you 
I checked it
it works  well
It is the  time to  close  this  issu

closed 2025-05-19 by Alvin1Zhang