espressif/esp-idf #15984: Panic when use browser to connect the esp32 https server will trigger panic with MBEDTLS_SSL_PROTO_TLS1_3=y setting

AxelLin, 2025-05-19:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-3514-g719b1b2be0

### Espressif SoC revision.

ESP32-D0WDR2-V3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM-32E-N8R2 module on a customized board.

### Power Supply used.

USB

### What is the expected behavior?

connect to esp32's https server should work.

### What is the actual behavior?

Test on v5.5 branch:
The esp32 runs a https server, and use browser to connect the https server will trigger panic.
This issue only happens when MBEDTLS_SSL_PROTO_TLS1_3=y.
Hit below panic when MBEDTLS_DYNAMIC_BUFFER=y && MBEDTLS_SSL_PROTO_TLS1_3=y

Core  1 register dump:
PC      : 0x40158926  PS      : 0x00060a30  A0      : 0x8015c384  A1      : 0x3ffdc610
--- 0x40158926: ssl_parse_record_header at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_msg.c:3876
---  (inlined by) ssl_get_next_record at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_msg.c:4983
---  (inlined by) mbedtls_ssl_read_record at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_msg.c:4323

A2      : 0x3f92b554  A3      : 0x00000000  A4      : 0x9159015a  A5      : 0x00000000
A6      : 0x00006680  A7      : 0x367cd507  A8      : 0x00000005  A9      : 0x3ffdc5f0
A10     : 0x00000000  A11     : 0x00000005  A12     : 0x3ffe4e28  A13     : 0x3ffe4e20
A14     : 0x00000005  A15     : 0x3f92b554  SAR     : 0x00000001  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace: 0x40158923:0x3ffdc610 0x4015c381:0x3ffdc650 0x4015b739:0x3ffdc670 0x401597ae:0x3ffdc710 0x401597fb:0x3ffdc730 0x4015292e:0x3ffdc750 0x401529b3:0x3ffdc770 0x40151dad:0x3ffdc790 0x400f298b:0x3ffdc7b0 0x40156ea1:0x3ffdc7e0 0x40155cef:0x3ffdc810
--- 0x40158923: ssl_parse_record_header at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_msg.c:3863
---  (inlined by) ssl_get_next_record at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_msg.c:4983
---  (inlined by) mbedtls_ssl_read_record at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_msg.c:4323
--- 0x4015c381: mbedtls_ssl_tls13_fetch_handshake_msg at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls13_generic.c:64
--- 0x4015b739: ssl_tls13_process_client_hello at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls13_server.c:1959
---  (inlined by) mbedtls_ssl_tls13_handshake_server_step at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls13_server.c:3475
--- 0x401597ae: mbedtls_ssl_handshake_step at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:4618
---  (inlined by) mbedtls_ssl_handshake_step at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:4558
--- 0x401597fb: mbedtls_ssl_handshake at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:4670
---  (inlined by) mbedtls_ssl_handshake at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:4647
--- 0x4015292e: esp_mbedtls_server_session_continue_async at /home/axel/esp/esp-idf/components/esp-tls/esp_tls_mbedtls.c:1082
--- 0x401529b3: esp_mbedtls_server_session_create at /home/axel/esp/esp-idf/components/esp-tls/esp_tls_mbedtls.c:1037
--- 0x40151dad: esp_tls_server_session_create at /home/axel/esp/esp-idf/components/esp-tls/esp_tls.c:734
--- 0x400f298b: httpd_ssl_open at /home/axel/esp/esp-idf/components/esp_https_server/src/https_server.c:178
--- 0x40156ea1: httpd_sess_new at /home/axel/esp/esp-idf/components/esp_http_server/src/httpd_sess.c:216
--- 0x40155cef: httpd_accept_conn at /home/axel/esp/esp-idf/components/esp_http_server/src/httpd_main.c:124
---  (inlined by) httpd_server at /home/axel/esp/esp-idf/components/esp_http_server/src/httpd_main.c:323
---  (inlined by) httpd_thread at /home/axel/esp/esp-idf/components/esp_http_server/src/httpd_main.c:339

Hit below panic when MBEDTLS_DYNAMIC_BUFFER=n && MBEDTLS_SSL_PROTO_TLS1_3=y
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4016f3d6  PS      : 0x00060030  A0      : 0x80169a6c  A1      : 0x3ffdc460
--- 0x4016f3d6: mbedtls_rsa_set_padding at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/rsa.c:998

A2      : 0x00000001  A3      : 0x00000001  A4      : 0x00000009  A5      : 0x3ffdd780
A6      : 0x00000040  A7      : 0x3ffdc470  A8      : 0x8016f3e6  A9      : 0x3ffdc440
A10     : 0x3f4473d8  A11     : 0x00000040  A12     : 0x0000003e  A13     : 0x00000001
A14     : 0x00000000  A15     : 0x00000004  SAR     : 0x0000001c  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000071  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace: 0x4016f3d3:0x3ffdc460 0x40169a69:0x3ffdc480 0x4016004b:0x3ffdc4c0 0x4016015f:0x3ffdc640 0x4015e173:0x3ffdc670 0x40159732:0x3ffdc710 0x4015977f:0x3ffdc730 0x401528b2:0x3ffdc750 0x40152937:0x3ffdc770 0x40151d69:0x3ffdc790 0x400f298b:0x3ffdc7b0 0x40156e25:0x3ffdc7e0 0x40155c73:0x3ffdc810
--- 0x4016f3d3: mbedtls_rsa_set_padding at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/rsa.c:989
--- 0x40169a69: mbedtls_pk_sign_ext at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/pk.c:1351
--- 0x4016004b: ssl_tls13_write_certificate_verify_body at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls13_generic.c:988
--- 0x4016015f: mbedtls_ssl_tls13_write_certificate_verify at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls13_generic.c:1037
--- 0x4015e173: ssl_tls13_write_certificate_verify at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls13_server.c:2818
---  (inlined by) mbedtls_ssl_tls13_handshake_server_step at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls13_server.c:3511
--- 0x40159732: mbedtls_ssl_handshake_step at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:4618
---  (inlined by) mbedtls_ssl_handshake_step at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:4558
--- 0x4015977f: mbedtls_ssl_handshake at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:4670
---  (inlined by) mbedtls_ssl_handshake at /home/axel/esp/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:4647
--- 0x401528b2: esp_mbedtls_server_session_continue_async at /home/axel/esp/esp-idf/components/esp-tls/esp_tls_mbedtls.c:1082
--- 0x40152937: esp_mbedtls_server_session_create at /home/axel/esp/esp-idf/components/esp-tls/esp_tls_mbedtls.c:1037
--- 0x40151d69: esp_tls_server_session_create at /home/axel/esp/esp-idf/components/esp-tls/esp_tls.c:734
--- 0x400f298b: httpd_ssl_open at /home/axel/esp/esp-idf/components/esp_https_server/src/https_server.c:178
--- 0x40156e25: httpd_sess_new at /home/axel/esp/esp-idf/components/esp_http_server/src/httpd_sess.c:216
--- 0x40155c73: httpd_accept_conn at /home/axel/esp/esp-idf/components/esp_http_server/src/httpd_main.c:124
---  (inlined by) httpd_server at /home/axel/esp/esp-idf/components/esp_http_server/src/httpd_main.c:323
---  (inlined by) httpd_thread at /home/axel/esp/esp-idf/components/esp_http_server/src/httpd_main.c:339

### Steps to reproduce.

Just set MBEDTLS_SSL_PROTO_TLS1_3=y and running a https server on esp32, then use browser to connect the server will trigger the panic.

### Debug Logs.

```plain

```

closed 2025-06-01 by espressif-bot

AxelLin, 2025-06-17:
@Ashish285 Can you apply the fix to stable branches?

Ashish285, 2025-06-17:
Hi @AxelLin , the fix has been merged internally to v5.5 branch. This will be available with the next release. You can try it out by checking out to the current master branch. 

Thanks