espressif/esp-idf #15986: The file system of the SD card is not working properly

LiuShunhao, 2025-05-19:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-3510-gaaebc37467

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

After mounting the SD card, files can be opened, written to and saved normally on the SD card

### What is the actual behavior?

After successfully mounting the SD card: 
1. Run the official [demo](https://github.com/espressif/esp-idf/tree/master/examples/storage/fatfs/getting_started) Both opening and writing were successful, but when checking the SD card resources, no saved files were found. 
2. 2. Failed to open the existing files in the mount directory.

### Steps to reproduce.

See the detailed description above

### Build or installation Logs.

```plain

```

LiuShunhao, 2025-05-19:
The SD card is in FAT format

pacucha42, 2025-05-19:
Hi @LiuShunhao, 
the IDF examples usually work smoothly, hopefully it's just something small to fix.
I need a bit more of the context, please:
- how did you check the presence of the files? External reader (eg PC) or by opening expected filepaths from your application (ie possibly modified example code)?
- what SD card do you use? (type, capacity, etc)
- could you please share the application log here? I mean the command line output (copy&paste is fine). It is sufficient to grab the default logs for now (verbosity INFO), but it will be likely needed to rerun your app with DEBUG or VERBOSE loglevel again

Thank you

LiuShunhao, 2025-05-19:
Add some details:
The SD card used is in FAT format with a capacity of 16G.
2. When checking the contents of the card on the computer through a card reader, example.txt was not found.
3. A new.txt file was created on the computer using a card reader, inserted into the device, and the program was run to modify the content of the.txt file. The log display was also successful. Then, when checking the content of the card on the computer through the card reader, it was found that the.txt file still had the original content.
4. The log will be supplemented later

LiuShunhao, 2025-05-19:
This is the log of the [demo：](https://github.com/espressif/esp-idf/tree/master/examples/storage/fatfs/getting_started)
[log.txt](https://github.com/user-attachments/files/20281451/log.txt)

LiuShunhao, 2025-05-19:
[image]

pacucha42, 2025-05-19:
> 3. A new.txt file was created on the computer using a card reader, inserted into the device, and the program was run to modify the content of the.txt file. 

Could you share the code that handles the file operations? As it sounds like a code mistake, not a failure. For instance, using "w" flag on opening files discards the file context on each fopen() and starts writing from offset 0

Regarding the logs: I see nothing suspicious, just a sooth run. Please, set the loglevel to VERBOSE (idf.py menuconfig), recompile and run again. Then you can share the output in a file again, it will be way larger this time.

Thank you

LiuShunhao, 2025-05-19:
The code was tested using the official [demo,](https://github.com/espressif/esp-idf/tree/master/examples/storage/fatfs/getting_started) with only one line of printed code added on top of the original one

LiuShunhao, 2025-05-20:
After set the loglevel to VERBOSE:

[log_250520.txt](https://github.com/user-attachments/files/20320707/log_250520.txt)

LiuShunhao, 2025-05-20:
[image]

Code

[fatfs_getting_started_main.txt](https://github.com/user-attachments/files/20320755/fatfs_getting_started_main.txt)

eternal-echo, 2025-05-20:
@LiuShunhao

Hello, you're using the wrong example. The code you're running (`examples/storage/fatfs/getting_started`) is for SPI Flash storage, not SD cards.

For SD card over SPI, use the example in `examples/storage/sd_card/sdspi` instead.

The files you're trying to find aren't on the SD card because the example you're using stores them in the SPI Flash partition, not on the SD card.

LiuShunhao, 2025-05-20:
Hello, when I was running the demo you mentioned(examples/storage/sd_card/sdspi), an error occurred and the SD card initialization failed. The chip I was using is ESP32s3. The following is the log:

`PS F:\SdkProject\LX_ESP32\esp-idf\sdcard_spi_test>  & 'c:\Users\Administrator\.espressif\python_env\idf5.5_py3.11_env\Scripts\python.exe' 'C:\Users\Administrator\esp\master\esp-idf\tools\idf_monitor.py' -p COM10 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --make '''c:\Users\Administrator\.espressif\python_env\idf5.5_py3.11_env\Scripts\python.exe'' ''C:\Users\Administrator\esp\master\esp-idf\tools\idf.py''' --target esp32s3 'f:\SdkProject\LX_ESP32\esp-idf\sdcard_spi_test\build\sdcard_spi_test.elf'
--- Warning: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM10 instead...
--- esp-idf-monitor 1.6.2 on \\.\COM10 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x15d8
load:0x403c8700,len:0xd24
load:0x403cb700,len:0x2f84
entry 0x403c8924
I (24) boot: ESP-IDF v5.5-dev-3372-g38628f98b9 2nd stage bootloader
I (25) boot: compile time May 20 2025 13:57:37
I (25) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (29) boot: efuse block revision: v1.3    
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : DIO  
I (40) boot.esp32s3: SPI Flash Size : 2MB  
I (44) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00100000
I (77) boot: End of partition table
I (81) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0dc44h ( 56388) map
I (98) esp_image: segment 1: paddr=0001dc6c vaddr=3fc95000 size=023ach (  9132) load
I (100) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=2c118h (180504) map
I (135) esp_image: segment 3: paddr=0004c140 vaddr=3fc973ac size=007cch (  1996) load
I (135) esp_image: segment 4: paddr=0004c914 vaddr=40374000 size=10f10h ( 69392) load
I (154) esp_image: segment 5: paddr=0005d82c vaddr=600fe000 size=0001ch (    28) load
I (160) boot: Loaded app from partition at offset 0x10000
I (161) boot: Disabling RNG early entropy source...
I (171) cpu_start: Multicore app
I (180) cpu_start: Pro cpu start user code
I (180) cpu_start: cpu freq: 160000000 Hz
I (180) app_init: Application information:
I (181) app_init: Project name:     sdcard_spi_test
I (185) app_init: App version:      v5.5-dev-3510-gaaebc37467
I (191) app_init: Compile time:     May 20 2025 13:57:27
I (196) app_init: ELF file SHA256:  e987c5b0d...
I (200) app_init: ESP-IDF:          v5.5-dev-3372-g38628f98b9
I (205) efuse_init: Min chip rev:     v0.0
I (209) efuse_init: Max chip rev:     v0.99
I (213) efuse_init: Chip rev:         v0.2
I (217) heap_init: Initializing. RAM available for dynamic allocation:
I (223) heap_init: At 3FC984E8 len 00051228 (324 KiB): RAM
I (228) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (234) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (239) heap_init: At 600FE01C len 00001FCC (7 KiB): RTCRAM
I (245) spi_flash: detected chip: gd
I (247) spi_flash: flash io: dio
W (250) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (263) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (269) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (276) main_task: Started on CPU0
I (296) main_task: Calling app_main()
I (296) example: Initializing SD card
I (296) example: Using SPI peripheral
I (296) example: Mounting filesystem
E (316) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
E (316) example: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.
I (316) main_task: Returned from app_main()
`

LiuShunhao, 2025-05-20:
What configurations do I still need before running the program

haberturdeur, 2025-05-21:
Hi, this issue usually occurs when the connection between the card and the ESP has problems.

What ESP32S3 board are you using?
Are you using an on-board SDcard slot, or are you using external module and connecting it yourself?
Could you share your wiring setup, either a diagram or a photo?

Please check if the pins in menuconfig match the actual physical connections.

If you are using external module, please also check the connection, long wires and bad contacts may cause issues.

Thank you.

LiuShunhao, 2025-05-22:
> Hi, this issue usually occurs when the connection between the card and the ESP has problems.
>
> What ESP32S3 board are you using? Are you using an on-board SDcard slot, or are you using external [...]

This is the ESP32S3 board that I am using now。 It uses an on-board SDcard slot,Below is the picture of the board

[image]

LiuShunhao, 2025-05-22:
[image]

rrtandler, 2025-05-23:
@LiuShunhao ,
The log you have posted is assuming following pin assignments:
PIN 2 CLK
PIN 4 MOSI
PIN 5 MISO
PIN 8 CS

The board you are using - Korvo - has different pin assignment:
PIN 15 CLK
PIN 4 MOSI
PIN 7 MISO
PIN -- CS Note: the CS is always set to logical 1

See the ESP32S3 Korvo's [datasheet](https://dl.espressif.com/dl/schematics/SCH_ESP32-S3-Korvo-2_V3.1.2_20240116.pdf)

To change the pin to signal assignment please follow the README.MD section in the example.

Alvin1Zhang, 2025-12-02:
Thanks for reporting, will close due to inactivity, feel free to reopen.