espressif/esp-idf #15993: ESP32 Stuck After Enabling Flash Encryption (Secure Boot NOT Enabled)

Vaibhav0216, 2025-05-20:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi everyone,

I'm working on ESP32 with ESP-IDF v5.3 and have enabled Flash Encryption in development mode.

**What’s Working:**

- Flashing works fine (idf.py flash)
- Flash encryption efuse (FLASH_CRYPT_CNT = 1) is set
- Secure Boot V1 and V2 are not enabled (ABS_DONE_0 = 0, ABS_DONE_1 = 0)

**The Problem:**
After flashing, the serial monitor shows:
```
--- esp-idf-monitor 1.5.0 on \\.\COM5 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
```
But there is no output from the chip – not even bootloader logs.

**What I’ve Tried:**

- Verified efuses with espefuse.py summary
- Full flash erase + clean build + idf.py flash
- Different COM ports, USB cables, baud rates

**My Questions:**

1. Is this behavior expected when Flash Encryption is enabled but a wrong/unencrypted image is flashed?
2. Could the device hang due to a bad encryption key or mismatched flash image?
3. What is the best recovery strategy when Flash Encryption is enabled and device is unresponsive?

Any help would be greatly appreciated. Thanks!

AdityaHPatwardhan, 2025-05-26:
@Vaibhav0216,

Could you share the entire output of `idf.py monitor` ?
By default the chip should at least provide some ROM prints even if nothing is flashed. There might be a connection issue at your end. 
You may refer to instructions provided [here](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/establish-serial-connection.html) to re-check your serial connection.

AdityaHPatwardhan, 2025-05-27:
@Vaibhav0216 
Did you try manually resetting the chip manually by pushing the RST button on the devkit?
Ideally the chip should get reset automatically after `Hard resetting via RTS pin...`

There is usually a small jumper attached on the esp32 devkit which facilitates this software enabled reset. If that is removed then it may affect software reset as well.

Please follow instructions mentioned [here](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html)

Vaibhav0216, 2025-05-29:
@AdityaHPatwardhan , 
I am using esp32 wroom ( custom board ). This same condition occurs in different board when I first enabled the secure boot v2 and then if flash encrypt not happing automatically I done it by using command by first set FLASH_CRYPT_CNT to 1 and generated flash encryption key ( in .bin ) then burn at the BLOCK 1.

```

Flash fuses:
FLASH_CRYPT_CNT (BLOCK0)                           Flash encryption is enabled if this field has an o = 1 R/W (0b0000001)
                                                   dd number of bits set
FLASH_CRYPT_CONFIG (BLOCK0)                        Flash encryption config (key tweak bits)           = 15 R/W (0xf)

Security fuses:
UART_DOWNLOAD_DIS (BLOCK0)                         Disable UART download mode. Valid for ESP32 V3 and = False R/W (0b0)
                                                    newer; only
ABS_DONE_0 (BLOCK0)                                Secure boot V1 is enabled for bootloader image     = False R/W (0b0)
ABS_DONE_1 (BLOCK0)                                Secure boot V2 is enabled for bootloader image     = True R/W (0b1)
DISABLE_DL_ENCRYPT (BLOCK0)                        Disable flash encryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_DECRYPT (BLOCK0)                        Disable flash decryption in UART bootloader        = False R/W (0b0)
KEY_STATUS (BLOCK0)                                Usage of efuse block 3 (reserved)                  = False R/W (0b0)
SECURE_VERSION (BLOCK3)                            Secure version for anti-rollback                   = 0 R/W (0x00000000)
BLOCK1 (BLOCK1)                                    Flash encryption key
   = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/-
BLOCK2 (BLOCK2)                                    Security boot key
   = cd 1d a0 a7 8f fc c5 89 dd 9f ac 5d e3 b9 a6 36 00 19 8b 55 26 5b 3a 17 dc 3b c2 9e 60 10 9a c9 R/-
BLOCK3 (BLOCK3)                                    Variable Block 3
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

```

AdityaHPatwardhan, 2025-06-03:
I think the issue here is because of the custom board that you are using.
As mentioned earlier esp32 had two modes
1) Download mode - This is the mode where esp32 expects some data on console ( UART/USB). In this case the ROM of esp32 has opened up a command interface for external entity to interact with. This is used by flashing utilities such as esptool etc.
2) SPI Flash Boot mode- In this case, the esp32 directly boots up the firmware flashed on the device. It first checks for the bootloader present at 0x1000 (for esp32) and then goes ahead.

The selection of these modes is based on the `STRAPPING PINS` which are part of the esp32 SoC. Based on their state the esp32 would either go in on of the modes.

In your case the device is going to download mode by default, you need to modify the state of the strapping pins in order to change the respective mode.

You can refer to [Automatic Bootloader](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#automatic-bootloader) section to see how the automatic boot mode selection is done in the esp32 devkit. You need to implement same hardware at your end.

Vaibhav0216, 2025-06-10:
Hi @AdityaHPatwardhan ,
but other time before i enable the secure boot on this devices( enabled flash encryption already ) this problem not occurs.

AdityaHPatwardhan, 2025-06-10:
Is it possible to share a successfully auto-reset log here?
We may get some idea about what has changed. Ideally secure boot would affect boot mode if ROM DL mode is disabled ( In that case chip cant enter ROM DL mode, but this eFuse is not burned in your case)

AdityaHPatwardhan, 2025-06-10:
Not this log, It is already present above.
I would like to see the log where the device boots normally.
Maybe after enabling only flash encryption, or nothing is enabled

closed 2025-06-10 by Vaibhav0216