espressif/esp-idf #15994: _invalid_pc_placeholder at /home/axel/esp/esp-idf/components/xtensa/xtensa_vectors.S:2235

AxelLin, 2025-05-20:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-3514-g719b1b2be0

### Espressif SoC revision.

ESP32-D0WDR2-V3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM-32E-N8R2 module

### Power Supply used.

USB

### What is the expected behavior?

Should successfully boot without resetting a few times.

### What is the actual behavior?

I sometime got below error when test with v5.5 branch.
It looks like the device reset a few times and then successfully boot.
However, if I switch back to build with v5.4, no such issue.
This looks like a bug in v5.5 branch.

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 153911750, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:296
load:0x40078000,len:13500 
ho 0 tail 12 room 4
load:0x40080400,len:3196
--- 0x40080400: _invalid_pc_placeholder at /home/axel/esp/esp-idf/components/xtensa/xtensa_vectors.S:2235

entry 0x4008057c
--- 0x4008057c: call_start_cpu0 at /home/axel/esp/esp-idf/components/bootloader/subproject/main/bootloader_start.c:25

ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 153911750, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:296
load:0x40078000,len:13500
ho 0 tail 12 room 4
load:0x40080400,len:3196
--- 0x40080400: _invalid_pc_placeholder at /home/axel/esp/esp-idf/components/xtensa/xtensa_vectors.S:2235

csum err:0xdb!=0x8b
ets_main.c 384
ets Jul 29 2019 12:21:46

rst:0x7 (TG0WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 153911750, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:296
load:0x40078000,len:13500
ho 0 tail 12 room 4
load:0x40080400,len:3196
--- 0x40080400: _invalid_pc_placeholder at /home/axel/esp/esp-idf/components/xtensa/xtensa_vectors.S:2235

entry 0x4008057c
--- 0x4008057c: call_start_cpu0 at /home/axel/esp/esp-idf/components/bootloader/subproject/main/bootloader_start.c:25

I (610) cpu_start: Multicore app
D (610) cpu_start: Pro cpu up
D (610) cpu_start: Starting app cpu, entry point is 0x40080e24
--- 0x40080e24: call_start_cpu1 at /home/axel/esp/esp-idf/components/esp_system/port/cpu_start.c:196

D (0) cpu_start: App cpu up
D (703) clk: RTC_SLOW_CLK calibration value: 15594575
```

### Steps to reproduce.

idf.py monitor will show the symptom.

### Debug Logs.

```plain

```

SoucheSouche, 2025-05-21:
Hi @AxelLin, I couldn't reproduce the issue. I have asked another colleague to try  and reproduce it as well without success. Could you try and give us more information on how to potentially reproduce the issue ?

SoucheSouche, 2025-05-21:
@AxelLin, maybe if you could give us the exact commit where the issue starts happening for you that could help us with finding a way to reproduce it locally.

Also, have you tried to reproduce the issue on several esp32s? Could it be possible that it happens only on the devkit you were using when the problem occurred?

AxelLin, 2025-05-21:
FYI, someone also reported this on https://esp32.com/viewtopic.php?t=44433
This error occurred in esp-idf-v5.5-dev.
In the stable version esp-idf-v5.4 - everything is ok.

SoucheSouche, 2025-05-22:
Hello @AxelLin,

so I checked the forum. The original post was concerning the "--- 0x40080400: _invalid_pc_placeholder at ..." and has been answered correctly in my opinion, this is just a result of the monitoring tool trying to decode addresses as PC values when it sees one.

Concerning your issue with the chip resetting several before successfully booting, as I mentioned, we couldn't reproduce it so could you please provide a more consistent way to reproduce, the expect commit that introduced the issue. And could you also try several esp32 devkit to make sure this is not coming from a damaged HW?

AxelLin, 2025-05-22:
> Hello [@AxelLin](https://github.com/AxelLin),
>
> so I checked the forum. The original post was concerning the "--- 0x40080400: _invalid_pc_placeholder at ..." and has been answered correctly in my [...]

OK.

> Concerning your issue with the chip resetting several before successfully booting, as I mentioned, we couldn't reproduce it so could you please provide a more consistent way to reproduce, the expect [...]

I cannot reproduce the below error and chip resetting a few times before successfully booting now:
csum err:0xdb!=0x8b

Not sure if it is fixed by below commands:
$ python -m esptool --port /dev/ttyUSB0 write_flash_status --non-volatile 0
I run above because I hit "MD5 of file does not match data in flash".
I will check if can hit the issue.

closed 2025-05-23 by AxelLin

MarshTheBacca, 2025-07-03:
I had the same issue on ESP-IDF 5.5 during UART initialisation. I solved it by changing my `uart_port_t` to `UART_NUM_2` instead of `UART_PORT_0`. I believe the monitor was already occupying `UART_PORT_0`.

Be-ing, 2025-08-08:
I am also encountering this error with an infinite bootloop since updating to ESP-IDF 5.5. The full backtrace is:

```
ELF file SHA256: 7f1937cc6

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:5536
load:0x40078000,len:16552
load:0x40080400,len:3780
--- 0x40080400: _invalid_pc_placeholder at /home/be/sw/tangara-fw/lib/esp-idf/components/xtensa/xtensa_vectors.S:2235
entry 0x400805e0
--- 0x400805e0: call_start_cpu0 at /home/be/sw/tangara-fw/lib/esp-idf/components/bootloader/subproject/main/bootloader_start.c:25
I (40) quad_psram: This chip is ESP32-D0WD
I (40) esp_psram: Found 8MB PSRAM device
I (41) esp_psram: Speed: 80MHz
I (42) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (50) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
I (59) cpu_start: Multicore app
I (74) cpu_start: Pro cpu start user code
I (74) cpu_start: cpu freq: 240000000 Hz
I (74) app_init: Application information:
I (79) app_init: Project name:     tangara
I (83) app_init: App version:      1.4.0
I (88) app_init: Compile time:     Aug  8 2025 10:58:37
I (94) app_init: ELF file SHA256:  7f1937cc6...
I (99) app_init: ESP-IDF:          v5.5
I (104) efuse_init: Min chip rev:     v3.0
I (108) efuse_init: Max chip rev:     v3.99 
I (113) efuse_init: Chip rev:         v3.1
I (118) heap_init: Initializing. RAM available for dynamic allocation:
I (125) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (131) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (138) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (144) heap_init: At 3FFCE380 len 00011C80 (71 KiB): DRAM
I (150) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (156) heap_init: At 3FFE4350 len 00019FC0 (103 KiB): D/IRAM
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4009b107  PS      : 0x00060d30  A0      : 0x8009b2c7  A1      : 0x3ffe3520  
--- 0x4009b107: multi_heap_internal_lock at /home/be/sw/tangara-fw/lib/esp-idf/components/heap/multi_heap.c:171
A2      : 0x00000000  A3      : 0x00060d23  A4      : 0x00060d20  A5      : 0x00060d23  
A6      : 0x00000001  A7      : 0x0000cdcd  A8      : 0x3ffc513c  A9      : 0x00000001  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0xb33fffff  A13     : 0x3ffe3c58  
A14     : 0x00018f01  A15     : 0x00000000  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
--- 0x4000c2e0: memcpy in ROM
--- 0x4000c2f6: memcpy in ROM

Backtrace: 0x4009b104:0x3ffe3520 0x4009b2c4:0x3ffe3540 0x40082e8e:0x3ffe3560 0x400830a2:0x3ffe35b0 0x400d4798:0x3ffe35d0 0x400d47b3:0x3ffe3bc0 0x400d33c0:0x3ffe3be0 0x400d33fd:0x3ffe3c10 0x40081e89:0x3ffe3c40 0x40079886:0x3ffe3c90 |<-CORRUPTED
--- 0x4009b104: multi_heap_internal_lock at /home/be/sw/tangara-fw/lib/esp-idf/components/heap/multi_heap.c:170
--- 0x4009b2c4: multi_heap_malloc at /home/be/sw/tangara-fw/lib/esp-idf/components/heap/multi_heap_poisoning.c:253
--- (inlined by) multi_heap_malloc at /home/be/sw/tangara-fw/lib/esp-idf/components/heap/multi_heap_poisoning.c:243
--- 0x40082e8e: create_new_task_stats_entry at /home/be/sw/tangara-fw/lib/esp-idf/components/heap/heap_task_info.c:164
--- 0x400830a2: heap_caps_update_per_task_info_alloc at /home/be/sw/tangara-fw/lib/esp-idf/components/heap/heap_task_info.c:319
--- 0x400d4798: heap_caps_init at /home/be/sw/tangara-fw/lib/esp-idf/components/heap/heap_caps_init.c:211
--- 0x400d47b3: __esp_system_init_fn_init_heap at /home/be/sw/tangara-fw/lib/esp-idf/components/heap/heap_caps_init.c:27
--- 0x400d33c0: do_system_init_fn at /home/be/sw/tangara-fw/lib/esp-idf/components/esp_system/startup.c:132
--- 0x400d33fd: do_core_init at /home/be/sw/tangara-fw/lib/esp-idf/components/esp_system/startup.c:170
--- (inlined by) start_cpu0_default at /home/be/sw/tangara-fw/lib/esp-idf/components/esp_system/startup.c:203
--- 0x40081e89: call_start_cpu0 at /home/be/sw/tangara-fw/lib/esp-idf/components/esp_system/port/cpu_start.c:836
--- 0x40079886: set_cache_and_start_app at /home/be/sw/tangara-fw/lib/esp-idf/components/bootloader_support/src/bootloader_utility.c:1155
--- (inlined by) unpack_load_app at /home/be/sw/tangara-fw/lib/esp-idf/components/bootloader_support/src/bootloader_utility.c:890
--- (inlined by) load_image at /home/be/sw/tangara-fw/lib/esp-idf/components/bootloader_support/src/bootloader_utility.c:800

ELF file SHA256: 7f1937cc6
```

Fjuzz-MatejL, 2025-08-12:
Hi all this error is still there in 5.5 stable version
is there any possible workaround to delete it from printing from bootloader?

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:2, clock div:2
secure boot v2 enabled
secure boot verification succeeded
load:0x3fff00c0 len:0x9e4
load:0x40078000 len:0x4d0c
load:0x40080400 len:0xbb4
--- 0x40080400: _invalid_pc_placeholder at PATH