espressif/esp-idf #15995: esp_gpio_is_reserved() behaviour

vvb333007, 2025-05-20:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Version I use is v5.4.1-1-g2f7dcd862a-dirty.

Whenever we install new UART or I2C drivers, corresponding GPIO's are marked as "RESERVED". Deinstallation of the driver does not clear reserved flag on GPIOs

nopnop2002, 2025-05-21:
Won't that port be marked as "RESERVED" with or without the UART or I2C drivers installed?

suda-morris, 2025-05-21:
I see this call chain `uart_driver_delete`->`uart_release_pin`->`esp_gpio_revoke`. So the GPIO used by UART(and reserved) can be release by `uart_driver_delete`. @vvb333007

vvb333007, 2025-05-21:
Now I realized that it is probably Arduino bug, not IDF :-/. Anyway:

I wrote a simple sketch, using an Arduino framework, It just displays pin 16 state: reserved or not.

Sketch outputs:
....
Pin 16 is not reserved
Pin 16 is not reserved
Pin 16 is not reserved
...

Next step is that I initialize UART2 on pins 16 and 17 (IO_MUX pins), by calling uartBegin():

......
[ 13259][V][esp32-hal-uart.c:552] uartBegin(): UART2 baud(115200) Mode(8000001c) rxPin(16) txPin(17)
[ 13269][V][esp32-hal-uart.c:650] uartBegin(): UART2 not installed. Starting installation
[ 13277][V][esp32-hal-uart.c:660] uartBegin(): UART2 RX FIFO full threshold set to 112 (value requested: 112 || FIFO Max = 128)
[ 13289][V][esp32-hal-uart.c:681] uartBegin(): Setting UART2 to use REF_TICK clock
[ 13298][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 16 successfully set to type UART_RX (2) with bus 0x3ffbdd18
[ 13309][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 17 successfully set to type UART_TX (3) with bus 0x3ffbdd18
[ 13320][V][esp32-hal-uart.c:732] uartBegin(): UART2 initialization done.

Sketch outputs:
....
Pin 16 is reserved
Pin 16 is reserved
Pin 16 is reserved
Pin 16 is reserved
....

Shutting down UART2 via uartEnd():

[ 19051][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 16 successfully set to type INIT (0) with bus 0x0
[ 19063][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 17 successfully set to type INIT (0) with bus 0x0

Sketch outputs:

Pin 16 is reserved
Pin 16 is reserved
Pin 16 is reserved
Pin 16 is reserved

The sketch itself is:

#include <Arduino.h>
#include "espshell.h"

#if __has_include("esp_private/esp_gpio_reserve.h")
#  include "esp_private/esp_gpio_reserve.h"
#elif __has_include("esp_gpio_reserve.h")
#  include "esp_gpio_reserve.h"
#else
#  warning "esp_gpio_reserve.h is not found, lets see if it will compile at all"
#endif

void setup() {
  Serial.begin(115200);
}

void loop() {
  while(1) {
    printf("Pin 16 is %sreserved\r\n",esp_gpio_is_reserved(1ULL << 16) ? "" : "not ");
    delay(1500);
  }
}

vvb333007, 2025-05-21:
In addition to the above: if i keep creating/deleting UART2 on every pin available they all eventually become reserved.

vvb333007, 2025-05-21:
May be I should close this issue, and I will reopen it in arduino-esp32 as it seems to be Anrduino Core feature

> You can check the port status with gpio_dump_io_configuration().

Yeah thats also. This function calls esp_gpio_is_reserved() anyway.

> 
> ```
> void app_main()
> {
>     gpio_dump_io_configuration(stdout, SOC_GPIO_VALID_GPIO_MASK);
> }
> ```
> 
> Some ports are "RESERVED" from the start.

vvb333007, 2025-05-21:
Investigation shows, that uart_driver_delete() is called within uartEnd(). So it seems to be IDF bug, not Arduino specific

suda-morris, 2025-05-21:
It turns out, this commit just fixed it: https://github.com/espressif/esp-idf/commit/007a497483b60fda44c34ed02408668b9314f570

closed 2025-05-28 by espressif-bot