espressif/esp-idf #16000: About ESP32C5 ECO2

williamleeag, 2025-05-21:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, I want to know what the difference between the ESP32C5 ECO1 is and ECO2.

closed 2025-05-21 by ginkgm

williamleeag, 2025-05-22:
> Hi [@williamleeag](https://github.com/williamleeag)
> 
> ECO1 is engineering samples for early customers. Please contact your sales for more information about it. Thanks.

Thanks for your reply, I will ask the PCB factory to contact with you.

Lucia-Rubio-E, 2025-05-22:
I am currently working on a project that requires the ESP32-C5 with ECO2 (v1.1 or v1.2 silicon).
Could you please let me know where I can purchase modules or chips with this version?

Thanks.

ginkgm, 2025-05-22:
Hi @Lucia-Rubio-E , 

ESP32-C5-DevKitC-1-N8R4 with v1.0 chip now available at:

Taobao (China): https://item.taobao.com/item.htm?ft=t&id=846843964795
AliExpress (Overseas): https://es.aliexpress.com/item/3256808604473710.html?gatewayAdapt=glo2esp4itemAdapt
Amazon US: https://www.amazon.com/dp/B0F6YKGL75
Digikey: https://www.digikey.cn/en/products/detail/espressif-systems/ESP32-C5-DEVKITC-1-N8R4/26658349?s=N4IgTCBcDaIKIGUAKBmMBaAwgVnQEQFMA3AaQEsAXTdARnQDkAOAJQBYQBdAXyA
Mouser: https://www.mouser.com/ProductDetail/Espressif-Systems/ESP32-C5-DevKitC-1-N8R4?qs=sqEgtWRSLJ2%2Fdhsv380LjQ%3D%3D

But please know, there is no such v1.1/v1.2 silicon for now, maybe you mean the devkit version?

ginkgm, 2025-05-22:
Hi @williamleeag ,

v0.1 chip (aka ECO1) is engineering sample, sorry but I don't have any information to share in public.

However for v1.0 chip (aka ECO2) there are preview version for datasheet and TRM :

https://www.espressif.com/sites/default/files/documentation/esp32-c5-wroom-1_wroom-1u_datasheet_en.pdf
 
https://www.espressif.com/sites/default/files/documentation/esp32-c5_technical_reference_manual_en.pdf

williamleeag, 2025-05-23:
> Hi [@williamleeag](https://github.com/williamleeag) ,
>
> v0.1 chip (aka ECO1) is engineering sample, sorry but I don't have any information to share in public.
>
> However for v1.0 chip (aka ECO2) there [...]

Thanks for your reply.