espressif/esp-idf #16002: if() funktion in CMakeLists.txt changes the compilation order of the components.

Gemelon, 2025-05-21:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I have a list with all reqiered components in my project. Whith config,I like to switch some components on or off.

```
set (requires "esp_event")
list(APPEND requires "base")
if(DEFINED CONFIG_BLINK_ENABLED)
   list(APPEND requires "blink")
endif()
set_property(GLOBAL PROPERTY requires_property ${requires})

```

### What is the actual behavior?

Instead the ${requires} variable is empty wenn compiling the base componente. When I remove the IF statement the content of the ${requires} variabel is as expected.

### Steps to reproduce.

1. this compiles fine
```
set (requires "esp_event")
list(APPEND requires "base")
list(APPEND requires "blink")
set_property(GLOBAL PROPERTY requires_property ${requires})

```
2. This compiles with error regardless of whether the variable is set or not

```
set (requires "esp_event")
list(APPEND requires "base")
if(DEFINED CONFIG_BLINK_ENABLED)
   list(APPEND requires "blink")
endif()
set_property(GLOBAL PROPERTY requires_property ${requires})

```
3. I have inserted two messages, one in this CMakeLists.txt at the main section:

```
set (requires "esp_event")
list(APPEND requires "base")
if(DEFINED CONFIG_BLINK_ENABLED)
   message("Blink is enabled")
   list(APPEND requires "blink")
endif()
set_property(GLOBAL PROPERTY requires_property ${requires})

```
and one in the CMakeLists.txt in the base component:

`message("Requires: ${requires}")
`

What I found out is, without IF statement the output of the two messages is:
```
Blink is enabled
Requires: .....

```
with IF statement the output of the two messages is:
```
Requires:
Blink is enabled

```
It has changed the order of compilation and Requires is empty because blink is not compiled yet.

### Build or installation Logs.

```plain

```

igrr, 2025-05-21:
Hi @Gemelon, you are probably running into a limitation of the IDF build system that component dependencies (REQUIRES/PRIV_REQUIRES) can't themseleves depend on Kconfig options (see e.g. https://github.com/espressif/esp-idf/issues/11757#issuecomment-1610829219)

In the current build system, you can only use conditions which depend on IDF build properties such as `IDF_TARGET`, when deciding which components to depend on. You can also use `idf_component_optional_requires` function to attach another component as a dependency after calling idf_component_register — however there is a limitation that that other component has to already be included into the build (either as a dependency of another component, or by specifying it explicitly by setting the COMPONENTS variable at project level).

We are planning to change this in IDF 6.0 — in the new build system it will be possible to use Kconfig options to decide which components to buid.

igrr, 2025-05-21:
It's indeed not mentioned in the docs, but you can find many usages of it in esp-idf repository, for example:

https://github.com/espressif/esp-idf/blob/27d68f57e6bdd3842cd263585c2c352698a9eda2/components/esp_driver_cam/CMakeLists.txt#L38-L40

and here is the description:

https://github.com/espressif/esp-idf/blob/27d68f57e6bdd3842cd263585c2c352698a9eda2/tools/cmake/component.cmake#L602-L613

Gemelon, 2025-05-21:
Yes, but I have to use this function inside the file it consumet it.  That makes the whole idea absurd, becaus I want a list like all the excamle projects have with only one REQUIRES ${requires} in every component, so I don't need to copy all the entrys every time I create a new component. I have only to put REQUIRES ${requires} and all requiries are there. This works quite well so far, but some times I do need to switch off a componet like blink, when the board don't has a LED.