espressif/esp-idf #16004: TinyUSB backend breaks in ESP-IDF v5.4 (esp32sx/dcd_esp32sx.c: USB0 and macros undeclared)

Talha-DPL, 2025-05-21:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**Description:**
When migrating a project from ESP-IDF v5.1 to v5.4, the build fails due to multiple missing macros and registers (e.g., USB0, USB_OUT_EP_NUM, USB_DO_SNAK0_M, etc.) in the esp32sx/dcd_esp32sx.c backend of TinyUSB. It is throwing around 240 errors. I am only using CDC related to usb.

** Environment**
ESP-IDF version: v5.4.1
TinyUSB version: managed_components/espressif__tinyusb (custom or older)
Board: ESP32-S3

**Expected Behavior**
Project builds successfully with CDC class enabled and USB stack configured as it was building in v5.1.

peter-marcisovsky, 2025-05-22:
Hi @Talha-DPL 

Are you using IDF CLI (command line `idf.py build` .. ) to build, or are you building using an IDE, like Espressif VSCode extension or similar?

Regarding the information about the `espressif__tinyusb` version you provided:
> TinyUSB version: managed_components/espressif__tinyusb (custom or older)

Are you using a custom version of `espressif__tinyusb` ?

closed 2025-06-05 by peter-marcisovsky

peter-marcisovsky, 2025-06-05:
Hi @Talha-DPL happy to hear, that the issue has been resolved. Thanks for confirmation.