espressif/esp-idf #16005: USB Host: Some devices not generating input events

dventurb, 2025-05-21:
**Environment**

- Module or chip used: ESP32S3
- IDF version 5.2
- Build System: idf.py
- Operating System: Linux
- Power Supply: External 

**Problem Description**

I'm having issues with USB Host on the ESP32-S3, I'm trying to read input data from HID devices like Keyboards, RFID readers and QR code/Bar code readers. 
My code is based of [hid_host_example.c](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/host/hid/main/hid_host_example.c), I have tested 4 different USB HID devices, 2 of them work fine and I can read the input events. The other 2 get detected as KEYBOARD HID, but do not generate any input report events. 

All devices work fine on PC.

I use `lsusb` to see the info from the device and the only pattern between the devices that don't generate any input is that they both have `bInterval = 1` and others that work one have `bInterval = 10` and the other have `bInterval = 20`. 

Thank you!

TD;LR: The ESP32-S3 detects the HID device, but no input reports are ever received.

dventurb, 2025-05-21:
Hello @roma-jam, thanks for the answer. 
Here is the output for the 4 devices:

**Keyboard (work)**:

```shell
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 1.10
bDeviceClass 0x0
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 8
idVendor 0x1a2c
idProduct 0x2d23
bcdDevice 1.10
iManufacturer 1
iProduct 2
iSerialNumber 0
bNumConfigurations 1
<0x1b>[0;32mI (31774) CLASS: Getting config descriptor<0x1b>[0m
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 59
bNumInterfaces 2
bConfigurationValue 1
iConfiguration 0
bmAttributes 0xa0
bMaxPower 100mA
	*** Interface descriptor ***
	bLength 9
	bDescriptorType 4
	bInterfaceNumber 0
	bAlternateSetting 0
	bNumEndpoints 1
	bInterfaceClass 0x3
	bInterfaceSubClass 0x1
	bInterfaceProtocol 0x1
	iInterface 0
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x81	EP 1 IN
		bmAttributes 0x3	INT
		wMaxPacketSize 8
		bInterval 10
	*** Interface descriptor ***
	bLength 9
	bDescriptorType 4
	bInterfaceNumber 1
	bAlternateSetting 0
	bNumEndpoints 1
	bInterfaceClass 0x3
	bInterfaceSubClass 0x0
	bInterfaceProtocol 0x0
	iInterface 0
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x82	EP 2 IN
		bmAttributes 0x3	INT
		wMaxPacketSize 8
		bInterval 10
``` 

**HID RFID Reader (work) **:

```shell
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 1.10
bDeviceClass 0x0
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 8
idVendor 0x3eb
idProduct 0x123
bcdDevice 0.00
iManufacturer 1
iProduct 2
iSerialNumber 3
bNumConfigurations 1
<0x1b>[0;32mI (1004) CLASS: Getting config descriptor<0x1b>[0m
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 52
bNumInterfaces 2
bConfigurationVa<break>
<0x81>1
iConfiguration 0
bmAttributes 0x80
bMaxPower 100mA
	*** Interface descriptor ***
	bLength 9
	bDescriptorType 4
	bInterfaceNumber 0
	bAlternateSetting 0
	bNumEndpoints 1
	bInterfaceClass 0x3
	bInterfaceSubClass 0x1
	bInterface<break>
rotocol 0x1
	iInterface 0
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x81	EP 1 IN
		bmAttributes 0x3	INT
		wMaxPacketSize 8
		bInterval 20
	*** Interface descriptor ***
	bLength 9
	bDescriptopType 4
	bInterfaceNumber 1
	bAlternateSetting 0
	bNumEndpoints 0
	bInterfaceClass 0x3
	bInterfaceSubClass 0x0
	bInterfaceProtocol 0x0
	iInterface 0
``` 

**HID QR Code (not working)**:

```shell
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 1.10
bDeviceClass 0x0
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 64
idVendor 0x2010
idProduct 0x7638
bcdDevice 0.00
iManufacturer 1
iProduct 2
iSerialNumber 3
bNumConfigurations 1
<0x1b>[0;32mI (4204) CLASS: Getting config descriptor<0x1b>[0m
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 34
bNumInterfaces 1
bConfigurationValue 1
iConfiguration 0
bmAttributes 0x80
bMaxPower 400mA
	*** Interface descriptor ***
	bLength 9
	bDescriptorType 4
	bInterfaceNumber 0
	bAlternateSetting 0
	bNumEndpoints 1
	bInterfaceClass 0x3
	bInterfaceSubClass 0x1
	bInterfaceProtocol 0x1
	iInterface 0
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x83	EP 3 IN
		bmAttributes 0x3	INT
		wMaxPacketSize 8
		bInterval 1
``` 

**HID RFID Reader (not working)**:

```shell

*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 1.10
bDeviceClass 0x0
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 64
idVendor 0x1a86
idProduct 0xdd01
bcdDevice 0.00
iManufacturer 1
iProduct 2
iSerialNumber 3
bNumConfigurations 1
<0x1b>[0;32mI (20664) CLASS: Getting config descriptor<0x1b>[0m
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 41
bNumInterfaces 1
bConfigurationValue 1
iConfiguration 0
bmAttributes 0x80
bMaxPower 300mA
	*** Interface descriptor ***
	bLength 9
	bDescriptorType 4
	bInterfaceNumber 0
	bAlternateSetting 0
	bNumEndpoints 2
	bInterfaceClass 0x3
	bInterfaceSubClass 0x0
	bInterfaceProtocol 0x0
	iInterface 0
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x82	EP 2 IN
		bmAttributes 0x3	INT
		wMaxPacketSize 64
		bInterval 1
		*** Endpoint descriptor ***
		bLength 7
		bDescriptorType 5
		bEndpointAddress 0x2	EP 2 OUT
		bmAttributes 0x3	INT
		wMaxPacketSize 64
		bInterval 1
```

tore-espressif, 2025-05-23:
@dventurb Thanks for the report. Of course, we test with different USB devices, but I could not reproduce the error even with bInterval=1. Could you please test with latest esp-idf? Either from master branch or the latest 5.4.1 release?

roma-jam, 2025-05-23:
Hi @dventurb, 

thanks for the debug output. 
I can see that two devices which don't generate the input request a bit more current than a regular keyboard: bMaxPower 400mA and 300mA.

Just to eliminate the problem of the amount of current to power the optics of QR-reader or RF-part of the RFID reader, could you try to connect the devices via external hub with external power supply that allows to power the downstream ports with at least not less than 0.5A?

dventurb, 2025-05-23:
@roma-jam @tore-espressif thanks for the answer again, I tested with the latest esp-idf  version (v5.4.1) using the code  [hid_host_example.c](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/host/hid/main/hid_host_example.c) and I still face the same issue: two same devices still not working. 

For the current problem, I separate the D+/D- wires from the GND/Vcc wires in USB cable, the GND/Vcc goes direct to the PC (supply 500mA) and the D+/D- wires goes to ESP32-S3. I first tested with one device that already work, and still work fine, however when I connect the QR Code Reader it is detected as KEYBOARD but no input receive. 

[image]

[image]

Also I added printf in every function for debugging, and seems that the problem is no detect input events, the code flow doesn't reach `HID_HOST_INTERFACE_EVENT_INPUT_REPORT` .

Any tips? Is there another lib for this (maybe tinyusb) ? Thanks again!

roma-jam, 2025-05-23:
Hi @dventurb, 

thanks for additional information. 

There are two lines in the example, which set the protocol and set idle for the device. Could you comment them out and try to connect the devices which are not working again?

```
        ESP_ERROR_CHECK(hid_host_device_open(hid_device_handle, &dev_config));
        // Remove set boot protocol & idle=0 
        // if (HID_SUBCLASS_BOOT_INTERFACE == dev_params.sub_class) {
        //     ESP_ERROR_CHECK(hid_class_request_set_protocol(hid_device_handle, HID_REPORT_PROTOCOL_BOOT));
        //     if (HID_PROTOCOL_KEYBOARD == dev_params.proto) {
        //         ESP_ERROR_CHECK(hid_class_request_set_idle(hid_device_handle, 0, 0));
        //     }
        // }
        ESP_ERROR_CHECK(hid_host_device_start(hid_device_handle));
```

In this case, the logic which parses the report won't work, because it is only for boot protocol.

To print the raw data, you can add the line right after getting the report data: 

```
        ESP_ERROR_CHECK(hid_host_device_get_raw_input_report_data(hid_device_handle,
                                                                  data,
                                                                  64,
                                                                  &data_length));

        ESP_LOG_BUFFER_HEX(TAG, data, data_length); // Print the HEX data
```

Let me know if that helps.

roma-jam, 2025-05-26:
Hi, @dventurb, 

thanks for the feedback, apparently the only one thing that could help us is the usb bus data log.  

Could you please capture the usb traffic with this device (connected to a PC Host) with Wireshark and share the log here? 

Thanks.

dventurb, 2025-05-26:
Hi @roma-jam, thank you.

I've capture the USB traffic from the QR Code device using Wireshark connected to a PC. I'm kind of a newbie in wireshark, so I uploud the log file here:

```
1	0.000000	3.19.3	host	USB	72	URB_INTERRUPT in
2	0.000065	host	3.19.3	USB	64	URB_INTERRUPT in
3	0.001941	3.19.3	host	USB	72	URB_INTERRUPT in
4	0.002058	host	3.19.3	USB	64	URB_INTERRUPT in
5	0.002237	host	3.19.0	USB	65	URB_CONTROL out
6	0.002992	3.19.0	host	USB	64	URB_CONTROL out
7	0.003931	3.19.3	host	USB	72	URB_INTERRUPT in
8	0.003983	host	3.19.3	USB	64	URB_INTERRUPT in
9	0.005928	3.19.3	host	USB	72	URB_INTERRUPT in
10	0.005965	host	3.19.3	USB	64	URB_INTERRUPT in
11	0.007916	3.19.3	host	USB	72	URB_INTERRUPT in
12	0.007945	host	3.19.3	USB	64	URB_INTERRUPT in
13	0.009918	3.19.3	host	USB	72	URB_INTERRUPT in
14	0.009947	host	3.19.3	USB	64	URB_INTERRUPT in
15	0.011939	3.19.3	host	USB	72	URB_INTERRUPT in
16	0.011965	host	3.19.3	USB	64	URB_INTERRUPT in
17	0.013922	3.19.3	host	USB	72	URB_INTERRUPT in
18	0.013949	host	3.19.3	USB	64	URB_INTERRUPT in
19	0.015921	3.19.3	host	USB	72	URB_INTERRUPT in
20	0.015948	host	3.19.3	USB	64	URB_INTERRUPT in
21	0.017915	3.19.3	host	USB	72	URB_INTERRUPT in
22	0.017943	host	3.19.3	USB	64	URB_INTERRUPT in
23	0.019917	3.19.3	host	USB	72	URB_INTERRUPT in
24	0.019945	host	3.19.3	USB	64	URB_INTERRUPT in
25	0.021920	3.19.3	host	USB	72	URB_INTERRUPT in
26	0.021947	host	3.19.3	USB	64	URB_INTERRUPT in
27	0.023919	3.19.3	host	USB	72	URB_INTERRUPT in
28	0.023946	host	3.19.3	USB	64	URB_INTERRUPT in
29	0.025918	3.19.3	host	USB	72	URB_INTERRUPT in
30	0.025946	host	3.19.3	USB	64	URB_INTERRUPT in
31	0.027920	3.19.3	host	USB	72	URB_INTERRUPT in
32	0.027951	host	3.19.3	USB	64	URB_INTERRUPT in
33	0.029922	3.19.3	host	USB	72	URB_INTERRUPT in
34	0.029954	host	3.19.3	USB	64	URB_INTERRUPT in
35	0.031926	3.19.3	host	USB	72	URB_INTERRUPT in
36	0.031957	host	3.19.3	USB	64	URB_INTERRUPT in
37	0.033932	3.19.3	host	USB	72	URB_INTERRUPT in
38	0.033961	host	3.19.3	USB	64	URB_INTERRUPT in
39	0.035933	3.19.3	host	USB	72	URB_INTERRUPT in
40	0.035961	host	3.19.3	USB	64	URB_INTERRUPT in
41	0.037919	3.19.3	host	USB	72	URB_INTERRUPT in
42	0.037947	host	3.19.3	USB	64	URB_INTERRUPT in
43	0.039847	3.19.3	host	USB	72	URB_INTERRUPT in
44	0.039866	host	3.19.3	USB	64	URB_INTERRUPT in
45	0.041918	3.19.3	host	USB	72	URB_INTERRUPT in
46	0.041943	host	3.19.3	USB	64	URB_INTERRUPT in
47	0.043919	3.19.3	host	USB	72	URB_INTERRUPT in
48	0.043951	host	3.19.3	USB	64	URB_INTERRUPT in
49	0.045917	3.19.3	host	USB	72	URB_INTERRUPT in
50	0.045947	host	3.19.3	USB	64	URB_INTERRUPT in
51	0.047927	3.19.3	host	USB	72	URB_INTERRUPT in
52	0.047957	host	3.19.3	USB	64	URB_INTERRUPT in
53	0.049919	3.19.3	host	USB	72	URB_INTERRUPT in
54	0.049948	host	3.19.3	USB	64	URB_INTERRUPT in
55	0.051910	3.19.3	host	USB	72	URB_INTERRUPT in
56	0.051941	host	3.19.3	USB	64	URB_INTERRUPT in
57	0.053908	3.19.3	host	USB	72	URB_INTERRUPT in
58	0.053938	host	3.19.3	USB	64	URB_INTERRUPT in
59	0.055907	3.19.3	host	USB	72	URB_INTERRUPT in
60	0.055937	host	3.19.3	USB	64	URB_INTERRUPT in
61	0.057921	3.19.3	host	USB	72	URB_INTERRUPT in
62	0.057952	host	3.19.3	USB	64	URB_INTERRUPT in
63	0.059931	3.19.3	host	USB	72	URB_INTERRUPT in
64	0.059961	host	3.19.3	USB	64	URB_INTERRUPT in
65	0.061929	3.19.3	host	USB	72	URB_INTERRUPT in
66	0.061959	host	3.19.3	USB	64	URB_INTERRUPT in
67	0.062510	host	3.19.0	USB	65	URB_CONTROL out
68	0.062833	3.19.0	host	USB	64	URB_CONTROL out
``` 

[usb_host.pcapng.zip](https://github.com/user-attachments/files/20439762/usb_host.pcapng.zip)

Thanks again for the support.

roma-jam, 2025-05-26:
Hi @dventurb, 

thanks a lot for the efforts! 

Almost there, but I need the whole log, starting from the device connection and it's enumeration.

So, there is no need to apply filters or anything. Just launch the capture, then attach the device and scan the qr code. 

Also, you can just attach the *.zip file, no necessary to post the log in text. 

Thanks.

dventurb, 2025-05-26:
Hello @roma-jam, thanks for the info.

I've now capture the full USB traffic, starting when the device is connected and the QR Code scan. Let me know if it's everything right. 

[usb_host.pcapng.zip](https://github.com/user-attachments/files/20440238/usb_host.pcapng.zip)

Thanks.

roma-jam, 2025-05-26:
Hi @dventurb, 

Yes, thanks a lot. Much better, but unfortunately I can see only Integrated Hub and NuPhy AIR 75 HE Keyboard, connected to the 2nd port of this hub. 

Could you please double check, that you have only one usb controller or (in case you have two of them) you capture the traffic from the right one. 

You can try to verify it, finding the following lines in the log, such as: 

[image]

Where you can check the descriptor data in the window below, when you select the `Get DESCRIPTOR Response String` line:

[image]
 

I assume you should see there something, that is somehow related to the QR scanner...

Thanks a lot for you help and collaboration.

roma-jam, 2025-05-27:
Hi @dventurb, 

I don't see any specifics in the device configuration, but let's try to imitate the driver. Maybe there are some connection of the internal logic in the QR scanner. 

To do so, we need to add two more things:
- requesting the HID Report Descriptor (used for reports descriptions)
- send the output report, to synchronize LEDs states (used to indicate the Num, Caps and other locks)

Please, add these lines after device opening: 

```
        ESP_ERROR_CHECK(hid_host_device_open(hid_device_handle, &dev_config));
        if (HID_SUBCLASS_BOOT_INTERFACE == dev_params.sub_class) {
            if (HID_PROTOCOL_KEYBOARD == dev_params.proto) {
                ESP_ERROR_CHECK(hid_class_request_set_idle(hid_device_handle, 0, 0));
                // Get Report Descriptor
                size_t report_desc_len = 0;
                uint8_t *report_descriptor = hid_host_get_report_descriptor(hid_device_handle, &report_desc_len);
                ESP_LOGI(TAG, "Report Descriptor (%d bytes):", report_desc_len);
                ESP_LOG_BUFFER_HEX(TAG, report_descriptor, report_desc_len);
                // Set LEDs state
                uint8_t led_state[1] = { 0 };
                ESP_ERROR_CHECK(hid_class_request_set_report(hid_device_handle,
                                                                 HID_REPORT_TYPE_OUTPUT, 
                                                                 0, 
                                                                 led_state, 
                                                                 1));
            }
        }
        ESP_ERROR_CHECK(hid_host_device_start(hid_device_handle));
```

Let me know if that helps.

dventurb, 2025-05-27:
Hello @roma-jam, thanks again.

I've added that code that you suggested, the device is sucessfully recognized as keyboard and the report discriptor is printed correctly, however still not receiving input from it. 

This is from QR Code Reader:
[image]

This is from RFID Reader (working):
[image]

I also added a `ESP_LOG_BUFFER_HEX` in the begging of `hid_host_interface_callback` to see the input data, and this is what I'm getting repeatdly:

```c
void hid_host_interface_callback(hid_host_device_handle_t hid_device_handle,
                                 const hid_host_interface_event_t event,
                                 void *arg)
{
    uint8_t data[64] = { 0 };
    size_t data_length = 0;
    hid_host_dev_params_t dev_params;
    ESP_ERROR_CHECK(hid_host_device_get_params(hid_device_handle, &dev_params));

    ESP_ERROR_CHECK(hid_host_device_get_raw_input_report_data(hid_device_handle, data, 64, &data_length));
    ESP_LOG_BUFFER_HEX("DATA:", data, data_length);
    hid_host_keyboard_report_callback(data, data_length);
``` 

[image]

I don't think this is the correct data, any advice? Thanks again!

roma-jam, 2025-05-27:
Hi @dventurb, 

Just to be sure, what is done: 
1. SET_IDLE(0)
2. GET_REPORT_DESCRIPTOR()
3. SET_REPORT()

I didn't mentioned, that we need to keep `hid_class_request_set_idle`, but it is in the code.  Please, double check that you also have it in your code.  

Also, has something changed in the amount of working devices after adding steps 2. and 3.? (still 2/2?)

> and this is what I'm getting repeatdly

This can be when we omit the SET_IDLE(0). But usually, device just reporting zeroes, but in you case the device reporting KEY_PRESSED (0x39) and KEY_RELEASED (0x00).  

Is something happening with a device, while it continuously reporting the key scancode?

dventurb, 2025-05-29:
Hello, @roma-jam, thanks for the help.

Just to close that issue, I get two more QR code scanners, the same model to the other that don't work, to check if the issue was because the hardware of the device.

[image]

After testing all 3 devices using the code [hid_host_example.c](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/host/hid/main/hid_host_example.c), none of the 3 devices work. However, when I added the lines that you suggested to request the report descriptor, 2 of the 3 devices started working fine on ESP. I still don't understand why this step is required. 

```c
size_t report_desc_len = 0;
uint8_t *report_descriptor = hid_host_get_report_descriptor(hid_device_handle, &report_desc_len);
ESP_LOGI("HID", "Report Descriptor (%d bytes):", report_desc_len);
ESP_LOG_BUFFER_HEX("HID", report_descriptor, report_desc_len);            
``` 

The RFID reader started work fine too after adding the same report descriptor lines. However, I encountered another issue, the card number appears with SHIFT, `)*^(^$*$&(` instead of the real digits `0869648479`, as seen on PC. 
It's seems like PC may ignore the shift modifiers for RFID Readers, to solve this I read the `iProduct` and if it contained the word "rfid", I disable the shift for that device.

```c
hid_host_dev_info_t info;
hid_host_get_device_info(hid_device_handle, &info);
char utf8_buf[64];
 wchar_to_utf8(utf8_buf, 64, info.iProduct);
char *str = strstr(lowercase_string(utf8_buf), "rfid");
if(str != NULL){
    is_rfid = true;
}else {
    is_rfid = false;
}

if(is_rfid == false){
    shift = kb_report->modifier & (KEYBOARD_MODIFIER_LEFTSHIFT | KEYBOARD_MODIFIER_RIGHTSHIFT);
}else {
     shift = 0;
}
``` 

Maybe there is a reason for this, and perhaps there is better solution. I just wanted to say thank you for the support and for the solution.

closed 2025-05-29 by dventurb