espressif/esp-idf #16006: Bug Report: Missing `freertos` CMake dependencies in multiple core components

spicysoul, 2025-05-21:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

### Environment

- Chip: ESP32-P4
- ESP-IDF version: v5.4.1 (official release)
- OS: Windows 11 x64
- Toolchain: Xtensa or RISC-V GCC from ESP-IDF installer
- Project type: Custom application using esp_timer / esp_event / freertos

---

### Bug Report: Missing `freertos` CMake dependencies in multiple core components

When building a minimal application using `esp_event`, `esp_timer`, and `freertos`, the following error appears:

fatal error: freertos/FreeRTOS.h: No such file or directory

The root cause is that the following components include `freertos` headers, but **do not declare `REQUIRES freertos` in their `CMakeLists.txt`**:

- `esp_timer`
- `esp_event`
- `esp_pm`
- `esp_mm`
- `esp_eth` (transitively through `esp_driver_gpio` → `hal/eth_types.h`)

---

### Reproduction Steps

1. Create a new project with a single file `main.c`:
```c
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "esp_timer.h"
#include "esp_event.h"

void app_main(void) {
    ESP_LOGI("APP", "P4 test running");
}

2. Use this CMakeLists.txt

idf_component_register(SRCS "main.c"
                       INCLUDE_DIRS "."
                       PRIV_REQUIRES
                       freertos
                       esp_timer
                       esp_event
                       esp_system
                       esp_pm
                       esp_mm)

3. Run:

idf.py set-target esp32p4
idf.py fullclean
idf.py build

Expected Behavior
The project builds successfully since all required components are present and explicitly listed in PRIV_REQUIRES.

### What is the actual behavior?

Build fails with : 
fatal error: freertos/FreeRTOS.h: No such file or directory

Even though freertos is part of IDF, CMake doesn't register the include path properly because internal REQUIRES are missing.

Suggestion

Please review and correct internal CMake dependencies in the following components:

esp_event: add REQUIRES freertos esp_system

esp_timer: add REQUIRES freertos esp_system

esp_mm: add REQUIRES freertos

esp_eth: ensure proper REQUIRES hal path for eth_types.h

Additional Notes
This issue only affects ESP32-P4 + ESP-IDF v5.4.1 because these components depend on freertos and esp_system but do not explicitly declare those dependencies in their CMake config.

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain

```

igrr, 2025-05-21:
@spicysoul Could you please attach the complete project to reproduce the issue?

Generally the dependency on freertos doesn't have to be specified since freertos is in _common requirements_ — dependencies attached to every component in the build:

https://github.com/espressif/esp-idf/blob/27d68f57e6bdd3842cd263585c2c352698a9eda2/tools/cmake/build.cmake#L278-L279

therefore the component doesn't have to be mentioned in the requirements list explicitly, unless you somehow modify `__COMPONENT_REQUIRES_COMMON` at the project level.

spicysoul, 2025-05-21:
Thank you for your response.

Here is the complete minimal reproducible example as requested. It’s based on the standard ESP-IDF 5.4.1 environment targeting ESP32-P4.

- Attached: full project zip (esp_hosted_slave_patched.zip)
- Included: sdkconfig.defaults

I did not modify `__COMPONENT_REQUIRES_COMMON`, but I still get this error:

 -- > fatal error: freertos/FreeRTOS.h: No such file or directory

This happens when using `esp_event`, `esp_timer`, or `esp_pm` without explicitly declaring `REQUIRES freertos`, despite freertos being listed in the common requirements.

Let me know if you need the full `build/` diagnostic archive or want me to reproduce this with verbose flags.

Thanks again!

[esp_hosted_slave_patched.zip](https://github.com/user-attachments/files/20375868/esp_hosted_slave_patched.zip)

o-marshmallow, 2025-05-22:
Hello @spicysoul ,

Thanks for the details and for the archive, I tried your project on my side but was unable to reproduce your issue, I am also using `v5.4.1`, I used the following to clean and rebuild:
```
rm -rf build
idf.py set-target esp32p4
idf.py build
```

Please check your installation of ESP-IDF, make sure the git submodules are updated:
```
git submodule update --recursive
```

spicysoul, 2025-05-22:
> Hello [@spicysoul](https://github.com/spicysoul) ,
>
> Thanks for the details and for the archive, I tried your project on my side but was unable to reproduce your issue, I am also using `v5.4.1`, I [...]

oh, so sorry... i think this is my fault..

i tried another PC in my office. 

theres nothing problem it.  

so sorry. everybody. :(

o-marshmallow, 2025-05-22:
@spicysoul No problem, thanks for coming back with the confirmation 😄