espressif/esp-idf #16008: Heap overflow detection improvement

bryghtlabs-richard, 2025-05-21:
### Is your feature request related to a problem?

Today, I found a piece of code had corrupted a heap-buffer. (This is problem is mine alone)

However, I was surprised to find that it overwrote it by 1.7MB before a thread on the other core noticed.

From the hardware available, it should be possible to make the heap much more aggressive at detecting these, and sometimes detect use-after-free as well.

### Describe the solution you'd like.

I'd like to suggest that the ESP-HEAP treat physical memory as an array of MMU pages, and that access to those MMU pages be enabled/disabled as needed.

Ex:

- On heap creation, access to the first page of the heap is enabled(and will stay enabled forever). Other pages will be disabled.
- Once an allocation extends to the next page(s), access to those pages will be enabled.
- When allocations are freed, if the coalesced area covers a whole page, access to that page will be disabled.
- Access to a disabled page will fault and trap the error.

### Describe alternatives you've considered.

I won't write any more bugs ever again :)

igrr, 2025-05-21:
One thing to note is that ESP chips don't have synchronous MMU traps implemented in hardware until ESP32-P4, and even there they are disabled by default for performance reasons. There is also only an MMU for Flash and PSRAM access, internal RAM and the peripherals are accessed directly.

ESP-IDF will implement on-demand paging once there is hardware available to support this (e.g. implementing RISC-V Sv32 MMU).

o-marshmallow, 2025-05-22:
Hello @bryghtlabs-richard ,

Thanks for your suggestion, as @igrr  said, we don't have a hardware MMU for the internal RAM.

Regarding your issue, in ESP-IDF we provide a heap corruption detection feature that can be enabled via the `menuconfig`: `Component config -> Heap memory debugging -> Heap corruption detection`. You can either enable light poisoning (`CONFIG_HEAP_POISONING_LIGHT`) or comprehensive (`CONFIG_HEAP_POISONING_COMPREHENSIVE`).

This feature is documented here: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/heap_debug.html#heap-corruption-detection

igrr, 2025-05-26:
> to run it quite fast and track head and tails to see why its overwrite and who the offender is

If I understood your suggestion correctly:
- Overriding allocation functions is already possible. That's how tracking malloc/free/etc. currently works.
- It's not necessary to override allocation functions to check the poisoned bytes — this is already done by default if comprehensive heap corruption detection is enabled.
- Regarding identifying the "offender":
   - If the out-of-bounds write overflows the buffer just a bit (e.g. by overwriting the byte immediately after the allocated block), we can assume that the code related to that buffer is responsible. If heap allocation tracing is enabled, we do in theory have the information about the code where each block was allocated from. However presently, IDF's heap component doesn't connect these two features — poisoning and tracing. This is indeed a useful, if complex, improvement we could do.
  - If the out-of-bounds write overflows the allocated buffer by more than the size of the poisoning region, only instrumentation techniques such as [kASAN](https://github.com/androidoffsec/baremetal_kasan) can help — at the expense of code size and RAM usage. This would also be a useful improvement, however the code size increase may be prohibitive for many applications. Also there is an issue with recompiling the libraries (libc, Wi-Fi and Bluetooth pre-compiled libraries) with instrumentation options enabled.
  - If the out-of-bounds write overflows the allocated buffer by a lot, then the suggestion raised in the OP (protecting whole MMU pages) can help generate a trap which will directly point to the offending code.

o-marshmallow, 2025-08-08:
Thanks for the feedback, I am closing this issue since this is not something we can do right now with our current hardware.

closed 2025-08-08 by o-marshmallow

bryghtlabs-richard, 2025-08-08:
Thank you for your consideration 👍