espressif/esp-idf #16012: Increase in Total image size proportional to growing libbtdm_app.a after upgrading ESP-IDF

kobe-dace, 2025-05-23:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi everyone,

After upgrading the ESP-IDF used in a Bluetooth (NimBLE) project for **ESP32-C3**, I noticed that the reported Total image size became significantly larger than before.

### What I’ve tried

1. Started from the official example examples/bluetooth/nimble/bleprph.
2. Added the line
    ```
    CONFIG_IDF_TARGET="esp32c3"
    ```
   to the example’s **sdkconfig** so the test is locked to ESP32-C3.
3. Built the same example on the following ESP-IDF tags: **v5.0.0 → v5.2.5**.
4. Recorded the Total image size reported by the build system for each version.

```mermaid
---
config:
    xyChart:
        width: 1000
        height: 600
    themeVariables:
        xyChart:
          plotColorPalette: "#ff0000"
---
xychart-beta
    title "Total image sizes"
    x-axis [v50, v501, v502, v503, v504, v505, v506, v507, v508, v51, v511, v512, v513, v514, v515, v516, v52, v521, v522, v523, v524, v525]
    y-axis "Size" 530000 --> 610000

    line [546106, 551748, 553620, 557374, 560728, 571998, 572984, 588358, 585172, 565152, 566690, 574562, 584242, 591456, 609714, 598026, 591676, 597348, 600910, 610666, 599684, 599684]
```

While all sections grow slightly, the **size-components** output shows that the `libbtdm_app.a` **flash_total** field increases the most—and appears to account for nearly all of the overall growth.

### Questions

1. As ESP-IDF versions advance, is the increase in Total image size caused by the expanding `libbtdm_app.a` essentially unavoidable?
2. Besides the optimizations listed in the Size Optimization Guide (https://docs.espressif.com/projects/esp-idf/en/v5.4.1/esp32/api-guides/performance/size.html), are there any additional, Bluetooth-specific techniques to keep the image size under control?

Any help would be greatly appreciated. Thanks!

esp-cjh, 2025-05-23:
Hi, @kobe-dace 
We have already optimized this issue. In the future, it will be possible to reduce the RAM and Flash size of `libbtdm_app.a` by disabling certain configuration options. 
Currently, this merge request has not yet been integrated into the IDF on GitHub. Once it is merged, I will provide the commit.

kobe-dace, 2025-05-23:
That’s fantastic news—thanks for tackling this optimization! I’m happy to hear RAM / Flash usage will be tunable soon. When the merge lands, please drop a note here; I’ll gladly pull the commit, test it, and share feedback. Keep up the great work, and thanks again!

esp-cjh, 2025-05-28:
Hi, @kobe-dace 
The change has been merged to github. This is the commit 1d739291ca4bdbd577b58c41244dd53d3d4c3548.
For examples/bluetooth/nimble/bleprph, You can configure the controller as follows to reduce RAM/Flash usage.
[image]

esp-cjh, 2025-05-29:
Hi, @kobe-dace 
Release tag v5.2.6 will include this change.

kobe-dace, 2025-05-29:
Thanks for the update—I look forward to using the fix in the next release.